Koenkk/zigbee2mqtt #10529: [Problem]: Ikea LED1624G9 - UNSUPPORTED_ATTRIBUTE when changing color temperature

artur3, 2022-01-03:
### What happened?

 Ikea LED1624G9, TRADFRI bulb E27 CWS opal 600lm

No possibility to change color temperature. 

Publish 'get' 'color_temp' to 'TradfriBulb1' failed: 'Error: Read 0x90fd9ffffee4429a/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Zigbee2MQTT 1.22.2 commit: 414c51f5

### What did you expect to happen?

Change color temperature

### How to reproduce it (minimal and precise)

Refresh state (in web UI) or change color temperature

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210319

### Adapter

zStack3x0

### Debug log

debug 2022-01-03 10:48:00: Received MQTT message on 'zigbee2mqtt/TradfriBulb1/get' with data '{"color_temp":""}'
debug 2022-01-03 10:48:00: Publishing get 'get' 'color_temp' to 'TradfriBulb1'
debug 2022-01-03 10:48:00: Received Zigbee message from 'TradfriBulb1', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorMode":1,"currentX":39059,"currentY":25421}' from endpoint 1 with groupID 0
debug 2022-01-03 10:48:00: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x90fd9ffffee4429a!
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color_mode', payload 'xy'
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/update-state', payload 'idle'
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/state', payload 'ON'
error 2022-01-03 10:48:00: Publish 'get' 'color_temp' to 'TradfriBulb1' failed: 'Error: Read 0x90fd9ffffee4429a/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
debug 2022-01-03 10:48:00: Error: Read 0x90fd9ffffee4429a/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:310:23)
    at Endpoint.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:430:22)
    at Object.convertGet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1069:13)
    at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:272:21)
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/brightness', payload '151'
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-x', payload '0.596'
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-y', payload '0.3879'
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-hue', payload '16'
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-saturation', payload '100'
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color_temp', payload '580'
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/power_on_behavior', payload 'off'
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/last_seen', payload '2022-01-03T10:48:00+01:00'
info  2022-01-03 10:48:00: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/linkquality', payload '42'
debug 2022-01-03 10:48:02: Received MQTT message on 'zigbee2mqtt/TradfriBulb1/set' with data '{"color_temp":454}'
debug 2022-01-03 10:48:02: Publishing 'set' 'color_temp' to 'TradfriBulb1'
debug 2022-01-03 10:48:02: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x90fd9ffffee4429a!
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color_mode', payload 'xy'
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/update-state', payload 'idle'
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/state', payload 'ON'
debug 2022-01-03 10:48:02: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x90fd9ffffee4429a!
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color_mode', payload 'color_temp'
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/brightness', payload '151'
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/update-state', payload 'idle'
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-x', payload '0.596'
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-y', payload '0.3879'
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/state', payload 'ON'
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-hue', payload '16'
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/brightness', payload '151'
info  2022-01-03 10:48:02: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-saturation', payload '100'
[... 178 lines cut ...]
info  2022-01-03 10:51:13: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/power_on_behavior', payload 'off'
info  2022-01-03 10:51:13: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/last_seen', payload '2022-01-03T10:51:13+01:00'
info  2022-01-03 10:51:13: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/linkquality', payload '90'
debug 2022-01-03 10:51:14: Received MQTT message on 'zigbee2mqtt/TradfriBulb1/get' with data '{"color_temp":""}'
debug 2022-01-03 10:51:14: Publishing get 'get' 'color_temp' to 'TradfriBulb1'
debug 2022-01-03 10:51:14: Received Zigbee message from 'TradfriBulb1', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorMode":1,"currentX":39059,"currentY":25421}' from endpoint 1 with groupID 0
debug 2022-01-03 10:51:14: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x90fd9ffffee4429a!
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color_mode', payload 'xy'
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/update-state', payload 'idle'
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/state', payload 'ON'
error 2022-01-03 10:51:14: Publish 'get' 'color_temp' to 'TradfriBulb1' failed: 'Error: Read 0x90fd9ffffee4429a/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
debug 2022-01-03 10:51:14: Error: Read 0x90fd9ffffee4429a/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:310:23)
    at Endpoint.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:430:22)
    at Object.convertGet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1069:13)
    at Publish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.ts:272:21)
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/brightness', payload '151'
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-x', payload '0.596'
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-y', payload '0.3879'
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-hue', payload '16'
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color-saturation', payload '100'
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/color_temp', payload '580'
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/power_on_behavior', payload 'off'
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/last_seen', payload '2022-01-03T10:51:14+01:00'
info  2022-01-03 10:51:14: MQTT publish: topic 'zigbee2mqtt/TradfriBulb1/linkquality', payload '90'

flatsiedatsie, 2022-01-03:
I suspect I just noticed the same issue:

```
2022-01-03 11:19:51.604 INFO   : zigbee2mqtt-adapter: 
2022-01-03 11:19:51.611 INFO   : zigbee2mqtt-adapter: z2m stdout: Zigbee2MQTT:error 2022-01-03 11:19:51: Exception while calling fromZigbee converter: Cannot read property 'hasOwnProperty' of undefined}
2022-01-03 11:19:51.612 INFO   : zigbee2mqtt-adapter: 
2022-01-03 11:19:51.616 INFO   : zigbee2mqtt-adapter: z2m stdout: Zigbee2MQTT:error 2022-01-03 11:19:51: Failed to read state of '0xccccccfffecXXXXX' after reconnect (Read 0xccccccfffXXXXX1/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))

```

ALaDoffe, 2022-01-05:
Same issue on my side also.

```
debug 2022-01-05 15:29:59: Received MQTT message on 'zigbee2mqtt/0x5c0272fffe583793/get' with data '{"color_temp":""}'
debug 2022-01-05 15:29:59: Publishing get 'get' 'color_temp' to '0x5c0272fffe583793'
debug 2022-01-05 15:29:59: Received Zigbee message from '0x5c0272fffe583793', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorMode":1,"currentX":30015,"currentY":26869}' from endpoint 1 with groupID 0
debug 2022-01-05 15:29:59: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x5c0272fffe583793!
info  2022-01-05 15:29:59: MQTT publish: topic 'zigbee2mqtt/0x5c0272fffe583793', payload '{"brightness":100,"color":{"x":0.458,"y":0.41},"color_mode":"xy","color_temp":368,"color_temp_startup":368,"linkquality":21,"power_on_behavi
or":"previous","state":"OFF","update":{"state":null},"update_available":null}'
error 2022-01-05 15:29:59: Publish 'get' 'color_temp' to '0x5c0272fffe583793' failed: 'Error: Read 0x5c0272fffe583793/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"timeout":10000,"disableResponse":f
alse,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
debug 2022-01-05 15:30:00: Error: Read 0x5c0272fffe583793/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"
direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/srv/zigbee2mqtt-test/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:310:23)
    at Endpoint.read (/srv/zigbee2mqtt-test/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:430:22)
    at Object.convertGet (/srv/zigbee2mqtt-test/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1069:13)
    at Publish.onMQTTMessage (/srv/zigbee2mqtt-test/lib/extension/publish.ts:272:21)
```

flatsiedatsie, 2022-01-06:
@ALaDoffe does a firmware update of the bulb fix the issue?

artur3, 2022-01-06:
My is already on this version:
    20211025
    2.3.086

ALaDoffe, 2022-01-07:
@flatsiedatsie unfortunaly firmware update does not resolve the issue.
But perhaps there is a possible workaround with it ?

flatsiedatsie, 2022-01-07:
I wonder.. perhaps my earlier issue is related to this.

https://github.com/Koenkk/zigbee2mqtt/issues/10639

Roosted7, 2022-01-09:
I'm experiencing the same issue with all of the Tradfri lights that I recently updated

They do not change color when using `color_temp`, the `color_xy` commands still work as expected. Weirdly enough, the requested `color_temp` changes do not produce any error. The state json correctly updates, including a new LQI. It does however indeed error out when reconnecting.

The annoying consequence is that (through Nabu Casa / HA) Google Assistant is not capable of changing to the common 'white' colors... :(

I tried the dev branch and resetting + re-joining the light - but that seem to make a difference

Roosted7, 2022-01-09:
Of course I'm hoping this can be fixed 'properly' in the Zigbee stack - but for the meantime, I've created this workaround Home Assistant automation:

```
alias: Color_temp Tradfri fix
description: 'Fix the missing color_temp behaviour in recent new IKEA Tradfri firmware (when using Zigbee2MQTT). Make sure that the Zigbee2MQTT "color_sync" device setting is enabled, which is the default value.'
trigger:
  - platform: state
    entity_id:
      - light.YOURLIGHTHERE
      - light.YOURLIGHTHERE2
    attribute: color_mode
    to: color_temp
condition: []
action:
  - service: light.turn_on
    target:
      entity_id: '{{ trigger.entity_id }}'
    data:
      transition: 0.5
      xy_color: '{{ trigger.to_state.attributes.xy_color }}'
mode: parallel
max: 10
```

Just replace the entity_id with the one of your lights that don't respond to color temperature commands :)
It utilizes the `color_sync` translation, that updates the `xy_color` state when the `color_temp` is updated, to push a working xy_color value to the triggered entitiy.

Tawmu, 2022-01-10:
Can confirm this is an issue also. color_temp doesn't work at all with these bulbs.

Thanks to @Roosted7 for the temporary workaround 🥳

chiragd, 2022-02-09:
Are there any thoughts on how to fix this permanently?

chiragd, 2022-03-12:
Friendly reminder

Koenkk, 2022-04-12:
Not sure what we can do from the z2m perspective here, seems to some version of this bulb don't support color temperature.

chiragd, 2022-04-12:
When I have the bulb paired to the ikea tradfri hub it supports both colour temperature and colour via home assistant. When I pair it with Z2M, it no longer supports colour temperature via home assistant or the Z2M UI. Why is there a difference?

Koenkk, 2022-04-12:
@chiragd I believe via HA it will set the color temperature via the color X/Y

Koenkk, 2022-04-13:
@Roosted7 I expect that the translation is done in the TRADFRI gateway (but haven't verified this)

Xitro01, 2022-04-19:
> @Roosted7 I expect that the translation is done in the TRADFRI gateway (but haven't verified this)

Isn't it an idea to disable color temperature for this bulb and maybe create a feature like "color_sync", but then the other way around? Color temp -> x/y instead of x/y -> color temp.

chiragd, 2022-05-25:
This is still an issue. When paired with the tradfri hub everything works great. I can even ask Google assistant to set various colour temps of white and it works fine. But now when paired to zigbee2mqtt, the bulb doesn't respond to colour temp changes. I was planning to remove the tradfri hub from my set up but now I will have to keep it just to support a few bulbs. Also the home assistant and z2m UI both state this bulb can support colour temp but the UI doesn't do anything, which causes confusion for other users in the house when the bulb is unresponsive. 

Is there anything that can be done or should I just re-pair with the tradfri hub?

Roosted7, 2022-05-25:
@chiragd Did you try my workaround from Jan 9?

I know it's not a true 'fix' - but I've been using this for a while and I can't tell any difference from my workaround and native lights

Tawmu, 2022-07-10:
Unfortunately this is still an issue

Xitro01, 2022-08-18:
> @chiragd Did you try my workaround from Jan 9?
> 
> I know it's not a true 'fix' - but I've been using this for a while and I can't tell any difference from my workaround and native lights

Used this for a long while, but with one of the Z2M updates it suddenly doesn't report color_temp anymore. So the workaround is not working anymore. Anybody found another workaround?

Xitro01, 2022-09-03:
@Koenkk Could you hack something for this?

Koenkk, 2022-09-06:
@Xitro01 I'm on holiday so can't test, can you check with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const color = require('zigbee-herdsman-converters/lib/color');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    colortemp: {
        key: ['color_temp'],
        options: [exposes.options.color_sync(), exposes.options.transition()],
        convertSet: async (entity, key, value, meta) => {
            const c = new color.ColorXY.fromMireds(value);
            const zclData = {transtime: utils.getTransition(entity, key, meta).time, x: c.x, y: c.y};
            await entity.command('lightingColorCtrl', 'moveToColor', zclData, utils.getOptions(meta.mapped, entity));
            return {
                state: libColor.syncColorState({'color_mode': constants.colorMode[2], 'color_temp': value}, meta.state,
                    entity, meta.options, meta.logger), readAfterWriteTime: payload.transtime * 100,
            };
        },
    }
}

const definition = {
    zigbeeModel: ['TRADFRI bulb E27 CWS opal 600lm', 'TRADFRI bulb E26 CWS opal 600lm', 'TRADFRI bulb E14 CWS opal 600lm',
        'TRADFRI bulb E12 CWS opal 600lm', 'TRADFRI bulb E27 C/WS opal 600'],
    model: 'LED1624G9',
    vendor: 'IKEA',
    description: 'TRADFRI LED bulb E14/E26/E27 600 lumen, dimmable, color, opal white',
    extend: extend.light_onoff_brightness_colortemp_color(),
    toZigbee: [tz.light_onoff_brightness, tz.light_color, tzLocal.colortemp],
    meta: {supportsHueAndSaturation: false},
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

Xitro01, 2022-09-06:
@Koenkk I placed the ext_converter.js next to configuration.yaml, with the code above. In the zigbee2mqtt folder.
After that I restart Z2M, but the bulbs are still not showing color_temp.

Koenkk, 2022-09-06:
Whoops, updated https://github.com/Koenkk/zigbee2mqtt/issues/10529#issuecomment-1237750671

Xitro01, 2022-09-06:
> Whoops, updated [#10529 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/10529#issuecomment-1237750671)

Had to add three dependencies: utils, libColor and constants, after that the following error:
`Publish 'set' 'color_temp' to 'Gang' failed: 'ReferenceError: payload is not defined'`

Koenkk, 2022-09-06:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/10529#issuecomment-1237750671 again

Xitro01, 2022-09-06:
> Updated [#10529 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/10529#issuecomment-1237750671) again

Seems to work now (in the UI), but in reality the lamp just turns off completely setting any color_temp.
(Please add the 3 dependencies in your code as well, these are required.)

Koenkk, 2022-09-07:
I would expect that the converter now sends an x/y color to the bulb. I will check further when I'm home from holiday (so I can test it myself).

Xitro01, 2022-09-20:
> I would expect that the converter now sends an x/y color to the bulb. I will check further when I'm home from holiday (so I can test it myself).

Had some more time, found this in my logs:
``Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x000d6ffffe21df92!``

Koenkk, 2022-09-21:
I've added support for `color_temp` now, let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Xitro01, 2022-09-21:
> I've added support for `color_temp` now, let me know if it works.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

That is amazing work, I think a lot of people will be very happy about this. Thanks!

I will wait till it arrives in the default branch, using Home Assistant OS and don't want to get into trouble with my config by switching to edge.

Xitro01, 2022-10-01:
Works perfect after the just released update. This issue can be closed now! Thanks alot @Koenkk

closed 2022-10-01 by Koenkk

TebbeUbben, 2022-10-06:
`color_temp_step` and `color_temp_move` still don't seem to be working. `color_temp_step` fails with an error:

```
Zigbee2MQTT:debug 2022-10-06 13:11:29: Received MQTT message on 'zigbee2mqtt/light_desk/set' with data '{"color_temp_step":17}'
Zigbee2MQTT:debug 2022-10-06 13:11:29: Publishing 'set' 'color_temp_step' to 'light_desk'
Zigbee2MQTT:debug 2022-10-06 13:11:29: Received Zigbee message from 'light_desk', type 'readResponse', cluster 'lightingColorCtrl', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-10-06 13:11:29: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x588e81fffe209e0f!
Zigbee2MQTT:error 2022-10-06 13:11:29: Exception while calling fromZigbee converter: Cannot read properties of undefined (reading 'hasOwnProperty')}
Zigbee2MQTT:debug 2022-10-06 13:11:29: TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')
    at Function.fromObject (/app/node_modules/zigbee-herdsman-converters/lib/color.js:207:17)
    at Function.fromMireds (/app/node_modules/zigbee-herdsman-converters/lib/color.js:220:24)
    at Object.syncColorState (/app/node_modules/zigbee-herdsman-converters/lib/color.js:750:32)
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/converters/fromZigbee.js:620:51)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51)
    at EventEmitter.emit (node:events:525:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:513:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
Zigbee2MQTT:error 2022-10-06 13:11:29: Publish 'set' 'color_temp_step' to 'light_desk' failed: 'Error: Read 0x588e81fffe209e0f/1 lightingColorCtrl(["colorTemperature"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Zigbee2MQTT:debug 2022-10-06 13:11:29: Error: Read 0x588e81fffe209e0f/1 lightingColorCtrl(["colorTemperature"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:721:17)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
```

`color_temp_move` doesn't produce any errors in the logs, but still doesn't change the color temperature.

Is there a chance to get those working, too?