Koenkk/zigbee2mqtt #10535: [New device support]: Konke model: 'KK-ES-J01W', Smart Brightness, Room Temperature and Humidity Sensor with LCD display

convicte, 2022-01-03:
### Link

http://m.konkesmart.com/sensor/temperature-and-humidity-sensor/temperature-sensor-and-humidity-sensor.html

### Database entry

{"id":43,"type":"EndDevice","ieeeAddr":"0x84fd27fffe98ff53","nwkAddr":61437,"manufId":4712,"manufName":"_TYZB01_fi5yftwv","powerSource":"Battery","modelId":"TS0222","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[0,1,3,1024,1026,1029,2821,64704],"outClusterList":[25],"clusters":{"msIlluminanceMeasurement":{"attributes":{"measuredValue":21462}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2580}},"msRelativeHumidity":{"attributes":{"measuredValue":4210}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":9,"stackVersion":2,"hwVersion":1,"dateCode":"20211012","swBuildId":"01-2019120401","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1641210297117,"useImplicitCheckin":true}

### Comments

Seems to be working OK with the converter listed below.
Obviously the image is missing but pairs successfully every time and reports all the values.

[image]

It's not reporting as often as I would want and sometimes does not respond to brief parameter changes (covering with one's hand for a second does not trigger lux change unless you hold it for 2-3 sec).

Nonetheless, a very nice looking device which should work well for a wall mounted LCD environmental sensor and luminance detector for light automations.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const definition = {
    fingerprint: [{modelID: 'TS0222', manufacturerName: '_TYZB01_fi5yftwv'}],
    model: 'KK-ES-J01W',
    vendor: 'Konke',
    description: 'Smart Brightness, Room Temperature and Humidity Sensor',
    fromZigbee: [fz.battery, fz.illuminance, fz.humidity, fz.temperature],
    toZigbee: [],
    exposes: [e.battery(), e.illuminance(), e.illuminance_lux().withUnit('lx'), e.humidity(), e.temperature()],
};
module.exports = definition;
```

Koenkk, 2022-01-03:
looks good, could you make a pr? You can do this by clicking the edit/pencil icon here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/konke.js

convicte, 2022-01-03:
The PR is in, but I seem to have messed up the closing bracket and semicolon on the PR. It's however locked now, and I can't edit this out. 

Thank you!

Koenkk, 2022-01-04:
Thanks, added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-04 by Koenkk