Koenkk/zigbee2mqtt #10538: [Problem]: network scan results in 'Failed to execute LQI' for lumi.plug.maus01

bigbangus, 2022-01-03:
### What happened?

After updating to the latest CC2652 firmware and zigbee2mqtt docker version, my network scan results in a failed to execute LQI for the only Aquara plug on my network.
As a secondary issue, since my update in the last day or so, my plug no longer seems to be routing traffic. The network map no longer shows it being used for any of the devices. Also, my network scans don't report LQIs for my devices. It used to show all the LQIs. Is there some kind of initialization period when you update the firmware on your CC2652 stick?

### What did you expect to happen?

No error when performing a network scan to generate a map
Network map has device LQIs populated
Network shows some devices connected to plug (router)

### How to reproduce it (minimal and precise)

It's the current state. If I reflash and restart zigbee2mqtt the same thing happens.

FYI running in docker on unraid 6.10rc2.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211217

### Adapter

TubeZB CC2652P2 Based Zigbee to PoE Coordinator V2

### Debug log

[herdsman-debug-aquara-plug-lqi.txt](https://github.com/Koenkk/zigbee2mqtt/files/7802374/herdsman-debug-aquara-plug-lqi.txt)

bigbangus, 2022-01-03:
Network map:
[image]

alex-savin, 2022-01-03:
I have the same issue with the same coordinator (TubeZB CC2652P2 Based Zigbee to PoE Coordinator V2)

rainisto, 2022-01-08:
I'm also having failed to execute LQI and failed to execute routing table with zzh (CC2652R_coordinator_20211217.hex) against Ikea TRADFRI routers.

bigbangus, 2022-02-02:
Seems to work ok in version 1.23.0. Gonna close this out.

closed 2022-02-02 by bigbangus