Koenkk/zigbee2mqtt #10540: z2m changes every day the valve date/time or reset it - valve schedule/automatic heating use false hour/day for schedule

gmarchart, 2022-01-03:
### What happened?

Zigbee2mqtt
Current version: 1.22.2-1
Home Assistant 2021.12.7
Host-Betriebssystem | Home Assistant OS 7.0
-- | --
Update-Channel | stable
Supervisor-Version | supervisor-2021.12.2
Docker-Version | 20.10.9

Hersteller
Lidl
Modell
368308_2010

Zigbee Modell
TS0601
Zigbee Hersteller
_TZE200_chyvmhay
Beschreibung
Silvercrest radiator valve with thermostat

every day z2m changes the date/time on valve or do a reset of it. this did not occurs if the valve is not binded/used by z2m. so no issue with valve. if i set the date/time today on valve - lates tomorrow morning the valve have the false date/time again.

no external converter used, only the internal.

so the automatic schedule from valve is unusable. 

with the former edge version with custom converter this was not happend, only occurs since i use the official add on with the internal converter for this supported valve. valve always gets a false year/month/hour/minute - so the internal schedule of the valve does the wrong heating/no heating to a false time...

thanks for fixing that issue - makes the automatic schedule on valve unusable !

### What did you expect to happen?

that z2m did not reset/change the date/time on the valve every day.

### How to reproduce it (minimal and precise)

if i set today the date/time - tomorrow morning the date/time is wrong on valve - every day i change it to the corrext value next day it is wrong ... make the automatic modus/schedule of the valve unusable.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26660700

### Adapter

deconz

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7802752/log.txt)

groenmarsmannetje, 2022-01-04:
Same problem as #10489. This has been fixed in dev/edge version.

gmarchart, 2022-01-04:
> Same problem as #10489. This has been fixed in dev/edge version.

Ok. Thanks. Will check it today one more time with dev/edge version and give a feedback.

gmarchart, 2022-01-05:
> Same problem as #10489. This has been fixed in dev/edge version.

hello, seems to be ok - after 1 day no date/time change on the valve occurs - looks good !
i will wait one more day, if nothing goes wrong, i will close this issue and wait until #10489 will be in the official branch added.
for now i can go with edge - all ok.

thanks !

gmarchart, 2022-01-06:
After 2 days no issue. Works as expected with this edge branche now.

closed 2022-01-06 by gmarchart

FilipCZE, 2022-01-30:
Hi I still have this problem... Im using z2m, do I need to switch to edge? r how can I fix this?? thanks!