Koenkk/zigbee2mqtt #10542: [Problem]: Warning "UnhandledPromiseRejectionWarning"

Ivano62, 2022-01-03:
### What happened?

Since version 1.22.1 this warnings are logged repeatedly in the log 

`(node:7) UnhandledPromiseRejectionWarning: RangeError: Offset is outside the bounds of the DataView
    at DataView.getUint16 (<anonymous>)
    at parseReadReceivedDataResponse (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:257:32)
    at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:458:32)
    at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:405:59)
    at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:63:59)
    at Driver.emit (events.js:400:28)
    at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)
    at Parser.emit (events.js:400:28)
    at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/zigbee-herdsman/node_modules/slip/src/slip.js:189:18)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 563)
(node:7) UnhandledPromiseRejectionWarning: undefined
`

### What did you expect to happen?

The warnings do not occur as they previously did 

### How to reproduce it (minimal and precise)

In normal execution

### Zigbee2MQTT version

1.22.2-dev (commit: 23bdc18)

### Adapter firmware version

0x26720700

### Adapter

RaspBee 2

### Debug log

`(node:7) UnhandledPromiseRejectionWarning: RangeError: Offset is outside the bounds of the DataView
    at DataView.getUint8 (<anonymous>)
    at parseReadReceivedDataResponse (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:214:33)
    at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:458:32)
    at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:405:59)
    at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:63:59)
    at Driver.emit (events.js:400:28)
    at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)
    at Parser.emit (events.js:400:28)
    at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/zigbee-herdsman/node_modules/slip/src/slip.js:189:18)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 563)
(node:7) UnhandledPromiseRejectionWarning: undefined
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 565)
(node:7) UnhandledPromiseRejectionWarning: undefined
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 567)
(node:7) UnhandledPromiseRejectionWarning: undefined
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 569)
(node:7) UnhandledPromiseRejectionWarning: undefined
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 571)
`

Koenkk, 2022-01-03:
@chrishae could you check this?

ChrisHae, 2022-01-11:
I can not recreate the errors.

@Ivano62 could you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Ivano62, 2022-01-11:
@ChrisHae Ithink the problem is with docker. The log I provided is the output of `docker container logs zigbee2mqtt` command

ChrisHae, 2022-01-11:
When did this error occurred the first time?

Ivano62, 2022-01-11:
I think after latest docker upgrade to version 20.10.12

Ivano62, 2022-01-11:
I've captured some log, the problem appears to occur after Home Assistant reboot
[log20220111.txt](https://github.com/Koenkk/zigbee2mqtt/files/7847196/log20220111.txt)

ChrisHae, 2022-01-13:
Perhaps I can implement something to prevent that error. But I need a more detailed debug log. Have you done this to enable debug log?

> To enable debug logging in the Zigbee2MQTT Docker container add -e DEBUG=zigbee-herdsman* to your docker run command. The logging can be retrieved via docker logs ZIGBEE2MQTT_CONTAINER_NAME > log.txt 2>&1

Ivano62, 2022-01-13:
@ChrisHae I stopped log registration right after the first occurrence of the error. You can find it on line 11344. Thanks
[log_docker.txt](https://github.com/Koenkk/zigbee2mqtt/files/7862268/log_docker.txt)

ChrisHae, 2022-01-25:
The fix for the unhandled rejection warnings should be in the next dev release.
Could you check this?

Ivano62, 2022-01-25:
@ChrisHae  Dev 5b9b7e4?

Ivano62, 2022-01-26:
@ChrisHae The same errors with latest dev (c4c02d2)

Koenkk, 2022-01-26:
@Ivano62 the changes are not in the latest dev yet, @ChrisHae is looking into it.

Ivano62, 2022-01-28:
@ChrisHae I've installed dev 2721cc1, after about 20 minutes I get

(node:7) UnhandledPromiseRejectionWarning: undefined

(Use `node --trace-warnings ...` to show where the warning was created)

(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)

(node:7) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

(node:7) UnhandledPromiseRejectionWarning: undefined

(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 4)

(node:7) UnhandledPromiseRejectionWarning: undefined

(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 6)

(node:7) UnhandledPromiseRejectionWarning: undefined

(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 8)

(node:7) UnhandledPromiseRejectionWarning: undefined

(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 10)

(node:7) UnhandledPromiseRejectionWarning: undefined

(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 12)

(node:7) UnhandledPromiseRejectionWarning: undefined

(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 14)

ChrisHae, 2022-01-31:
Should be fixed now.

Ivano62, 2022-01-31:
I'm sorry, I've installed dev 9fc413d and see the same errors:

...
(node:7) UnhandledPromiseRejectionWarning: Error: received frame size to small
    at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:455:15)
    at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:414:59)
    at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:63:59)
    at Driver.emit (events.js:400:28)
    at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)
    at Parser.emit (events.js:400:28)
    at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)
    at slip.Decoder.p.handleEnd (/app/node_modules/zigbee-herdsman/node_modules/slip/src/slip.js:189:18)
    at slip.Decoder.p.decode (/app/node_modules/zigbee-herdsman/node_modules/slip/src/slip.js:143:32)
    at Parser._transform (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:39:22)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 30)
...

ChrisHae, 2022-01-31:
No problem. That should be the last error of this kind fixed in the next release.

Ivano62, 2022-01-31:
@ChrisHae on dev 0429027 I get:

(node:8) UnhandledPromiseRejectionWarning: undefined
(Use `node --trace-warnings ...` to show where the warning was created)
(node:8) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)

(node:8) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

ChrisHae, 2022-02-01:
I have no clue where this error originated from. Do you have any more detailed debug output like the one above (which contains function names and line numbers) ?

Ivano62, 2022-02-01:
@ChrisHae I stopped log registration right after the second occurrence of the error. You can find it on lines 22339 and 22882. Thank
[log_docker.txt](https://github.com/Koenkk/zigbee2mqtt/files/7978035/log_docker.txt)
s

ChrisHae, 2022-02-02:
You got several unhandled exceptions warnings because some frames were not received or sent correctly. I did my best to remove all these warnings. Please try the next release it should get rid of the last exception you posted. It would be nice if you provide a debug log again because I added some debug output that would be helpfull for me.

That said the source of the errors you received is going deeper. I would recommend that you set the "delay" value in the zigbee2mqtt configuration to a higher value. You could try values of 50, 100 or 200 and see if you get less errors.

Ivano62, 2022-02-02:
@ChrisHae I tried latest 1.23.0. Could you explain me the 'delay' parameter? Thanks

[log_docker.txt](https://github.com/Koenkk/zigbee2mqtt/files/7986211/log_docker.txt)

ChrisHae, 2022-02-02:
In the configuration.yaml in the zigbee2mqtt data directory add e.g. `adapter_delay:200` in the advanced section (I meant adapter_delay not delay)

the default value is 0 and it adds some milliseconds between the data send/ and receiving requests. This can help when sending and receiving many requests at the same time.

ChrisHae, 2022-02-02:
P.S. I think this was not the latest commit I uploaded (Commit 105165f) perhaps there will be a new release soon.

Ivano62, 2022-02-02:
@ChrisHae on dev fae8e96 with adapter_delay:50, after about three hours there isn't any errors.

ysimonx, 2022-02-23:
Hello.

Same problem here, with raspberry 3 model B, raspberry OS, docker, official zigbee2mqtt docker, phoscon conbee II (usb)

my yaml is : 

```
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: false

frontend:
  port: 8080

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://172.17.0.1:1883'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0

advanced:
  adapter_delay: 200
```

logs are

`      

Using '/app/data' as data directory

Zigbee2MQTT:info  2022-02-23 18:10:41: Logging to console and directory: '/app/data/log/2022-02-23.18-10-38' filename: log.txt

Zigbee2MQTT:info  2022-02-23 18:10:41: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61ba)

Zigbee2MQTT:info  2022-02-23 18:10:41: Starting zigbee-herdsman (0.14.10)

CREATED DECONZ ADAPTER

Zigbee2MQTT:info  2022-02-23 18:10:42: zigbee-herdsman started (resumed)

Zigbee2MQTT:info  2022-02-23 18:10:42: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'

Zigbee2MQTT:info  2022-02-23 18:10:42: Currently 0 devices are joined:

Zigbee2MQTT:info  2022-02-23 18:10:42: Zigbee: disabling joining new devices.

Zigbee2MQTT:info  2022-02-23 18:10:43: Connecting to MQTT server at mqtt://172.17.0.1:1883

Zigbee2MQTT:info  2022-02-23 18:10:43: Connected to MQTT server

Zigbee2MQTT:info  2022-02-23 18:10:44: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

Zigbee2MQTT:info  2022-02-23 18:10:44: Started frontend on port 0.0.0.0:8080

Zigbee2MQTT:info  2022-02-23 18:10:44: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"6ea61ba","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":false,"version":"1.23.0"}'

(node:8) UnhandledPromiseRejectionWarning: RangeError: Offset is outside the bounds of the DataView

    at DataView.getUint32 (<anonymous>)
[... 145 lines cut ...]
(node:8) UnhandledPromiseRejectionWarning: RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 15. Received 16

    at new NodeError (internal/errors.js:322:7)

    at boundsError (internal/buffer.js:85:9)

    at Buffer.readUInt8 (internal/buffer.js:251:5)

    at BuffaloZcl.readUInt8 (/app/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:46:35)

    at BuffaloZcl.readGdpFrame (/app/node_modules/zigbee-herdsman/src/zcl/buffaloZcl.ts:275:31)

    at BuffaloZcl.read (/app/node_modules/zigbee-herdsman/src/zcl/buffaloZcl.ts:443:25)

    at Function.parsePayloadCluster (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:245:47)

    at Function.parsePayload (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:216:25)

    at Function.fromBuffer (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:186:30)

    at DeconzAdapter.checkReceivedGreenPowerIndication (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1050:29)

(node:8) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 7)

      `

Koenkk, 2022-02-23:
@ChrisHae could you check this?

ysimonx, 2022-02-23:
@Koenkk & @ChrisHae 

I just saw that there is a specific setting for conbee II

so, i changed configuration.yaml with

`
# Serial settings
serial:
  adapter: deconz
  port: /dev/ttyACM0
`

still log errors with

`
Using '/app/data' as data directory

Zigbee2MQTT:info  2022-02-23 19:26:52: Logging to console and directory: '/app/data/log/2022-02-23.19-26-49' filename: log.txt

Zigbee2MQTT:info  2022-02-23 19:26:53: Starting Zigbee2MQTT version 1.23.0 (commit #6ea61ba)

Zigbee2MQTT:info  2022-02-23 19:26:53: Starting zigbee-herdsman (0.14.10)

CREATED DECONZ ADAPTER

Zigbee2MQTT:info  2022-02-23 19:26:54: zigbee-herdsman started (resumed)

Zigbee2MQTT:info  2022-02-23 19:26:54: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'

Zigbee2MQTT:info  2022-02-23 19:26:54: Currently 0 devices are joined:

Zigbee2MQTT:info  2022-02-23 19:26:54: Zigbee: disabling joining new devices.

Zigbee2MQTT:info  2022-02-23 19:26:55: Connecting to MQTT server at mqtt://172.17.0.1:1883

Zigbee2MQTT:info  2022-02-23 19:26:56: Connected to MQTT server

Zigbee2MQTT:info  2022-02-23 19:26:56: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'

Zigbee2MQTT:info  2022-02-23 19:26:56: Started frontend on port 0.0.0.0:8080

Zigbee2MQTT:info  2022-02-23 19:26:56: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"6ea61ba","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":false,"version":"1.23.0"}'

(node:7) UnhandledPromiseRejectionWarning: RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 15. Received 16

    at new NodeError (internal/errors.js:322:7)

    at boundsError (internal/buffer.js:85:9)

    at Buffer.readUInt8 (internal/buffer.js:251:5)

    at BuffaloZcl.readUInt8 (/app/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:46:35)

    at BuffaloZcl.readGdpFrame (/app/node_modules/zigbee-herdsman/src/zcl/buffaloZcl.ts:275:31)

    at BuffaloZcl.read (/app/node_modules/zigbee-herdsman/src/zcl/buffaloZcl.ts:443:25)

    at Function.parsePayloadCluster (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:245:47)

    at Function.parsePayload (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:216:25)

    at Function.fromBuffer (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:186:30)

    at DeconzAdapter.checkReceivedGreenPowerIndication (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1050:29)

(Use `node --trace-warnings ...` to show where the warning was created)

(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)

(node:7) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

(node:7) UnhandledPromiseRejectionWarning: RangeError: Offset is outside the bounds of the DataView

    at DataView.getUint32 (<anonymous>)

    at parseGreenPowerDataIndication (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:339:33)

    at parseFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:466:29)

    at processFrame (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/frameParser.ts:414:59)

    at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:63:59)

    at Driver.emit (events.js:400:28)

    at Driver.onParsed (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:784:14)

    at Parser.emit (events.js:400:28)

    at Parser.onMessage (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/parser.ts:30:14)

    at slip.Decoder.p.handleEnd (/app/node_modules/zigbee-herdsman/node_modules/slip/src/slip.js:189:18)
`

ysimonx, 2022-02-23:
it seems that i need to install deconz ... but i did not manage yet to install it correctly

ysimonx, 2022-02-23:
ok @ChrisHae : i managed to install deconz and to update firmware of conbee2

however, still have issue with zigbee2mqtt

can you help me please ?

cf logs

`
Zigbee2MQTT:info  2022-02-23 22:17:26: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"6ea61ba","coordinator":{"meta":{"maintrel":0,"majorrel":38,"minorrel":114,"product":0,"revision":"0x26720700","transportrev":0},"type":"ConBee2/RaspBee2"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":false,"version":"1.23.0"}'

(node:8) UnhandledPromiseRejectionWarning: RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 15. Received 16

    at new NodeError (internal/errors.js:322:7)

    at boundsError (internal/buffer.js:85:9)

    at Buffer.readUInt8 (internal/buffer.js:251:5)

    at BuffaloZcl.readUInt8 (/app/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:46:35)

    at BuffaloZcl.readGdpFrame (/app/node_modules/zigbee-herdsman/src/zcl/buffaloZcl.ts:275:31)

    at BuffaloZcl.read (/app/node_modules/zigbee-herdsman/src/zcl/buffaloZcl.ts:443:25)

    at Function.parsePayloadCluster (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:245:47)

    at Function.parsePayload (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:216:25)

    at Function.fromBuffer (/app/node_modules/zigbee-herdsman/src/zcl/zclFrame.ts:186:30)

    at DeconzAdapter.checkReceivedGreenPowerIndication (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:1050:29)
`

ChrisHae, 2022-03-01:
The error is thrown while reading the ZCLFrame. So it seems that the received greenpower frame is invalid.
I would need to see the zigbee sniffer log to be able to do sth.
A zigbee-herdsman debug log would also be helpfull. https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging
Could you tell me which device is causing the error? I could ask if someone has experiences with that device.

closed 2022-04-09 by github-actions[bot]