Koenkk/zigbee2mqtt #10543: Frient device = Develco device

RitaSghe, 2022-01-03:
### Link

https://frient.com/products/motion-sensor/

### Database entry

xxxx

### Comments

Frient device are clearly Develco device with another brand, but are the same ...
Motion sensor, smart plug, button, door sensor are the same ....

Can you duplicate the Develco device to pairing the Frient devices also?

rindlerblabla, 2022-01-03:
If you add a Frient device it is successfully added if there's support for the respective Develco device. They share the same ID which is recognised by zigbee2mqtt.

RitaSghe, 2022-01-04:
Yes, I was hoping so, but .....
FRIENT SMART PLUG: pairing ok but also in state OFF the plug send status constantly, every 2/3 seconds and I don't want this; when the status is ON, yes also for monitoring power, but why in OFF?

FRIENT MOTION SENSOR: unrecognized device but there is in Develco device, the same .....

RobLeighton67, 2022-01-04:
I'd also raise the issue that the MOSZB-140 motion sensor has very short minimum reporting intervals and low reporting thresholds - my device is reporting Illuminance and Temperature every 12s or so !

Claude2666, 2022-01-06:
I have also a Frient smart plug, and after pairing, several entries in the reporting tab of the front end appeared to reduce the number of reports. Maybe try remove and pair again?

[image]

I have noticed that for some devices it is possible to control the reporting via the reporting tab, but not for all...

RobLeighton67, 2022-01-06:
I've solved the issue by modifying the settings for MOSZB-140 in the Develco.js devices file.

RitaSghe, 2022-01-06:
@Claude2666
Thank you ....

closed 2022-02-14 by github-actions[bot]