Koenkk/zigbee2mqtt #10547: [Problem]: TuYa TV02 Thermostat doesn't update current temp unless local_temperature_calibration is changed

YvoTuk, 2022-01-03:
### What happened?

The TV02 local_temperature is stuck on the same temperature all the time.
Only when I change the local_temperature_calibration is changed, the local_temperature is updated.

### What did you expect to happen?

local_temperature would be updated frequently

### How to reproduce it (minimal and precise)

Install as is and check for variation of local_temperature

### Zigbee2MQTT version

1.22.2-1

### Adapter firmware version

20210120

### Adapter

CC2652RB Slaesh

### Debug log

not sure if this has anything to do with it:
22-01-03 20:58:06 WARNING (MainThread) [supervisor.addons.options] Unknown option 'new_api' for Zigbee2mqtt (45df7312_zigbee2mqtt)

Lipown, 2022-01-04:
Same issue here. Also valve will not open if current temp is not at least around 2°C below setpoint and will not close is closing slowly causing the setpoint overshoot by 4°C and more.  I did automation to publish local_temperature_calibration 0 value each few minutes to valve but this is not solution for all issues.

groenmarsmannetje, 2022-01-04:
Same problem as #10489. This has been fixed in dev/edge version

YvoTuk, 2022-01-04:
> Same issue here. Also valve will not open if current temp is not at least around 2°C below setpoint and will not close is closing slowly causing the setpoint overshoot by 4°C and more. I did [...]

I thought I saw the same thing happening to mine but didn't want to report it as I didn't spend time verifying what I thought.

To me it seems like the integration isn't quite compatible with the device. The operations are not compatible with the device (heat, cool, auto, auto, dry, fan only). From what I found out so far, only the modes are compatible (None (off), Auto, manual, and holiday)

Also, it's not possible to access or change the program through HA. You can see the program in Zigbee2MQTT, but it's not possible to change.

Koala-lyon, 2022-01-05:
Hey !
Same issue here, not possible to update the current temperature, unless by changing the offset, which is not a solution... 
Parameters HEAT, COOL, OFF available through Homebridge are not functionnal at all. It's only possible to change the SP. Possible to act on other settings throught z2m frontend web-page otherwise (except the schedule).
It's specified that it's not possible do /get the current temp, but at least the device should refresh the value from time to time no? 
I did not try to use the TV02 device with HA.

Lipown, 2022-01-07:
I am not able to close the valve either by setting to zero, heat stop or window open. It works time to time but if I check radiator each hour, the heat sometimes comming from it anyway. Since this valve does not report the status of itself in percent I am affraid that the control is not working.

Lipown, 2022-01-11:
Heating stop warmed my rooms to 26 degrees today. It does not working. Instead I will try to use 0°C to shut off the valve.

XiChar, 2022-01-18:
@Lipown did you find a solution? I got the same problem, my valve never fully closes. I tried heating_stop, open_window and setting the heating point below the current measured temperature, but the valve always remains slightly open.

Lipown, 2022-01-25:
> @Lipown did you find a solution? I got the same problem, my valve never fully closes. I tried heating_stop, open_window and setting the heating point below the current measured temperature, but the valve always remains slightly open.

I thouth it has some PID regulation so it needs to be calibrated for certain time. It seemed that all have worked for a week but today I cannot open the valve no matter what. My GF is freezing at home :/

vladi1234, 2022-02-05:
Hi, 
please read here. You can find everything there.

https://github.com/Koenkk/zigbee2mqtt/issues/8522#issuecomment-999039475
https://github.com/Koenkk/zigbee-herdsman-converters/issues/3142#issuecomment-976885156
https://github.com/Koenkk/zigbee2mqtt/issues/8522#issuecomment-990070405
https://github.com/Koenkk/zigbee2mqtt/issues/8522#issuecomment-997474913

jypsie, 2022-02-24:
same problem here, no update of current temp and I have the impression other things are not working too.

prometheus86, 2022-03-19:
Hi,
I  connected the TRV to a Tuya gateway. So it could update the firmware of the trv. Now it seems to update the current temp correctly for about one day.

Mr-Beat, 2022-03-23:
My my solution via /config/automations.yaml:
```

- id: '1646558426XXX'
  alias: 'Reed and Update Local Temp'
  description: 'No description'
  trigger:
  - platform: time_pattern
    minutes: /15
  condition: []
  action:
  - device_id: 13b35ca332032a902b5c8e570a1XXXXX
    domain: number
    entity_id: number.0x2c1165fffecXXXXX_local_temperature_calibration
    type: set_value
    value: 0
  mode: single
```

Lipown, 2022-04-03:
1. Any idea how to update firmware if you do not have tuya gateway but using usb coordinator please? 
2. How to control target temperature by external sensor? Example. Target is 22°C. When sun is shining, valve is open even if in the room is 27°C on the external temperature sensor.

21stce, 2022-04-06:
@Lipown 
1. This is somewhat documented on the website, but it's advanced and requires some equipment (I don't have it, for example). 
2. The device doesn't support closing the valve _completely_, which makes it a scam. You can close it the most by enabling `heating_stop` and disabling `frost_protection`.

Impact123, 2022-04-06:
[The update/image URL is already available here](https://github.com/Koenkk/zigbee-OTA/issues/104). Once the issue gets resolved an OTA update might not be too far away.

sebastianschauenburg, 2022-05-07:
> [The update/image URL is already available here](https://github.com/Koenkk/zigbee-OTA/issues/104). Once the issue gets resolved an OTA update might not be too far away.

Seeing the latest reply there, it currently seems we're out of luck for now.

closed 2022-06-16 by github-actions[bot]

rotilho, 2022-08-15:
Did anyone managed to upgrade without the Tuya Gateway?

J-Maertens, 2022-09-30:
Any news about upgrading firmware without the Tuya hub?

Tntdruid, 2022-11-15:
No news on firmware update whit no hub?

damon-t, 2022-11-28:
Is there any chance to get an update on this?

Pheggas, 2023-11-29:
Mine updates temp every half an hour. Even if the temperature drops 1°C, it doesn't care and will update exactly every half an hour