Koenkk/zigbee2mqtt #10548: Moes Smart Curtain Switch ZTS-EUR-C

adamis24, 2022-01-03:
### Link

https://www.aliexpress.com/item/1005003311312070.html?spm=a2g0o.productlist.0.0.98fb39fdrFmx3G&algo_pvid=83587ddc-e9a0-4d3c-a2b2-646e5051778a&algo_exp_id=83587ddc-e9a0-4d3c-a2b2-646e5051778a-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000025143742708%22%7D&pdp_pi=-1%3B18.53%3B-1%3B-1%40salePrice%3BEUR%3Bsearch-mainSearch

### Database entry

Router, TS0601, _TZE200_nhyj64w2, 0x2c1165fffe9e60d9

### Comments

I try to add this new device and I´m not succesfull

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
 
const definition = { 
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_nhyj64w2'}],
        zigbeeModel: ['TS0601'],  
        model: 'TS0202',
        vendor: 'TuYa', 
        description: 'Curtain switch',
		supports: 'open, close, stop, position',
		fromZigbee: [],
		toZigbee: [tz.cover_state],
 
        //fromZigbee: [fz.tuya_curtain, fz.ignore_basic_report],
        //toZigbee: [tz.tuya_curtain_control, tz.tuya_curtain_options],
 
        //from Zigbee: [fz.tuya_switch, fz.ZNCLDJ11LM_ZNCLDJ12LM_curtain_options_output, fz.ignore_basic_report, fz.ignore_time_read, fz.cover_position_tilt, fz.ZNCLDJ11LM_ZNCLDJ12LM_curtain_analog_output],
        //toZigbee: [],
        //exposes: [e.switch()],
};
 
module.exports = definition;
```

closed 2022-03-01 by github-actions[bot]

migor7, 2022-03-23:
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
This converter works. Calibration does not work. Possible device and does not support calibrationconst exposes = require('zigbee-herdsman-converters/lib/exposes');

const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");
 
const definition = { 
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_nhyj64w2'}],
        zigbeeModel: ['TS0601'],  
        model: 'TS0601',
        vendor: 'Moes', 
        description: 'Zigbee + RF curtain switch',
        fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
        toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options, tz.moes_cover_calibration, tz.cover_position_tilt],
        exposes: [e.cover_position().setAccess('position', ea.STATE_SET)],
    };
    
module.exports = definition;

adamis24, 2022-03-23:
Than you 👍👍

adamis24, 2022-03-26:
Hello, my friend.

Unfortunately, it still doesn't work. Need to change something in the file?

Dobrý den, posíláme Vám Cenovou nabídku dle Vašich požadavků. V případě
dotazů či nejasností nás neváhejte kontaktovat.

Milan Suchý OKNACL
dodávka a montáž plastových oken
Jezvé 18
CZ-47108
mobil: +420 776508508

migor7, 2022-03-28:
Greetings. Here is my config, which I added as an external converter in the
settings of zigbee2mqtt. Opening, closing and stop works, no calibration. I
don't know how to add it

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
        fingerprint: [{modelID: 'TS0601', manufacturerName:
'_TZE200_nhyj64w2'}],
        zigbeeModel: ['TS0601'],
        model: 'TS0601',
        vendor: 'Moes',
        description: 'Zigbee + RF curtain switch',
        fromZigbee: [fz.tuya_cover, fz.ignore_basic_report],
        toZigbee: [tz.tuya_cover_control, tz.tuya_cover_options,
tz.moes_cover_calibration, tz.cover_position_tilt],
        exposes: [e.cover_position().setAccess('position', ea.STATE_SET)],
    };

module.exports = definition;

adamis24, 2022-04-02:
Hello, my friend.

Moes switch is now working, but I have one problem. When I pull the
cover up in the morning with a command in Node Red, after a few
minutes (2-3 minutes) the cover closes itself. And so on and on ...

migor7, 2022-04-04:
> Hello, my friend. Moes switch is now working, but I have one problem. When I pull the cover up in the morning with a command in Node Red, after a few minutes (2-3 minutes) the cover closes itself. And so on and on ... po 28. 3. 2022 v 9:23 odesílatel migor7 ***@***.***> napsal:
> […](#)

I don't know how to help. Probably a problem with Node Red

adamis24, 2022-04-04:
Hello, its problem with switch.
I ordered a new one - LoraTap, it has support in ZB2MQTT
https://www.aliexpress.com/item/4001190275372.html?spm=a2g0o.order_list.0.0.21ef1802jwSXyp