Koenkk/zigbee2mqtt #10550: [Problem]: IKEA ICPSHC24-30EU-IL-1 is marked as unsupported device

BiasF, 2022-01-03:
### What happened?

I tried multiple times to Pair the IKEA ICPSHC24-30EU-IL-1 with no success.
The interview is always failed and the device marked as unsupported. 

- multiple reset. 
- Tried to pair directly near to my Slaesh's CC2652RB stick
- Powered off IKEA bulbs near by

### What did you expect to happen?

Successful paired device because it should be a supported device 

### How to reproduce it (minimal and precise)

- Reset the LED Driver
- Try to pair the Driver

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20201026

### Adapter

Slaesh's CC2652RB 

### Debug log

No logs available (Only interview is failed)

airsailor, 2022-01-07:
Same problem with sonoff dongle 3.0 plus with the last firmware.
The ICPSHC24-30EU-IL-1 is detected but is not supported by zigbee2mqqt.
Any help?

Twit123, 2022-01-08:
What does the log have during the interview process, it should show IKEA as the manufacturer and the model number that it is picking up from the unit.

Just checking that Z2M is being given the correct model number from the unit as manufacturers sometimes make small changes.

Simon

airsailor, 2022-01-08:
> 

Hello Simon.
Thanks for your reply.
I've tried again and again and always the same problem.
My logs are in the attached files
If you have any ideas please let me know.
Thanks
[image]

airsailor, 2022-01-08:
> 
[image]

BiasF, 2022-01-09:
> [image]

Its the same message as for me :(

BiasF, 2022-02-09:
No idea?

airsailor, 2022-02-09:
The problem remains.

Koenkk, 2022-02-14:
The interview is failing, I suggest to first update your coordinator fw since you are running an outdated one (20201026)

goddenrich, 2022-02-25:
Im having the same issue
```
Starting interview of '0xb4e3f9fffe441ff1'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffe441ff1","ieee_address":"0xb4e3f9fffe441ff1","status":"started"},"type":"device_interview"}'
Error Failed to interview '0xb4e3f9fffe441ff1', device has not successfully been paired
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xb4e3f9fffe441ff1","ieee_address":"0xb4e3f9fffe441ff1","status":"failed"},"type":"device_interview"}'
```

closed 2022-04-05 by github-actions[bot]

danickel, 2022-10-08:
@goddenrich Are here any updates? I now bought the same IKEA device and I'm running into the same problem as you. Also using coordinator revision 20211217

goddenrich, 2022-10-08:
@danickel I didn't get it working and returned the device to IKEA

danickel, 2022-10-08:
ok thanks for the fast reply. I'm using also the TRADRI power supplies and there are working fine. 
@Koenkk: is there a chance to add support for this device?

Koenkk, 2022-10-09:
@danickel is is already supported for the `ICPSHC24-30EU-IL-1 `: https://www.zigbee2mqtt.io/devices/ICPSHC24-30EU-IL-1.html#ota-updates

yschen5812, 2023-07-03:
@danickel  @Koenkk  I guess for some reason there is a newer ICPSHC24-30EU-IL-1 (). I'm not sure if that would make sense to you but I bought my first one last year and was able to check/update the firmware via OTA (20211025 | 2.3.086). Then I bought another one recently, it shows up as ICPSHC24-30EU-IL-1 too however when checking new firmware it always show "No update available". The current firmware is 20181203 | 1.0.002

federicosantolini, 2023-11-26:
@yschen5812  I can confirm they changed something. The latest LED controllers are always unsupported and all the packaging seems relabeled. I got mine and despite it looking like the same old one, it cannot be paired like the others I bought last year. @Koenkk, is there anything I can send you to check if there is something we can do?
[image]
[image]
[image]
[image]
[image]