Koenkk/zigbee2mqtt #10552: [Problem]: Ikea STYRBAR Remote E2002 sends empty message on long press

stoinov, 2022-01-03:
### What happened?

I bought a new [Ikea STYRBAR remote control N2](https://www.zigbee2mqtt.io/devices/E2001_E2002.html)
The pairing was successful following the #10008 repair process.
I see empty messages  with only `linkquality` and `last_seen` when the left/right button is long pressed.

### What did you expect to happen?

receive messages that always have `action`.

### How to reproduce it (minimal and precise)

just long press the left/right buttons.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

zzh

### Debug log

```
Zigbee2MQTT:info  2022-01-03 23:02:58: MQTT publish: topic 'device/0xcc86ecfffe894140', payload '{"last_seen":1641250978123,"linkquality":87}'
Zigbee2MQTT:info  2022-01-03 23:02:58: MQTT publish: topic 'device/0xcc86ecfffe894140', payload '{"action":"on","last_seen":1641250978615,"linkquality":87}'
Zigbee2MQTT:info  2022-01-03 23:02:59: MQTT publish: topic 'device/0xcc86ecfffe894140', payload '{"last_seen":1641250979144,"linkquality":87}'
Zigbee2MQTT:info  2022-01-03 23:03:00: MQTT publish: topic 'device/0xcc86ecfffe894140', payload '{"action":"arrow_right_hold","last_seen":1641250980138,"linkquality":87}'
```

stoinov, 2022-02-04:
It is still present in version 1.23
```
Zigbee2MQTT:info  2022-02-04 10:33:57: MQTT publish: topic 'device/0xcc86ecfffe894140', payload '{"action":"arrow_left_click","last_seen":1643970837587,"linkquality":65}'
Zigbee2MQTT:info  2022-02-04 10:33:57: MQTT publish: topic 'device/0xcc86ecfffe894140', payload '{"last_seen":1643970837635,"linkquality":65}'
```

Koenkk, 2022-02-04:
Could you provide the debug log of this?

closed 2022-02-13 by github-actions[bot]