Koenkk/zigbee2mqtt #10555: [Problem]: automatic build on first start after update times out

klaernie, 2022-01-04:
### What happened?

starting after an update triggers the automatic rebuild with `npm run build` - but on lower-end hardware like a beaglebone black, this takes too long, so the start times out:

```
Jan 03 23:59:28 beaglebone systemd[1090]: Started zigbee2mqtt.
Jan 03 23:59:34 beaglebone bash[20971]: > zigbee2mqtt@1.22.2 start /opt/zigbee2mqtt
Jan 03 23:59:34 beaglebone bash[20971]: > node index.js
Jan 04 00:02:50 beaglebone bash[20971]: Building Zigbee2MQTT... (hash changed), failed
Jan 04 00:02:50 beaglebone bash[20971]: (node:20995) UnhandledPromiseRejectionWarning: Error: Command failed: npm run build
Jan 04 00:02:50 beaglebone bash[20971]: Aborted
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! code ELIFECYCLE
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! errno 134
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! zigbee2mqtt@1.22.2 build: `tsc && node index.js writehash`
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! Exit status 134
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR!
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! Failed at the zigbee2mqtt@1.22.2 build script.
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! A complete log of this run can be found in:
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR!     /home/debian/.npm/_logs/2022-01-03T23_02_50_210Z-debug.log
Jan 04 00:02:50 beaglebone bash[20971]: Aborted
Jan 04 00:02:50 beaglebone bash[20971]:     at ChildProcess.exithandler (child_process.js:294:12)
Jan 04 00:02:50 beaglebone bash[20971]:     at ChildProcess.emit (events.js:189:13)
Jan 04 00:02:50 beaglebone bash[20971]:     at maybeClose (internal/child_process.js:970:16)
Jan 04 00:02:50 beaglebone bash[20971]:     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
Jan 04 00:02:50 beaglebone bash[20971]: (node:20995) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
Jan 04 00:02:50 beaglebone bash[20971]: (node:20995) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Jan 04 00:02:50 beaglebone bash[20971]: Aborted
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! code ELIFECYCLE
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! errno 134
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! zigbee2mqtt@1.22.2 start: `node index.js`
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! Exit status 134
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR!
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! Failed at the zigbee2mqtt@1.22.2 start script.
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR! A complete log of this run can be found in:
Jan 04 00:02:50 beaglebone bash[20971]: npm ERR!     /home/debian/.npm/_logs/2022-01-03T23_02_50_510Z-debug.log
Jan 04 00:02:50 beaglebone systemd[1090]: zigbee2mqtt.service: Main process exited, code=killed, status=6/ABRT
Jan 04 00:02:50 beaglebone systemd[1090]: zigbee2mqtt.service: Failed with result 'signal'.
Jan 04 00:02:50 beaglebone systemd[1090]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Jan 04 00:02:50 beaglebone systemd[1090]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 1.
Jan 04 00:02:50 beaglebone systemd[1090]: Stopped zigbee2mqtt.
Jan 04 00:02:50 beaglebone systemd[1090]: Started zigbee2mqtt.
[further failed attempts]
```

Running `npm run build` manually works perfectly, but takes a bit longer:
```
debian@beaglebone:/opt/zigbee2mqtt$ time npm run build

> zigbee2mqtt@1.22.2 build /opt/zigbee2mqtt
> tsc && node index.js writehash

real    4m38.413s
user    4m22.256s
sys     0m5.541s
```

### What did you expect to happen?

zigbee2mqtt builds and starts

### How to reproduce it (minimal and precise)

use really low resources to run zigbee2mqtt, so that the `npm run build` takes longer than 3m16s.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

not relevant

### Adapter

not relevant

### Debug log

not relevant

Koenkk, 2022-01-04:
`134` means too low memory, not a timeout. So it seems so RAM needs to be freed to make the build finish succesfully.

klaernie, 2022-01-05:
So the act of stopping the service and running npm build outside of 'node index.js' frees just enough memory to make the build succeed?

klaernie, 2022-01-06:
Okay, gave it another try today with some more measurements, and it looks like about 160MB of RAM are used at most during the build:

```
debian@beaglebone:/opt/zigbee2mqtt$ /usr/bin/time -v npm run build

> zigbee2mqtt@1.22.2 build /opt/zigbee2mqtt
> tsc && node index.js writehash

        Command being timed: "npm run build"
        User time (seconds): 262.09
        System time (seconds): 5.79
        Percent of CPU this job got: 96%
        Elapsed (wall clock) time (h:mm:ss or m:ss): 4:36.86
        Average shared text size (kbytes): 0
        Average unshared data size (kbytes): 0
        Average stack size (kbytes): 0
        Average total size (kbytes): 0
        Maximum resident set size (kbytes): 158524
        Average resident set size (kbytes): 0
        Major (requiring I/O) page faults: 60
        Minor (reclaiming a frame) page faults: 98876
        Voluntary context switches: 7079
        Involuntary context switches: 52186
        Swaps: 0
        File system inputs: 40304
        File system outputs: 2264
        Socket messages sent: 0
        Socket messages received: 0
        Signals delivered: 0
        Page size (bytes): 4096
        Exit status: 0
debian@beaglebone:/opt/zigbee2mqtt$ 
```

I'd say it's fairly reasonable to assume a system has 200MB of RAM available - but if possible it'd be nice to have a more usable error message that points directly to the memory shortage.

Koenkk, 2022-01-06:
Added now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-06 by Koenkk

klaernie, 2022-01-07:
Thanks a lot, @Koenkk !