Koenkk/zigbee2mqtt #10557: [Problem]: _TZ3000_fisb3ajo discovered as TS0002

dkwireless, 2022-01-04:
### What happened?

Tuya 2CH neutral switch _TZ3000_fisb3ajo is discovered as TS0002

### What did you expect to happen?

Should be discovered as:
https://www.zigbee2mqtt.io/devices/QS-Zigbee-S04-2C-LN.html

### How to reproduce it (minimal and precise)

Discovered as such once paired

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 dongle

### Debug log

none

closed 2022-03-16 by github-actions[bot]

ArJay60, 2022-08-10:
Still the case. Can this also cause switch to after being paired getting error 205 after first few succesfull attempts? There are plenty of routers in the neigbourhood of the switch.

MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Tuinkamer Fan' failed: 'Error: Command 0xa4c13877af7ab682/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Data request failed with error: 'No network route' (205))'","meta":{"friendly_name":"Tuinkamer Fan"},"type":"zigbee_publish_error"}'

qweluke, 2022-11-23:
I'm having the same problem!