Koenkk/zigbee2mqtt #10559: [Problem]:  S26R2ZB goes offline after a period of time

paul-b-richardson, 2022-01-04:
### What happened?

After some time (24hr +) all my sockets appear offline and are unresponsive unless I turn the off/on physically at the socket.

### What did you expect to happen?

They remain online.

### How to reproduce it (minimal and precise)

Plug the socket in, wait ~24hr and try and access socket via zigbee2mqtt

### Zigbee2MQTT version

1.22.1 commit: 42569d6c

### Adapter firmware version

20211217

### Adapter

slaesh CC2652R stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7806859/log.txt)

daufinsyd, 2022-01-14:
It seems like we are facing the same issue : #10282

Hedda, 2022-02-06:
FYI, there is also more general discussions about non-specific performance issues with ITead Sonoff S26R2ZB plugs here -> https://github.com/Koenkk/zigbee2mqtt/discussions/10909

closed 2022-03-16 by github-actions[bot]