Koenkk/zigbee2mqtt #10562: [Problem]: UnhandledPromiseRejection

vk496, 2022-01-04:
### What happened?

When developing the Zlinky integration (https://github.com/Koenkk/zigbee-herdsman-converters/pull/3493), I receive a error message after some time (during this time, all is working properly)

```
UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "undefined".
Waiting for the debugger to disconnect...
```

I don't know how to get exactly line where that promise fails

### What did you expect to happen?

No crashing Z2M :)

### How to reproduce it (minimal and precise)

Using the PR with the device. Start and pair as usually. After some time (read of non reportable data I guess) it crashes.

### Zigbee2MQTT version

ALL DEV: 23bdc186 (Z2M), 8abf23a (herdsman), c1b2ff3b (herdsman-converters, the PR)

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Debug log

I did it twice, just for the case.

Log1:
```
zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 82 +5ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xdfcb EP:1 SeqNr. 66 request id: 107 +1ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 106 status: 2 +16ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +1ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +0ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +0ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 107 status: 0 +2ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +2ms
  zigbee-herdsman:deconz:driver query aps data confirm +2ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xdfcb EP:1 SeqNr. 67 request id: 108 +0ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 108 status: 0 +5ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 83 +1ms
  zigbee-herdsman:deconz:driver query aps data confirm +1ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xdfcb EP:1 SeqNr. 68 request id: 109 +0ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xdfcb request id: 98 confirm status: 0 +3ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +1ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 97 +27ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 109 status: 2 +1ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +1ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 84 +2ms
  zigbee-herdsman:deconz:driver query aps data confirm +1ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xdfcb EP:1 SeqNr. 69 request id: 110 +1ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xdfcb request id: 99 confirm status: 0 +5ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +1ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 98 +5ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 110 status: 2 +1ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +1ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 85 +1ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xdfcb EP:1 SeqNr. 70 request id: 111 +2ms
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 111 status: 2 +5ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +2ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xdfcb EP:1 SeqNr. 71 request id: 112 +3ms
[... 316 lines cut ...]
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 137 +4ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 185 +4ms
  zigbee-herdsman:deconz:driver query aps data confirm +2ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 186 +4ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xdfcb request id: 143 confirm status: 0 +12ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +3ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 142 +12ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 67 +0ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 187 +3ms
  zigbee-herdsman:deconz:driver query aps data confirm +1ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xdfcb request id: 144 confirm status: 0 +6ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +1ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 143 +6ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 68 +0ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 188 +2ms
  zigbee-herdsman:deconz:driver query aps data confirm +2ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xdfcb request id: 145 confirm status: 0 +5ms
  zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1 +1ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 144 +5ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 69 +0ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 189 +3ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint ERROR: [object Object] +4ms
UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "undefined".
Waiting for the debugger to disconnect...
```

vk496, 2022-01-04:
I think the problem is trying to do a lot of reads. There must be some queue in the Controller that have a limit.

If I read one by one with more time lapse, all works correctly

closed 2022-01-04 by vk496

mwthrane, 2022-02-05:
I have the same issue:
`UnhandledPromiseRejection: This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). The promise rejected with the reason "undefined".`