Koenkk/zigbee2mqtt #10563: [Problem]: Lights on command from HA not working

Dutchy-79, 2022-01-04:
### What happened?

I changed from cc2531 and ZHA to Sonoff 3.0 plus dongle with Zigbee2mqtt last week. (Running latest version of integration and HA version)
I changed the device names for my bar lights to the new names in automation and next day the automation was trigged on time.
1 out of 3 lights turned on, I checked and in HA dashboard the not working light showed ON state. Toggled the switch >off >on and it turned on.

Next morning I checked the automation and selected all 3 lights from device list again.

That evening none turned on but like before it did show on in HA dash.

just now I manually triggered the automation and 1 light did not turn on.

### What did you expect to happen?

I expected it to work just like it was before when running ZHA and a cc2531

### How to reproduce it (minimal and precise)

It seems random, First light 2 and 3 did not turn on, Then none and now light 1 did not turn on

### Zigbee2MQTT version

1.22.2 commit: unknown

### Adapter firmware version

20211217

### Adapter

Sonoff 3.0 Plus Dongle

### Debug log

Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Sonoff Zigbee Basic', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"0/0x00124B0019389E09","inactiveText":"COORD","minimumOffTime":0,"presentValue":122,"relinquishDefault":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"64037/0x00124B0014D96965","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"49419/0x00124B00210B7145","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"65046/0x00124B00210B788C","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"16116/0x00124B00210B6D29","inactiveText":"NEIGHBOR","presentValue":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"35079/0x00124B0019389E09","inactiveText":"PARENT","minimumOffTime":0,"presentValue":0,"relinquishDefault":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"16438/0xBC33ACFFFE281573","inactiveText":"CHILD_RFD","minimumOffTime":1,"presentValue":34,"relinquishDefault":2}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"49419/0x00124B00210B7145","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"65046/0x00124B00210B788C","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"16116/0x00124B00210B6D29","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received MQTT message on 'zigbee2mqtt/Lamp boven trap/set' with data '{"state": "ON", "brightness": 89}'
Publishing 'set' 'brightness' to 'Lamp boven trap'
Received MQTT message on 'zigbee2mqtt/Lamp boven bar 1/set' with data '{"state": "ON", "brightness": 89}'
Publishing 'set' 'brightness' to 'Lamp boven bar 1'
Received MQTT message on 'zigbee2mqtt/Lamp boven bar 2/set' with data '{"state": "ON", "brightness": 89}'
Publishing 'set' 'brightness' to 'Lamp boven bar 2'
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"0/0x00124B0019389E09","inactiveText":"COORD","minimumOffTime":0,"presentValue":122,"relinquishDefault":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"64037/0x00124B0014D96965","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"49419/0x00124B00210B7145","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"65046/0x00124B00210B788C","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Saving state to file /config/zigbee2mqtt_new/state.json
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"35079/0x00124B0019389E09","inactiveText":"PARENT","minimumOffTime":0,"presentValue":0,"relinquishDefault":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"16438/0xBC33ACFFFE281573","inactiveText":"CHILD_RFD","minimumOffTime":1,"presentValue":34,"relinquishDefault":2}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"0/0x00124B0019389E09","inactiveText":"COORD","presentValue":55}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"49419/0x00124B00210B7145","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0

Dutchy-79, 2022-01-04:
After turning on debug logging i ran automation again and no spots turned on, See log that says they did.

Dutchy-79, 2022-01-04:
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"0/0x00124B0019389E09","inactiveText":"COORD","minimumOffTime":0,"presentValue":122,"relinquishDefault":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"64037/0x00124B0014D96965","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"49419/0x00124B00210B7145","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"65046/0x00124B00210B788C","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"35079/0x00124B0019389E09","inactiveText":"PARENT","minimumOffTime":0,"presentValue":0,"relinquishDefault":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"16438/0xBC33ACFFFE281573","inactiveText":"CHILD_RFD","minimumOffTime":1,"presentValue":34,"relinquishDefault":2}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"49419/0x00124B00210B7145","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"65046/0x00124B00210B788C","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"0/0x00124B0019389E09","inactiveText":"COORD","minimumOffTime":0,"presentValue":122,"relinquishDefault":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"64037/0x00124B0014D96965","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"49419/0x00124B00210B7145","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"65046/0x00124B00210B788C","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Saving state to file /config/zigbee2mqtt_new/state.json
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"35079/0x00124B0019389E09","inactiveText":"PARENT","minimumOffTime":0,"presentValue":0,"relinquishDefault":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"16438/0xBC33ACFFFE281573","inactiveText":"CHILD_RFD","minimumOffTime":1,"presentValue":34,"relinquishDefault":2}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"0/0x00124B0019389E09","inactiveText":"COORD","presentValue":55}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"49419/0x00124B00210B7145","inactiveText":"NEIGHBOR","presentValue":1}' from endpoint 8 with groupID 0
Received MQTT message on 'zigbee2mqtt/Lamp boven trap/set' with data '{"state": "ON", "brightness": 89}'
Publishing 'set' 'brightness' to 'Lamp boven trap'
Received MQTT message on 'zigbee2mqtt/Lamp boven bar 1/set' with data '{"state": "ON", "brightness": 89}'
Publishing 'set' 'brightness' to 'Lamp boven bar 1'
Received MQTT message on 'zigbee2mqtt/Lamp boven bar 2/set' with data '{"state": "ON", "brightness": 89}'
Publishing 'set' 'brightness' to 'Lamp boven bar 2'
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router zonder antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"0/0x00124B0019389E09","inactiveText":"COORD","minimumOffTime":0,"presentValue":122,"relinquishDefault":0}' from endpoint 8 with groupID 0
[... cut ...] 5) at listOnTimeout (internal/timers.js:557:17) at processTimers (internal/timers.js:500:7)
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"35079/0x00124B0019389E09","inactiveText":"PARENT","minimumOffTime":0,"presentValue":0,"relinquishDefault":0}' from endpoint 8 with groupID 0
Received Zigbee message from 'Router met antenne', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0

Dutchy-79, 2022-01-04:
[image]
[image]

Dutchy-79, 2022-01-04:
2 status ON and one OFF, All 3 are OFF

Twit123, 2022-01-07:
Are you using Z2M groups and then running an automation against the group OR are you calling 3 separate on/off events?

As I can't group my blinds I have to fire 4 separate actions to lower/open them and I have to put a delay between the commands other wise at least one wont work. I use 3 seconds delay but you could probably go lower.

I have my lights in groups in Z2M and flash the lights (6 rooms) at dinner time and they all work together every time.

Simon.

Dutchy-79, 2022-01-08:
Calling 3 separate lights, I'll try creating a group instead.

Dutchy-79, 2022-01-14:
> Are you using Z2M groups and then running an automation against the group OR are you calling 3 separate on/off events?
>
> As I can't group my blinds I have to fire 4 separate actions to lower/open [...]

Still the same in a group, Sometimes one or two turn on and one or two say they are on in z2m and HA but ar not on.

Deleted the lights and now paired again.

If this does not work then I will go back to ZHA because before switching to Z2M I didn't have these problems.

Dutchy-79, 2022-01-16:
For now it seems like the lights all turn on, I think the problem was that I also included a brightness and since I removed that from the automation everything works.

closed 2022-01-16 by Dutchy-79