Koenkk/zigbee2mqtt #10569: [New device support]: OSRAM ZLO-CeilingTW-OS (old Surface Light)

smoki3, 2022-01-04:
### Link

https://www.osram.com/cb/lightify/lightify-home/lightify-home-faq/lightify_home_faqs.jsp

### Database entry

"id":3,"type":"Router","ieeeAddr":"0x00158d0001998aff","nwkAddr":32771,"manufId":4151,"manufName":"OSR\u0000Z","powerSource":"Mains (single phase)","modelId":"ZLO-CeilingTW-OS","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,4,3,6,8,5,768,65535,65535,4096],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"ZLO-CeilingTW-OS","manufacturerName":"OSR\u0000Z","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"\u0000\u0000\u0000\u0000","swBuildId":"V1.04.27\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"\u0000\u0000\u0000\u0000","swBuildId":"V1.04.27\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1640542523447,"useImplicitCheckin":true}

### Comments

Looks like my Lamp is not supported until now. The ZLO-CeilingTW-OS is a version of the Surface light TW. 

The Lamp is dimmable and has tunable white ( color temperature)

According to the OSRAM website it is using Z3.0 compared to the standard Surfce TW. 
[image]

### Supported color modes

color temperature

### Color temperature range

{"colorTempPhysicalMin":153} {"colorTempPhysicalMax":370}

smoki3, 2022-01-04:
https://github.com/ioBroker/ioBroker.zigbee/issues/1311

smoki3, 2022-01-04:
got it working with 

```   
 {
        zigbeeModel: ['ZLO-CeilingTW-OS'],
        model: 'AB401130055',
        vendor: 'OSRAM',
        description: 'OSRAM Surface TW Light',
        extend: extend.ledvance.light_onoff_brightness_colortemp(),
        ota: ota.ledvance,
    },
```

Koenkk, 2022-01-04:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-04 by Koenkk