Koenkk/zigbee2mqtt #10572: [Problem]: Can't trigger effect on IKEA LED1924G9

red-black-bonsai, 2022-01-04:
### What happened?

Tried to trigger an effect on my new IKEA LED1924G9 by sending an MQTT message to topic 
`zigbee2mqtt/Tradfri_Colour_LED_Bulb/set`:

    {"color": "#0000ff", "effect": "blink"}

Colour of the bulb changed to blue but bulb doesn't blink in any perceptible way.

### What did you expect to happen?

I expected the bulb to blink after sending that command.

### How to reproduce it (minimal and precise)

Pair IKEA LED1924G9, send command via MQTT as described above.

### Zigbee2MQTT version

1.22.2-dev commit: 4f887a1a

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Debug log

debug 2022-01-04 21:35:57: Received MQTT message on 'zigbee2mqtt/Tradfri_Colour_LED_Bulb/set' with data '{"color_temp":454}'
debug 2022-01-04 21:35:57: Skipping state because of Home Assistant
debug 2022-01-04 21:35:57: Publishing 'set' 'color_temp' to 'Tradfri_Colour_LED_Bulb'
info  2022-01-04 21:35:57: MQTT publish: topic 'zigbee2mqtt/Tradfri_Colour_LED_Bulb', payload '{"brightness":254,"color":{"h":27,"hue":27,"s":92,"saturation":92,"x":0.5056,"y":0.4152},"color_mode":"color_temp","color_temp":454,"linkquality":163,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}'
debug 2022-01-04 21:38:24: Received MQTT message on 'zigbee2mqtt/Tradfri_Colour_LED_Bulb/set' with data '{"color": "#0000ff", "effect": "blink"}'
debug 2022-01-04 21:38:24: Skipping state because of Home Assistant
debug 2022-01-04 21:38:24: Publishing 'set' 'color' to 'Tradfri_Colour_LED_Bulb'
debug 2022-01-04 21:38:24: Requested color_temp 574 is higher than maximum supported 454, using maximum!
debug 2022-01-04 21:38:24: Publishing 'set' 'effect' to 'Tradfri_Colour_LED_Bulb'
info  2022-01-04 21:38:25: MQTT publish: topic 'zigbee2mqtt/Tradfri_Colour_LED_Bulb', payload '{"brightness":254,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.1355,"y":0.0399},"color_mode":"xy","color_temp":454,"linkquality":160,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":null}'

sjorge, 2022-01-05:
I don't think you can mix effects with other properties in the payload. I don't have your exact bulb though but just using `effect` in the payload and nothing else seems to work fine for me on other tradfi bulbs and panels.

red-black-bonsai, 2022-01-05:
I just included the colour change to verify that the device gets the command and reacts. I had tried just the effect before and it didn't work either. Just verified:

Sending to `zigbee2mqtt/Tradfri_Colour_LED_Bulb/set`

    {"effect": "blink"}

does nothing.

The log mentions publishing set effect to the bulb. With log level debug for the herdsman it logs

    Zigbee2MQTT:debug 2022-01-04 22:04:38: Publishing 'set' 'effect' to 'Tradfri_Colour_LED_Bulb'
      zigbee-herdsman:controller:endpoint Command 0x2c1165fffed4eaaa/1 genIdentify.triggerEffect({"effectid":0,"effectvariant":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +17s

sjorge, 2022-01-06:
I wonder if it's an issue specific to the LED1924G9, I sadly don't own that exact one.

red-black-bonsai, 2022-01-07:
I managed to try out a LED1836G9 with my setup - with it the effects work. I'll have a look at the herdsman logs again...

red-black-bonsai, 2022-01-07:
It appears that for the LED1924G9 there is a `Data confirm error` after thta effect command is sent (which doesn't happen for the LED1836G9). Dunno if that tells you anything...

    zigbee-herdsman:controller:endpoint Command 0x8cf681fffe4de422/1 genIdentify.triggerEffect({"effectid":0,"effectvariant":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +2m
    zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x8cf681fffe4de422:60537/1 (0,0,1) +2m
    zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":60537,"destendpoint":1,"srcendpoint":1,"clusterid":3,"transid":7,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[1,7,64,0,0]}} +2m
    zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,121,236,1,1,3,0,7,0,30,5,1,7,64,0,0,230] +2m
    zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +2m
    zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms
    zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +1ms
    zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +2m
    zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
    zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,233,1,7,40] +83ms
    zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,233,1,7,40] +1ms
    zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [233,1,7] - 40 +1ms
    zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":233,"endpoint":1,"transid":7} +2m
    zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
    zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x8cf681fffe4de422:60537,233,0) +109ms
    zigbee-herdsman:adapter:zStack:adapter Wait 2000ms +2ms
    zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x8cf681fffe4de422:60537/1 (0,1,1) +2s
    zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":60537,"destendpoint":1,"srcendpoint":1,"clusterid":3,"transid":8,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[1,7,64,0,0]}} +2s
    zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,121,236,1,1,3,0,8,0,30,5,1,7,64,0,0,233] +2s
    zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +2s
    zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +1ms
    zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +1ms
    zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +2s
    zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
    zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,8,206] +54ms
    zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,8,206] +1ms
    zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,8] - 206 +1ms
    zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":8} +2s
    zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
    zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,3,0,121,236,1,1,0,185,0,231,252,235,0,0,5,8,7,11,64,0,121,236,29,202] +2s
    zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,3,0,121,236,1,1,0,185,0,231,252,235,0,0,5,8,7,11,64,0,121,236,29,202] +1ms
    zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,3,0,121,236,1,1,0,185,0,231,252,235,0,0,5,8,7,11,64,0,121,236,29] - 202 +1ms
    zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":3,"srcaddr":60537,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":185,"securityuse":0,"timestamp":15465703,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,7,11,64,0]}} +2s
    zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":7,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":64,"statusCode":0},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":60537,"endpoint":1,"linkquality":185,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}' +2m
    zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms

tomasliumparas, 2022-02-10:
I own two LED1924G9s, both having the same issue

ikilledmypc, 2022-02-28:
Same for me unfortunately I wonder if this bulb will ever get a firmware upgrade. looking at the current firmware it's already quite old: 

  | 2019062886LEL08N | 1.0.021

tashda, 2022-03-13:
Same goes for all 3 of my bulbs.

T-IOOO, 2022-03-16:
> Same for me unfortunately I wonder if this bulb will ever get a firmware upgrade. looking at the current firmware it's already quite old:
> 
>   | 2019062886LEL08N | 1.0.021

Ikea's changelog was updated yesterday (March 15). Now when I check for new firmware in the frontend I get the message "newer firmare on <IEEE> is newer than latest online".

closed 2022-04-24 by github-actions[bot]

arsenicks, 2022-10-24:
Interested in the thread.. I'm having the same issue, anybody was able to get the effects working ?

fliespl, 2023-01-29:
@arsenicks nope - still no go :/

AsherVo, 2023-02-02:
Just bought these and have the same issue. I noticed they also didn't blink when I performed the factory reset, so it seems like it's an issue with this specific model of bulb.

AsherVo, 2023-02-04:
Just posted on r/tradfri, which appears to be the only place to report bugs where IKEA might read about them. Feel free to upvote: https://www.reddit.com/r/tradfri/comments/10rzjuf/e12_rbg_led_bulbs_broken_effects/

MrZwickie, 2024-02-10:
Same thing here with Tradfri LED1923R5. Is there still no solution to this issue?

marazmarci, 2024-03-15:
FYI there's a new version of this bulb: [LED2103G5](https://www.ikea.com/gb/en/p/tradfri-led-bulb-e27-806-lumen-smart-wireless-dimmable-warm-white-globe-10541502/). It is already [supported](https://github.com/Koenkk/zigbee2mqtt/issues/21755) one the dev branch 4 days ago.

It has all the same parameters on paper, except for the new one being more energy efficient. It has an EU energy rating of E instead of an F.

Another user have [confirmed](https://github.com/Koenkk/zigbee2mqtt/issues/21755#issuecomment-1999678886) that effects are working on this new bulb!