Koenkk/zigbee2mqtt #10580: [Problem]:

drubens, 2022-01-04:
### What happened?

Hello dears,
happy new year!

I'm new on Zigbee2Mqtt so please be patience with me.

I've configured everything and everything seems working correctly on pairing side and HA communication but I'm having some issues with BTICINO stuff.

I've the living now L441C/N4411C/NT4411C dimmer and brightness seems not working. I can switch on and off the light but it is not possible at all dimmer the light.

Moreover all Bticino connected tools got a led that can be configured but I don't find anything about that.

Thanks in advance for any your help.
have a nice rest of 2022.
R

### What did you expect to happen?

Bticino L441C/N4411C/NT4411C should be dimmerable.
Moreover the Light of the led for Bticino connected series tools should be configurable. 

### How to reproduce it (minimal and precise)

Pair the L441C/N4411C/NT4411C with zigbee2Mqtt then try to dimmer. Log show the right message but nothing happens on the light.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7811042/log.txt)

Yonn-Trimoreau, 2022-01-10:
You should:
- Reset & associate your device back to the official BTicino gateway
- Wait for 24 hours for your device's firmware to be upgraded
- Reset & associate your device to your own Zigbee gateway
- Upgrade zigbee2mqtt to the latest version
- Enable dimmer & define values for ballast_minimum_level and ballast_maximum_level (1 and 254 usually)
- Try again setting the brightness

drubens, 2022-01-16:
Thank you @Yonn-Trimoreau  sorry for my late answer.
I will try to do that but I'm not sure if I got how can I do that (Enable dimmer & define values for ballast_minimum_level and ballast_maximum_level (1 and 254 usually). Maybe this is my issue, I've not activated something.

let's keep updated.
Grazie grazie
R

drubens, 2022-01-28:
@Yonn-Trimoreau unfortunately I followed what you suggested but no way to have Brightness.
I've also updated the slaesh usb coordinator.
the firmware of the dimmer looks the "20" on home+control app and after 1 week nothing updated and fw is still "20".

Any suggestion or different procedure that should I've to follow please? Any other user is experiencing same issue with dimmer?

thanks again yonn and everybody.
ciao

drubens, 2022-02-02:
I see full setup now.
Fixed.
Thanks for all your work.

[image]

closed 2022-02-02 by drubens