Koenkk/zigbee2mqtt #10581: [Problem]: Slow devices response after upgrade to version 1.22.2-1

HelioFilipe, 2022-01-05:
### What happened?

I was running homeassistant with Zigbee2mqtt version 1.22.1-1 and it was OK. After upgrade to 1.22.2-1 the devices start to respond about 10 seconds later.
If I restore to previous version everything is OK again.

### What did you expect to happen?

the devices get back to respond fast, like in the previous version

### How to reproduce it (minimal and precise)

just switch one device ON / OFF and it will happen

### Zigbee2MQTT version

1.22.2-1

### Adapter firmware version

20201127

### Adapter

CC2531

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7811370/log.txt)

HelioFilipe, 2022-01-06:
it takes 1min 04 secs to respond

Zigbee2MQTT:debug **2022-01-06 00:50:00**: Received MQTT message on 'zigbee2mqtt/Lampada Sala/set' with data '{"state": "OFF"}'
Zigbee2MQTT:debug **2022-01-06 00:50:00**: Publishing 'set' 'state' to 'Lampada Sala'
Zigbee2MQTT:info  **2022-01-06 00:51:04**: MQTT publish: topic 'zigbee2mqtt/Lampada Sala', payload '{"brightness":10,"color":{"x":0.3348,"y":0.343},"color_mode":"color_temp","color_temp":185,"linkquality":31,"state":"OFF"}'

Jpsy, 2022-02-07:
This is likely not cause by Zigbee2MQTT but by nested light groups. Please check if what you are switching are groups that contain other groups. A current workaround is to flatten out any groups so that they are only one level deep. This is probably not restricted to light groups but can be a problem with all kinds of groups.
See https://github.com/home-assistant/core/issues/65119#issuecomment-1030904603

fraintt, 2022-02-07:
Hi, 
I do not think so, I just now changed to the Sonoff Zigbee Stick, and there, when updateing to the mentioned version the problem does not occur. I also have this issue when switching a light on which is not a part of any Group.

Jpsy, 2022-02-07:
Okay. Sorry. The 10 second delay is so typical for the group issue that I was actually quite sure that this would be the same problem.

HelioFilipe, 2022-02-07:
The strange thing is that if I rollback to the previous version Zigbee2mqtt 1.22.1-1 this delay doesn´t happend.
(I am not using light groups or any other devices groups)

oywino, 2022-02-08:
This is a groups-within-groups issue.
It began with 2022.2.1
After upgrading to 2022.2.3 today it is still not solved.

fraintt, 2022-02-08:
> This is a groups-within-groups issue. It began with 2022.2.1 After upgrading to 2022.2.3 today it is still not solved.

I had this issue in 2021.12.x as well. It appeared when updateing from zigbee2mqtt 1.22.1-1   to ->   1.22.2-1.

closed 2022-03-19 by github-actions[bot]