Koenkk/zigbee2mqtt #10582: [Problem]: Moes BRT-100-TRV OTA updates

Ivano62, 2022-01-05:
### What happened?

I've a 'TS0601 (_TZE200_b6wax7g0)'.
If I run `zigbee2mqtt/bridge/request/device/ota_update/check` for this device I get `Error Failed to check if update available for 'thermoCucina' (No image available for imageType '5634')`, but the state of device is
`    "update": {
        "state": "available"
    },
    "update_available": true,`

[image]

I think this is an incongruence and I don't know what can happen if the update is started.

### What did you expect to happen?

If the firmware is not available, the possibility of updating is not activated, the status is consistent with the situation detected 

### How to reproduce it (minimal and precise)

Run `zigbee2mqtt/bridge/request/device/ota_update/check` for Moes BRT-100-TRV  -  TS0601 (_TZE200_b6wax7g0)

### Zigbee2MQTT version

1.22.2-dev commit: 4f887a1

### Adapter firmware version

0x26720700

### Adapter

RaspBee II

### Debug log

error 2022-01-05 09:31:53: Failed to check if update available for 'thermoCucina' (No image available for imageType '5634')

Ivano62, 2022-02-05:
The problem persiste.
Can't find a solution?
Thanks

DamGranger, 2022-03-05:
The error message still happens on 1.24.0

closed 2022-04-12 by github-actions[bot]