Koenkk/zigbee2mqtt #10586: Xiaomi H1 with neutral WS-EUK04 not updating status for right switch [Problem]:

daserra23, 2022-01-05:
### What happened?

This 2 way switch won't update it's status in the frontend unless I click "refresh" in the "exposes" tab. It worked with the previous version. The left button works fine.

Re-adding the device doesn't change anything, not does enabling or disabling "control relay/decoupled".

### What did you expect to happen?

I expected that the frontend would show the updated status after pressing the right button but it doesn't and therefore actions supposedly triggered by the right button are not triggered.

### How to reproduce it (minimal and precise)

Press the right button, observe the non-reaction of the device in the frontend, then click the "state_right" refresh icon and watch the status update.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210120

### Adapter

Slaesh CC2652RB

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7814984/log.txt)

daserra23, 2022-01-05:
After some testing I have fixed this issue by configuring the reporting manually. Not too much spamming with these settings.
[image]

closed 2022-01-05 by daserra23

ilvi80, 2022-01-27:
Не работает отвязка клавиш. Не переключает в логический режим