Koenkk/zigbee2mqtt #10590: Thermostat TS0601 (and similars) wrong "deadzone_temperature" expose

shymchenko, 2022-01-05:
### What happened?

The "deadzone_temperature" parameter should be set in increments of 1 degree, but set in increments of 0.1, as a result of which an incorrect huge value is sended to the thermostat!
If I use a custom converter with parameter e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1) , all is working correct.  

### What did you expect to happen?

Correct "deadzone_temperature" parameter setting

### How to reproduce it (minimal and precise)

Set "deadzone_temperature" in Zigbee2Mqtt interface

### Zigbee2MQTT version

1.22.2-dev commit: 4f887a1

### Adapter firmware version

20201010

### Adapter

zStack30x

### Debug log

[log (1).txt](https://github.com/Koenkk/zigbee2mqtt/files/7815341/log.1.txt)

shymchenko, 2022-01-05:
Probable the problem is  here:
exposes.js 
deadzone_temperature: () => new Numeric('deadzone_temperature', access.STATE_SET).withUnit('°C').withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat').**withValueMin(0.1)**.withValueMax(5),

Koenkk, 2022-01-05:
Please make a pull request with the fix.

shymchenko, 2022-01-05:
Sorry please, but I am not a programmer and not understand how make a pull request.
In the previous version, the parameter " withValueMin" was 1. In the new version, just a typo.
deadzone_temperature: () => new Numeric('deadzone_temperature', access.STATE_SET).withUnit('°C').withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat').**withValueMin(1)**.withValueMax(5),

Koenkk, 2022-01-06:
Added the fix as you suggested: 

```
e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1)
```

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-06 by Koenkk