Koenkk/zigbee2mqtt #10591: [Problem]: NPM CI Dont work

GlimpseTaha, 2022-01-05:
### What happened?

Hello, its my first time that i use Github.
I have a problem with zigbee2mqtt. I use a raspberry pi 4 2gb.

# Install dependencies (as user "pi")
cd /opt/zigbee2mqtt
npm ci
And after npm ci i get an error:
npm ERR! code EACCES
npm ERR! syscall mkdir
npm ERR! path /opt/zigbee2mqtt/node_modules
npm ERR! errno -13
npm ERR! Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/node_modules'
npm ERR!  [Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/node_module                                                                                                                                                             s'] {
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'mkdir',
npm ERR!   path: '/opt/zigbee2mqtt/node_modules'
npm ERR! }
npm ERR!
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the                                                                                                                                                              current user
npm ERR!
npm ERR! If you believe this might be a permissions issue, please double-check t                                                                                                                                                             he
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-01-05T15_27_38_268Z-debug.log
Can anyone help me. 

### What did you expect to happen?

It should run my command?

### How to reproduce it (minimal and precise)

I dont know

### Zigbee2MQTT version

I dont know

### Adapter firmware version

I dont know 

### Adapter

CC2531 

### Debug log

npm ERR! code EACCES
npm ERR! syscall mkdir
npm ERR! path /opt/zigbee2mqtt/node_modules
npm ERR! errno -13
npm ERR! Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/node_modules'
npm ERR!  [Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/node_module                                                                                                                                                             s'] {
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'mkdir',
npm ERR!   path: '/opt/zigbee2mqtt/node_modules'
npm ERR! }
npm ERR!
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the                                                                                                                                                              current user
npm ERR!
npm ERR! If you believe this might be a permissions issue, please double-check t                                                                                                                                                             he
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-01-05T15_27_38_268Z-debug.log

lux73, 2022-01-05:
the error log is clear:

```
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
```

it seems that User "pi" has not the rights execute npm ci in your zigbee2mqtt Directory!

```
chown -R pi /opt/zigbee2mqtt
chgrp -R pi /opt/zigbee2mqtt
```

should fix your Problem

GlimpseTaha, 2022-01-05:
Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/branches' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs/remotes/origin/HEAD' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs/remotes/origin' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs/remotes' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs/heads/master' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs/heads' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/HEAD' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/packed-refs' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/package.json' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.eslintrc.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/npm-shrinkwrap.json' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/CODE_OF_CONDUCT.md' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/data/configuration.yaml' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/data' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images/logo_large.png' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images/logo.png' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images/architecture.png' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images/logo_bee_only.png' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images/logo_vector.svg' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/cli.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.npmrc' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/lib/flushPromises.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/lib' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/assets/mock-external-converter.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/assets/exampleExtension.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/assets/mock-external-converter-multiple.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/assets' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/receive.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/otaUpdate.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/publish.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/settings.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/externalExtension.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/controller.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/onEvent.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/legacy/report.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/legacy/bridgeLegacy.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/legacy' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/logger.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/networkMap.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/externalConverters.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/utils.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/frontend.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/stub/mqtt.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/stub/data.js' wird geändert: Die Operation ist nicht erlaubt
[... cut ...]
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'mkdir',
npm ERR!   path: '/opt/zigbee2mqtt/node_modules'
npm ERR! }
npm ERR! 
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR! 
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator.

GlimpseTaha, 2022-01-05:
> the error log is clear:
>
>
>
> ```
>
> npm ERR! The operation was rejected by your operating system.
>
> npm ERR! It is likely you do not have the permissions to access this file as the current user
>
> ``` [...]

I tried it know and thats what i got: 
Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/branches' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs/remotes/origin/HEAD' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs/remotes/origin' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs/remotes' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs/heads/master' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs/heads' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/refs' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs/HEAD' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/logs' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git/packed-refs' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.git' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/package.json' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.eslintrc.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/npm-shrinkwrap.json' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/CODE_OF_CONDUCT.md' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/data/configuration.yaml' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/data' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images/logo_large.png' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images/logo.png' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images/architecture.png' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images/logo_bee_only.png' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images/logo_vector.svg' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/images' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/cli.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/.npmrc' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/lib/flushPromises.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/lib' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/assets/mock-external-converter.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/assets/exampleExtension.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/assets/mock-external-converter-multiple.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/assets' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/receive.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/otaUpdate.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/publish.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/settings.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/externalExtension.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/controller.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/onEvent.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/legacy/report.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/legacy/bridgeLegacy.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/legacy' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/logger.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/networkMap.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/externalConverters.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/utils.test.js' wird geändert: Die Operation ist nicht erlaubt
chown: der Eigentümer von '/opt/zigbee2mqtt/test/frontend.test.js' wird geändert: Die Operation ist nicht erlaubt
[... cut ...]
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'mkdir',
npm ERR!   path: '/opt/zigbee2mqtt/node_modules'
npm ERR! }
npm ERR! 
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR! 
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator.

lux73, 2022-01-05:
login as root & execute the commands i posted

closed 2022-02-12 by github-actions[bot]