Koenkk/zigbee2mqtt #10596: [New device support]: Aqara Smart Natural Gas Detector (lumi.sensor_gas.acn02 / JT-BZ-01AQ/A)

Wh1terat, 2022-01-05:
### Link

https://www.aqara.com/cn/productDetail/Aqara%20Smart%20Natural%20Gas%20Detector

### Database entry

{"id":22,"type":"Router","ieeeAddr":"0x54ef4410003228eb","nwkAddr":52159,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.sensor_gas.acn02","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,64704,4],"outClusterList":[10,25,4],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.sensor_gas.acn02","appVersion":13,"manufacturerName":"LUMI","powerSource":1,"zclVersion":3,"stackVersion":2,"hwVersion":1,"dateCode":"08-05-2021"}},"ssIasZone":{"attributes":{"iasCieAddr":"0xbc33acfffe81beef","zoneState":1}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[5,33,1,0,8,33,13,1,9,33,0,10,10,33,0,0,11,32,0,12,32,1,154,32,0,159,32,2,160,33,0,0,161,32,0,162,32,0,163,32,0,164,32,0,165,32,0,166,32,0]},"314":0,"315":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":13,"stackVersion":2,"hwVersion":1,"dateCode":"08-05-2021","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1641402395207,"useImplicitCheckin":true}

### Comments

Tried copying JTQJ-BF-01LM/BW - seemingly _nothing_ like it as everything reports null.

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const constants = require('zigbee-herdsman-converters/lib/constants');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const definition = {
    zigbeeModel: ['lumi.sensor_gas.acn02'],
    model: 'JT-BZ-01AQ/A',
    vendor: 'Xiaomi',
    description: 'Aqara Smart Natural Gas Detector',
    fromZigbee: [fz.ias_gas_alarm_1, fz.JTQJBF01LMBW_sensitivity, fz.JTQJBF01LMBW_gas_density],
    toZigbee: [tz.JTQJBF01LMBW_JTYJGD01LMBW_sensitivity, tz.JTQJBF01LMBW_JTYJGD01LMBW_selfest],
    exposes: [
        e.gas(), e.battery_low(), e.tamper(), exposes.enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high']),
        exposes.numeric('gas_density', ea.STATE), exposes.enum('selftest', ea.SET, ['']),
    ],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

closed 2022-01-16 by Koenkk

Otnow, 2022-01-16:
@Wh1terat, [the converter](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3685) I proposed has been merged into the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html), so if possible, please test it (_important: the firmware of the coordinator must be the latest version_).

Wh1terat, 2022-01-16:
Thanks @Otnow , I will test and let you know!