Koenkk/zigbee2mqtt #10599: [Problem]: npm start error

GlimpseTaha, 2022-01-05:
### What happened?

After i wrote npm start 

root@raspberrypi:/opt/zigbee2mqtt# npm start

> zigbee2mqtt@1.22.2 start
> node index.js
root@raspberrypi:/opt/zigbee2mqtt# npm start 

> zigbee2mqtt@1.22.2 start
> node index.js

Zigbee2MQTT:info  2022-01-05 18:59:21: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-01-05.18-59-19' filename: log.txt
Zigbee2MQTT:info  2022-01-05 18:59:21: Starting Zigbee2MQTT version 1.22.2 (commit #301ba14b)
Zigbee2MQTT:info  2022-01-05 18:59:21: Starting zigbee-herdsman (0.13.188)
CREATED DECONZ ADAPTER
Zigbee2MQTT:info  2022-01-05 18:59:22: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-01-05 18:59:22: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:info  2022-01-05 18:59:22: Currently 0 devices are joined:
Zigbee2MQTT:warn  2022-01-05 18:59:22: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-01-05 18:59:22: Allowing new devices to join.
Zigbee2MQTT:warn  2022-01-05 18:59:22: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2022-01-05 18:59:22: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-01-05 18:59:22: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:error 2022-01-05 18:59:22: MQTT failed to connect: connect ECONNREFUSED 127.0.0.1:1883
Zigbee2MQTT:error 2022-01-05 18:59:22: Exiting...
Zigbee2MQTT:info  2022-01-05 18:59:23: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-01-05 18:59:23: Stopped zigbee-herdsman
npm ERR! code 1
npm ERR! path /opt/zigbee2mqtt
npm ERR! command failed
npm ERR! command sh -c node index.js

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2022-01-05T17_59_23_787Z-debug.log

### What did you expect to happen?

To run npm start 

### How to reproduce it (minimal and precise)

I dont know 

### Zigbee2MQTT version

I dont know 

### Adapter firmware version

I dont know 

### Adapter

I dont know 

### Debug log

root@raspberrypi:/opt/zigbee2mqtt# npm start

> zigbee2mqtt@1.22.2 start
> node index.js
root@raspberrypi:/opt/zigbee2mqtt# npm start 

> zigbee2mqtt@1.22.2 start
> node index.js

Zigbee2MQTT:info  2022-01-05 18:59:21: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-01-05.18-59-19' filename: log.txt
Zigbee2MQTT:info  2022-01-05 18:59:21: Starting Zigbee2MQTT version 1.22.2 (commit #301ba14b)
Zigbee2MQTT:info  2022-01-05 18:59:21: Starting zigbee-herdsman (0.13.188)
CREATED DECONZ ADAPTER
Zigbee2MQTT:info  2022-01-05 18:59:22: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-01-05 18:59:22: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":88,"product":0,"revision":"0x26580700","transportrev":0},"type":"ConBee2/RaspBee2"}'
Zigbee2MQTT:info  2022-01-05 18:59:22: Currently 0 devices are joined:
Zigbee2MQTT:warn  2022-01-05 18:59:22: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2022-01-05 18:59:22: Allowing new devices to join.
Zigbee2MQTT:warn  2022-01-05 18:59:22: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2022-01-05 18:59:22: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2022-01-05 18:59:22: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:error 2022-01-05 18:59:22: MQTT failed to connect: connect ECONNREFUSED 127.0.0.1:1883
Zigbee2MQTT:error 2022-01-05 18:59:22: Exiting...
Zigbee2MQTT:info  2022-01-05 18:59:23: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-01-05 18:59:23: Stopped zigbee-herdsman
npm ERR! code 1
npm ERR! path /opt/zigbee2mqtt
npm ERR! command failed
npm ERR! command sh -c node index.js

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2022-01-05T17_59_23_787Z-debug.log

Koenkk, 2022-01-05:
This does not look like a z2m bug, you should fix your MQTT server url:

```
Zigbee2MQTT:error 2022-01-05 18:59:22: MQTT failed to connect: connect ECONNREFUSED 127.0.0.1:1883
```

closed 2022-01-05 by Koenkk

filipevaladas, 2022-01-18:
Hi. Newbie on this MQTT/Homebridge/rapsberry world!

I have this issue too. To have zigbee working I need to manually launch npm start on /opt/zigbee2mqtt. When I close the command (CTRL+C) zigbee goes down on Homebridge + Zigbee frontend.

How can I make NPM always auto start on raspberry boot, so I don´t have to manually start it?

Tks

FergusGeorge, 2022-01-21:
That's a different issue, but the answer you're seeking (i.e. how to run it as a _service_) is in the official docs: https://www.zigbee2mqtt.io/guide/installation/01_linux.html#optional-running-as-a-daemon-with-systemctl