Koenkk/zigbee2mqtt #10601: [Problem]: 2 datapoints are not known - update -Lidl Modell 368308_2010 / TS0601 Zigbee Hersteller _TZE200_chyvmhay

gmarchart, 2022-01-05:
### What happened?

2 exposes are unknown at the moment - binary one and binary two

Binary one = Summer/Winter Time on Valve automatic set (off/on)
Binary Two = Open Window detected (off/on) off = no open window detected/on=open windows detected

not sure how to determine the datapoints - maybe someone can give me a hint, then i can check :-)

### What did you expect to happen?

if on valve the Summer / Wintertime is set via Mode/Menu "dSt" to off - no winter/summertime automatic change is set = binary one  =off
if on valve the summer/wintertime is set via Mode/Menu "dSt" to on - winter/summertime automiatic changes are set = binary one =  on

if on valve open Window detected -> binary two = on
if on valve NO open Window detected -> binary two = off

### How to reproduce it (minimal and precise)

if you set on valve via "dSt" menu  the summer/winter time to automatic - (on) - binary one is not reconcnized =only a ? between on/off
if you get a "window open detection " you see binary two = on

### Zigbee2MQTT version

1.22.2-dev commit: 4f887a1

### Adapter firmware version

 0x26660700

### Adapter

deconz/conbee 2

### Debug log

.

gmarchart, 2022-01-20:
is there any need for some debug output, or anything else to fix that ?
how can i help here ?

Koenkk, 2022-01-20:
There are some docs on this, see https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

gmarchart, 2022-02-25:
no one have the same problem ? :-(

closed 2022-03-05 by github-actions[bot]

veeroohre, 2023-04-12:
I have the same problem and would like to get a solution or help (if possible with my limited abilities)