Koenkk/zigbee2mqtt #10604: [New device support]: TS011F' and manufacturer name '_TZ3000_bfn1w0mm

Domoke12, 2022-01-05:
### Link

https://aliexpress.com/item/1005003646930652.html

### Database entry

{"id":20,"type":"Router","ieeeAddr":"0x70b3d52b600062ad","nwkAddr":12089,"manufId":0,"manufName":"_TZ3000_bfn1w0mm","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6],"outClusterList":[3,25,6],"clusters":{"genBasic":{$

### Comments

I'm new at this, I saw this in tuya.js but I don't know how to proceed to make a external converter.  It's with power monitoring but I don't know if it is with polling or not

const TS011Fplugs = ['_TZ3000_5f43h46b', '_TZ3000_cphmq0q7', '_TZ3000_dpo1ysak', '_TZ3000_ew3ldmgx', '_TZ3000_gjnozsaz',
    '_TZ3000_jvzvulen', '_TZ3000_mraovvmm', '_TZ3000_nfnmi125', '_TZ3000_ps3dmato', '_TZ3000_w0qqde0g', '_TZ3000_u5u4cakc',
    '_TZ3000_rdtixbnu', '_TZ3000_typdpbpg', '_TZ3000_v1pdxuqq'];

### External converter

```shell
{
        fingerprint: TS011Fplugs.map((manufacturerName) => {
            return {modelID: 'TS011F', manufacturerName};
        }),
        model: 'TS011F_plug_1',
        description: 'Smart plug (with power monitoring)',
        vendor: 'TuYa',
        whiteLabel: [{vendor: 'LELLKI', model: 'TS011F_plug'}, {vendor: 'NEO', model: 'NAS-WR01B'},
            {vendor: 'BlitzWolf', model: 'BW-SHP15'}],
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage')],
```

Koenkk, 2022-01-06:
Added, thanks.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-06 by Koenkk

Domoke12, 2022-01-10:
Sorry, it's wrong with type 1, maybe it is a type 3 with polling. Or the seller false advertising with powering when it is not. Sorry for the trouble.

Domoke12, 2022-01-11:
Ok, I tested it with type3 with polling. does not work. False advertising from seller plug does not have power metering.

Koenkk, 2022-01-11:
@Domoke12 thanks, marked it as `TS011F_plug_2` now (this model doesn't support power monitoring)