Koenkk/zigbee2mqtt #10605: Badly recognized AwoX GU10 33946

Block4r, 2022-01-05:
### Link

https://www.zigbee2mqtt.io/devices/33943_33944_33946.html

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0xa4c138c90028459b","nwkAddr":59393,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genBasic":{"attributes":{"appVersion":0,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1641414533804,"useImplicitCheckin":true} {"id":8,"type":"Router","ieeeAddr":"0xa4c1386223798ea2","nwkAddr":62891,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genBasic":{"attributes":{"appVersion":0,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1641414533732,"useImplicitCheckin":true}

### Comments

Hey, 
I've bought two LED GU10 bulbs Awox 33946 (https://www.zigbee2mqtt.io/devices/33943_33944_33946.html) and It gets discovered as a different device - 33951/33948 (https://www.zigbee2mqtt.io/devices/33951_33948.html#awox-33951%252F33948) 
Which is not completely true - My model has the option to change colours as well.
Below I attach some screens and data form database.db.
How can I make this device working correctly?

[image]
[image]
[image]

**database.db:**
{"id":7,"type":"Router","ieeeAddr":"0xa4c138c90028459b","nwkAddr":59393,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genBasic":{"attributes":{"appVersion":0,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1641414533804,"useImplicitCheckin":true}
{"id":8,"type":"Router","ieeeAddr":"0xa4c1386223798ea2","nwkAddr":62891,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genBasic":{"attributes":{"appVersion":0,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1641414533732,"useImplicitCheckin":true}

SHxKM, 2022-01-05:
[These were merged together](https://github.com/Koenkk/zigbee2mqtt/issues/8940#issuecomment-942675116) (~3 months ago) so this is "expected" from Z2M's standpoint. It was due to a [user complaint which is titled very similarly](https://github.com/Koenkk/zigbee2mqtt/issues/8940) to this issue.

Block4r, 2022-01-05:
I saw that but it didnt help me. :( 
Version Z2M is on the first screen - 1.22.2-dev commit: 16842a3 
I tested on stable version Z2M and currently on Edge - "Dev" - the same issue.

Koenkk, 2022-01-06:
Fixed this for your model now @Block4r .

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-06 by Koenkk