Koenkk/zigbee2mqtt #10608: [Problem]: IKEA LED1624G9 does not expose color_temp or related settings

chiragd, 2022-01-06:
### What happened?

I had an ikea bulb paired with the ikea hub and used it via home assistant. It exposed colour, brightness and temperature controls.

I unpaired the bulb from the ikea hub and paired it with my controller for zigbee2mqtt. I can control the bulb fine, but in zigbee2mqtt and home assistant I cannot see an color_temp controls as [described in the documentation](https://www.zigbee2mqtt.io/devices/LED1624G9.html#ikea-led1624g9).

What steps can I take to get back the colour temp controls? I have a few bulbs to move over so I can turn off the ikea hub.

### What did you expect to happen?

Color_temp controls should be available for this bulb

### How to reproduce it (minimal and precise)

Pair the IKEA LED1624G9 with zigbee2mqtt. Observe that color temperature (tunable white) is not available.

### Zigbee2MQTT version

1.16.2 commit: 04c15f7

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

### Debug log

N/A

chiragd, 2022-01-06:
Sorry - I thought I had upgraded zigbee2mqtt, but I ran the old container instead of the new one. On the latest version this seems fine.

chiragd, 2022-01-06:
Actually, this doesn't work.

Here are the logs. I see the color_temp controls, but I cannot set them. The device doesn't respond.

`MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"color":{"x":0.458,"y":0.41},"color_mode":"xy","color_temp":368,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/remote1', payload '{"action":null,"battery":34,"linkquality":255,"update":{"state":"available"},"update_available":true}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":197,"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"color_temp_startup":370,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"color_temp_startup":65535,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"color_temp_startup":65535,"linkquality":255,"power_on_behavior":null,"state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"color_temp_startup":65535,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"color_temp_startup":65535,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.458,"y":0.41},"color_mode":"xy","color_temp":368,"color_temp_startup":65535,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":null},"update_available":null}'
Error Publish 'get' 'color' to 'tradfri_bulb' failed: 'Error: Read 0xbc33acfffeab4e33/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'color' to 'tradfri_bulb' failed: 'Error: Read 0xbc33acfffeab4e33/1 lightingColorCtrl([\"colorMode\",\"currentX\",\"currentY\",\"colorTemperature\"], {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"tradfri_bulb"},"type":"zigbee_publish_error"}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.458,"y":0.41},"color_mode":"xy","color_temp":368,"color_temp_startup":65535,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.458,"y":0.41},"color_mode":"xy","color_temp":368,"color_temp_startup":65535,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":null},"update_available":null}'
Error Publish 'get' 'color_temp' to 'tradfri_bulb' failed: 'Error: Read 0xbc33acfffeab4e33/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[... cut ...]
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"color_temp_startup":65535,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/tradfri_bulb', payload '{"brightness":236,"color":{"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"color_temp_startup":65535,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":null},"update_available":null}'`

Tawmu, 2022-01-10:
Can also confirm not working on LED1624G9. Bulbs used to support color temp on other Zigbee providers (ZHA etc)

closed 2022-01-10 by Koenkk