Koenkk/zigbee2mqtt #10612: [Problem]: OSRAM 4062172044776 DALI dimmer become unsupported after updating to 1.2.22

jpauoc, 2022-01-06:
### What happened?

After updating Zigbee2MQTT from version 1.22.1 to 1.22.2, the OSRAM 4062172044776 DALI dimmer stops working. According to the logs and web UI, the devices is reported as unsupported and it cannot be used. Unpairing and pairing again does not solve the issue, since the device is still reported as unsupported. 

### What did you expect to happen?

The device is supported without problems as previous versions of Z2M.

### How to reproduce it (minimal and precise)

Two options:
1- Pair the device on a previous supported Z2M version (between 1.18.3 and 1.2,21) and the update to 1.2.22.
2.- Update to 1.2.22 and try to pair the device 

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB USB coordinator

### Debug log

info  2022-01-05 23:25:12: DALI Zigbee (0x000*************): Not supported (Router)
debug 2022-01-05 23:25:13: Received MQTT message on 'zigbee2mqtt/DALI Zigbee/availability' with data 'offline'
info  2022-01-05 23:25:25: MQTT publish: topic 'zigbee2mqtt/DALI Zigbee', payload '{"brightness_l1":142,"state_l1":"OFF"}'
debug 2022-01-05 23:29:07: Received Zigbee message from 'DALI Zigbee', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":142}' from endpoint 10 with groupID 0
warn  2022-01-05 23:29:07: Received message from unsupported device with Zigbee model 'Zigbee 3.0 DALI CONV LI
debug 2022-01-05 23:30:09: Received Zigbee message from 'DALI Zigbee', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 10 with groupID 0
warn  2022-01-05 23:30:09: Received message from unsupported device with Zigbee model 'Zigbee 3.0 DALI CONV LI

jpauoc, 2022-01-06:
I'm not sure if the problem is related to a change of the ID of this device. Since 1,2,22 version is now supporting 4 different types of the same device and the original ID (4062172044776) now incorporates a underscore to identify each one (4062172044776_1, 4062172044776_2, 4062172044776_3 and 4062172044776_4). In my case, I assume that now should be identified as 4062172044776_2.

Koenkk, 2022-01-06:
Could you provide the data/database.db entry of your device?

jpauoc, 2022-01-07:
Sure (I just put * to blind addresses, but if you need to know these values, let me know):
{"id":39,"type":"Router","ieeeAddr":"0x000********","nwkAddr":*****,"manufId":4364,"manufName":"Osram Sylvania\u0000","powerSource":"Mains (single phase)","modelId":"Zigbee 3.0 DALI CONV LI\u0000","epList":[10,25,242],"endpoints":{"10":{"profId":260,"epId":10,"devId":257,"inClusterList":[0,3,4,5,6,8,769,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Zigbee 3.0 DALI CONV LI\u0000","manufacturerName":"Osram Sylvania\u0000","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":0,"hwVersion":5,"dateCode":"","swBuildId":"1.0.0.4-6.5.5.b"}},"genLevelCtrl":{"attributes":{"currentLevel":144}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00***********","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00************","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":300,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":3,"maxRepIntval":300,"repChange":1}],"meta":{}},"25":{"epId":25,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":0,"hwVersion":5,"dateCode":"","swBuildId":"1.0.0.4-6.5.5.b","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2,"reporting":1},"lastSeen":1641515993630,"useImplicitCheckin":true}

Koenkk, 2022-01-07:
Should be fixed now, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

jpauoc, 2022-01-07:
Hi, tested with the dev branch and now it works! Thx!

It is detected as 4062172044776_2. To be honest, I do not know if my device allows to manage one or two end-points, but since I'm using it to control only one, it is fine for me.

Just one detail: logs report global and end-point L1 state and brightness. State and brightness of L1 do not change when switching on/off or changing brightness (using Z2M service or the push button). Only global state does. These are the logs:

info  2022-01-07 19:47:16: MQTT publish: topic 'zigbee2mqtt/DALI Zigbee', payload '{"brightness":126,"brightness_l1":144,"linkquality":36,"state":"OFF","state_l1":"OFF"}'
info  2022-01-07 19:48:08: MQTT publish: topic 'zigbee2mqtt/DALI Zigbee', payload '{"brightness":126,"brightness_l1":144,"linkquality":36,"state":"ON","state_l1":"OFF"}'
info  2022-01-07 19:49:02: MQTT publish: topic 'zigbee2mqtt/DALI Zigbee', payload '{"brightness":126,"brightness_l1":144,"linkquality":39,"state":"ON","state_l1":"OFF"}'
info  2022-01-07 19:53:09: MQTT publish: topic 'zigbee2mqtt/DALI Zigbee', payload '{"brightness":126,"brightness_l1":144,"linkquality":36,"state":"ON","state_l1":"OFF"}'
info  2022-01-07 19:54:02: MQTT publish: topic 'zigbee2mqtt/DALI Zigbee', payload '{"brightness":126,"brightness_l1":144,"linkquality":36,"state":"ON","state_l1":"OFF"}'
info  2022-01-07 19:55:53: MQTT publish: topic 'zigbee2mqtt/DALI Zigbee', payload '{"brightness":126,"brightness_l1":144,"linkquality":36,"state":"OFF","state_l1":"OFF"}'
info  2022-01-07 19:56:17: MQTT publish: topic 'zigbee2mqtt/DALI Zigbee', payload '{"brightness":126,"brightness_l1":144,"linkquality":36,"state":"ON","state_l1":"OFF"}'
info  2022-01-07 19:56:21: MQTT publish: topic 'zigbee2mqtt/DALI Zigbee', payload '{"brightness":125,"brightness_l1":144,"linkquality":36,"state":"ON","state_l1":"OFF"}'
info  2022-01-07 19:56:24: MQTT publish: topic 'zigbee2mqtt/DALI Zigbee', payload '{"brightness":71,"brightness_l1":144,"linkquality":36,"state":"ON","state_l1":"OFF"}'
info  2022-01-07 19:56:27: MQTT publish: topic 'zigbee2mqtt/DALI Zigbee', payload '{"brightness":49,"brightness_l1":144,"linkquality":36,"state":"ON","state_l1":"OFF"}'

jpauoc, 2022-01-07:
BTW, one thing that I have been not able to do is to make the device to remember the state, so in case of a power loss, the light controlled by the DALI Zigbee device does not remain switch on when the power is back. I tried using the "remember_state" attribute unsuccessfully. But this was not working also on the previous versions.

Koenkk, 2022-01-07:
@jpauoc you might need to clear the cached state, stop z2m open data/state.json and remove the entry of this device.

jpauoc, 2022-01-07:
Yep. Cache cleared and now only reports the main state and brightness.

Thx!

closed 2022-01-07 by jpauoc

mario-tux, 2022-10-21:
Sorry for posting on this closed report. I would ask some information to @jpauoc as owner of the osram device. I still wondering how it works.
If I connect a dali dimmerable light and an ordinary push button, what does happen?
- can I switch on/off the light without any configuration or automation on the domotica backend (e.g. Home Assistant)?
- with a dimmerable light, can I dimm it keeping pushed the button?
- to make it work without HA, do I need to bind the ZigBee button exposed by the converter to the ZigBee light?

Thanks and sorry for the thread highjack.

jpauoc, 2022-11-14:
Hi @mario-tux, I did not realize about your question since now.

I have three push buttons (connected in parallel) and a DALI dimmable light connected to the Osram device (i.e., the 4062172044776_2 zigbee2mqtt configuration). Both can work independently from the Zigbee gateway. More concretely:
  - A single push switches on/off the DALI dimmable device
  - A long push dimmers up/down the DALI light
 
I tested this with zigbee2mqtt service down and it works. So I can confirm that push button does not require zigbee gateway for interacting with the DALI light. When zigbee2mqtt is up, the on/off and brightness states are updated when modified using the push button. However, brightness value received and kept by Zigbee gateway is slightly higher than the real one. So when switching off and on again the light, this retrieves the value from gateway and therefore it is higher, so it needs to be slightly corrected again on zigbee2mqtt interface.

The only thing I have not been able to make it work is remembering the previous state in case of power loss. So when this happens, the light switches on at top brightness... I do not know where this state information should be kept: in the Osram device or the DALI light. But the fact is that none of them are keeping this information and I need to periodically check that the light is not on... 

Hope it helps,