Koenkk/zigbee2mqtt #10613: [Problem]: Unable to pair removed devices after migrating to new coordinator

craggus2000, 2022-01-06:
### What happened?

I've just migrated from a CC2531 Texas Instruments USB coordinator to Tubes ZB PoE CC2652P2 PoE Coordinator. Everything seemed to be working fine, but then I noticed that some battery devices were not reporting. Ikea Tradfri Motion Detector and Sonoff Temperature Sensor SNZB-02. So removed them to re-pair and I am unable to re-pair them. I can't see anything obvious in any logs.
I then further tested by removing a mains powered light bulb, and was unable to re-pair that as well.
Then one of my battery Ikea switches battery died, so I replaced the battery with a new one, and this is not communicating either. Tried removing it, and re-pairing, but it won't re-pair either.
It seems I can't re-pair any device that has been removed or run out of battery.

### What did you expect to happen?

I expected to be able to re-pair them.

### How to reproduce it (minimal and precise)

Remove a device then try to re-pair.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210708

### Adapter

Tube's Zigbee CC2652P2 PoE Coordinator

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7821340/log.txt)

Ignore lines relating to failed to ping trafri driver 1, that's because I use it in automations. this is not part of my problem.

Tonio16, 2022-01-06:
Hello
You should have a look here https://github.com/Koenkk/Z-Stack-firmware/issues
Your issue is already reported there.
https://github.com/Koenkk/Z-Stack-firmware/issues/338
Or
https://github.com/Koenkk/zigbee2mqtt/issues/10339

Antoine

craggus2000, 2022-01-07:
Thanks.
I have now solved this, by shutting down my entire zigbee network and restarting. Pairing was then successful

closed 2022-01-07 by craggus2000