Koenkk/zigbee2mqtt #10614: [Problem]: Rob_200-017-0 get detected as Sunricher HK-plug-A

turnhob, 2022-01-06:
### What happened?

I do have the ROB_200_017-0 but when i add it to zigbee2mqtt (with a Conbee II) it tells me its not supported.
The model name that shows is: HK-PLUG-A and the manufacturar is Sunricher.
So i don't understand why it does that (I assume ROBB did use Sunricher and uses his own brand). But now im not able to use my Smart Plug.
Can U help me how to fix this. Im using zigbee2mqtt with home assistant and tried to create a converter file, but when i follow the instructions my add-on wont start anymore when i changed the configuration (as mentioned on your page).

Besides that.. Moving from deconz to zigbee2mqtt was the best thing i could have done.

### What did you expect to happen?

I expected the smart plug to be discover as ROB_200-017-0 from manufacturer ROBB

### How to reproduce it (minimal and precise)

Repairing the SmartPlug to Zigbee2mqtt gives the same "Unsupported" result everytime with the message that the HK-plug-A from Sunricher is not supported.

### Zigbee2MQTT version

 1.22.2

### Adapter firmware version

0x26660700

### Adapter

ConBee2/RaspBee2

### Debug log

debug 2022-01-06 13:02:20: Received Zigbee message from '0x60a423fffebe33bc', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Sunricher"}' from endpoint 1 with groupID null
debug 2022-01-06 13:02:20: Skipping message, definition is undefined and still interviewing
debug 2022-01-06 13:02:20: Received Zigbee message from '0x60a423fffebe33bc', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID null
debug 2022-01-06 13:02:20: Skipping message, definition is undefined and still interviewing
debug 2022-01-06 13:02:20: Received Zigbee message from '0x60a423fffebe33bc', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
debug 2022-01-06 13:02:20: Skipping message, definition is undefined and still interviewing
debug 2022-01-06 13:02:21: Received Zigbee message from '0x60a423fffebe33bc', type 'readResponse', cluster 'genBasic', data '{"appVersion":0}' from endpoint 1 with groupID null
debug 2022-01-06 13:02:21: Skipping message, definition is undefined and still interviewing
debug 2022-01-06 13:02:21: Received Zigbee message from '0x60a423fffebe33bc', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID null
debug 2022-01-06 13:02:21: Skipping message, definition is undefined and still interviewing
debug 2022-01-06 13:02:21: Received Zigbee message from '0x60a423fffebe33bc', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
debug 2022-01-06 13:02:21: Skipping message, definition is undefined and still interviewing
debug 2022-01-06 13:02:21: Received Zigbee message from '0x60a423fffebe33bc', type 'readResponse', cluster 'genBasic', data '{"dateCode":"NULL"}' from endpoint 1 with groupID null
debug 2022-01-06 13:02:21: Skipping message, definition is undefined and still interviewing
debug 2022-01-06 13:02:21: Received Zigbee message from '0x60a423fffebe33bc', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2.5.3_r2"}' from endpoint 1 with groupID null
debug 2022-01-06 13:02:21: Skipping message, definition is undefined and still interviewing
info  2022-01-06 13:02:21: Successfully interviewed '0x60a423fffebe33bc', device has successfully been paired
warn  2022-01-06 13:02:21: Device '0x60a423fffebe33bc' with Zigbee model 'HK-PLUG-A' and manufacturer name 'Sunricher' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info  2022-01-06 13:02:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x60a423fffebe33bc","ieee_address":"0x60a423fffebe33bc","status":"successful","supported":false},"type":"device_interview"}'
info  2022-01-06 13:02:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x60a423fffebe33bc","supported":false},"type":"pairing"}'[](url)

turnhob, 2022-01-06:
Did fix it myself. (not bad for a complete programming noob :)
In the converterfile robb.js I replaced zigbeeModel: ['ROB_200-017-0'] with ['HK-PLUG-A'] and after that it was supported.
But i don't know how to make a proper pull request. So I hope you will be able to adjust it in the file.

closed 2022-02-15 by github-actions[bot]

freakshock88, 2022-07-23:
It looks like this issue is still not resolved, facing the same thing on 1.26.0.

Koenkk, 2022-07-23:
Added! 
Changes will be available in the dev branch tomorrow. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) (and the next release on 1 August)

freakshock88, 2022-07-23:
> Added! Changes will be available in the dev branch tomorrow. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) (and the next release on 1 August)

Friggin awesome thanks Koen.

freakshock88, 2022-08-05:
I have just successfully added the plug using Z2M 1.27.0.
The switching works as expected. 
However, the energy sensor (Sum of consumed energy) stays on Null kWh.
Is that something that can be resolved? 
If you need me to create a new bug Koen, let me know.

Koenkk, 2022-08-06:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    zigbeeModel: ['ROB_200-017-0', 'HK-PLUG-A'],
    model: 'ROB_200-017-0',
    vendor: 'ROBB',
    description: 'Zigbee smart plug',
    fromZigbee: [fz.electrical_measurement, fz.on_off, fz.ignore_genLevelCtrl_report, fz.metering],
    toZigbee: [tz.on_off],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering', 'haElectricalMeasurement']);
        await reporting.onOff(endpoint);
        await reporting.readEletricalMeasurementMultiplierDivisors(endpoint);
        await reporting.readMeteringMultiplierDivisor(endpoint);
        await reporting.rmsVoltage(endpoint);
        await reporting.rmsCurrent(endpoint);
        await reporting.activePower(endpoint);
        await reporting.currentSummDelivered(endpoint);
    },
    exposes: [e.power(), e.current(), e.voltage().withAccess(ea.STATE), e.switch(), e.energy()],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed