Koenkk/zigbee2mqtt #10615: [Problem]: ZemiSmart Smart 6 key scene switch does not work

RJArmitage, 2022-01-06:
### What happened?

1.22.2 adds support for the Smart 6 key scene switch from ZemiSmart. Using this version, and also with hassio-edge I can add the device and it claims to be supported. However the buttons don't do anything. Checking the log I see that a command message is received but that zb2m reports there is no converter

### What did you expect to happen?

Clicking on a device button should send an action message

### How to reproduce it (minimal and precise)

1. Pair device
2. Wait for interview to complete
3. Click on a device button

### Zigbee2MQTT version

1.22.2-dev commit: 43ded11

### Adapter firmware version

0x26660700

### Adapter

ConBee2

### Debug log

[image]
[image]

Koenkk, 2022-01-06:
Did an attempt to fix it, could you check if it works?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

RJArmitage, 2022-01-06:
I'm using the hassio plugin so I'll need to wait for the edge plugin build to include your change. I'll let you know...

RJArmitage, 2022-01-06:
Have retested - that change seems to have got the messages flowing. However I now see that it is not handling battery reporting properly. The battery state is sent as a 4 byte integer in DP10. However the plugin is treating this as an unknown action message for button 10. So I would say that for dp10 you could parse as: 
```
const button = dpValue.dp;
if (button == 10) {
    const batteryValue = tuya.getDataValue(dpValue);
    return {battery: batteryValue};
} else { 
    const actionValue = tuya.getDataValue(dpValue);
    ....
    return {action: `button_${button}_${action}`};
}
```
and resurrect 
```
    exposes: [e.battery(), e.action([...
```

DHM47, 2022-01-06:
The device is working again with the latest edge version

Koenkk, 2022-01-06:
@RJArmitage applied, can this be closed?

RJArmitage, 2022-01-06:
I can retest tomorrow morning. I’ll let you know if we’re all ok.

RJArmitage, 2022-01-06:
Have retested the latest dev. All seems to be working correctly. Battery is being reported correctly and appears in Home Assistant. I think we can close this now. Thanks.

closed 2022-01-06 by RJArmitage

RJArmitage, 2022-01-07:
> I just tested and everything is working.
> @RJArmitage thanks for solving the battery reading.

No problem

DavidFFerreira, 2022-01-09:
no probleme now, everything works great :D