Koenkk/zigbee2mqtt #10617: [New device support]: Philips Hue Centura recessed spot light 929003047501

Klangen82, 2022-01-06:
### Link

https://www.philips-hue.com/en-gb/p/hue-white-and-colour-ambiance-centura-recessed-spotlight/8719514338647

### Database entry

Starting interview of '0x001788010bbb2e04' Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010bbb2e04","ieee_address":"0x001788010bbb2e04","status":"started"},"type":"device_interview"}' Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001788010bbb2e04"},"type":"pairing"}' Info Successfully interviewed '0x001788010bbb2e04', device has successfully been paired Warning Device '0x001788010bbb2e04' with Zigbee model '929003047501' and manufacturer name 'Philips' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### Comments

I bought 5pcs Philips Hue Centura recessed spotlights and it should have support for it in Z2M. https://www.zigbee2mqtt.io/supported-devices/#v=Philips&s=centura
But it seams like the model name have changed to  929003047501 instead of 5045131P7. I have never added a device to Z2M but I found that there is two ways. I tried just to add the following code to the devices.js under Philips:
``` 
  {
        zigbeeModel: ['929003047501'],
        model: '929003047501',
        vendor: 'Philips',
        description: 'Hue Centura',
        meta: {turnsOffAtBrightness1: true},
        extend: preset.hue.light_onoff_brightness_colortemp_color,
        ota: ota.zigbeeOTA,
    },
```
When I do this I get Support status = Unsupported so it will not work. I also tried to add a new device as external converter (929003047501.js) and then it get's supported but when I go to Expose tab I only see text and cannot control the light.

[image]

I have also tried the edge for HA but no luck.

### External converter

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const hueExtend = {
    light_onoff_brightness_colortemp_color: (options={}) => ({
        ...extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options}),
        toZigbee: extend.light_onoff_brightness_colortemp_color({supportsHS: true, ...options})
            .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
};

const definition = {
    zigbeeModel: ['929003047501'],
    model: '929003047501',
    vendor: 'Philips',
    description: 'Centura recessed spotlight',
    meta: {turnsOffAtBrightness1: true},
    fromZigbee: [fz.command_on_state, fz.command_off_state], // We will add this later
    toZigbee: [],
    exposes: [hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]})],
    ota: ota.zigbeeOTA,    
};

module.exports = definition;
```

### Supported color modes

light (state, brightness, color_temp, color_temp_startup, color_xy, color_hs), effect, linkquality

Klangen82, 2022-01-06:
Can add that I use HA as frontend and here are the version numbers I run at the moment.

Zigbee2MQTT version: 1.22.2 commit: unknown
Coordinator type: zStack3x0
Coordinator revision: 20211217
Frontend version: 0.6.67

Koenkk, 2022-01-06:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-06 by Koenkk

Klangen82, 2022-01-06:
> Added.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Thanks, it works fine now:)

Klangen82, 2022-01-06:
> Added.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Or sorry I spoke to early:/ This is a RGB device, it's not possible to control the color.

Koenkk, 2022-01-06:
@Klangen82 fixed.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Klangen82, 2022-01-06:
> @Klangen82 fixed.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Thanks again! Works fine now @Koenkk Keep up the good work!