Koenkk/zigbee2mqtt #10619: [Problem]: Incorrect Pearl 3157100 Features

Justahobby01, 2022-01-06:
### What happened?

The Centralite Pearl Thermostat model: 3157100

Unable to change temperature without setting a mix/max value, which is incorrect for this device.

### What did you expect to happen?

The thermostat expects to receive a specific temperature setpoint, not a range.

### How to reproduce it (minimal and precise)

I recently moved from ZHA to zigbee2mqtt, but Home Assistant is now requiring a min and max temperature.  Not only is it a convoluted way to adjust the thermostat, but I believe it is an incorrect implementation for this thermostat since the ```hvac_modes``` is not ```heat_cool```. Curiously, the thermostat shows the correct modes under developer tools, but the climate card requires a min/max, and I am now unable to call the ```climate.set_temperature``` without the min/max value.

### Zigbee2MQTT version

 1.22.2 commit: 414c51f

### Adapter firmware version

20210708

### Adapter

Sonoff USB 3.0

### Debug log

Debug Publishing 'set' 'occupied_cooling_setpoint' to 'Thermostat'
Error Publish 'set' 'occupied_heating_setpoint' to 'Thermostat' failed: 'Error: Write 0x000d6f000ade91f6/1 hvacThermostat({"occupiedHeatingSetpoint":2000}, {"sendWhenActive":true,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'HARDWARE_FAILURE')'
Debug Error: Write 0x000d6f000ade91f6/1 hvacThermostat({"occupiedHeatingSetpoint":2000}, {"sendWhenActive":true,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'HARDWARE_FAILURE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:310:23) at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:392:22) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1278:13) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
Debug Received Zigbee message from 'Thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedCoolingSetpoint":2222}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Thermostat', payload '{"battery":98,"fan_mode":"auto","fan_state":"ON","linkquality":87,"local_temperature":19.39,"local_temperature_calibration":null,"occupied_cooling_setpoint":22.22,"occupied_heating_setpoint":20,"running_state":"idle","system_mode":"cool","voltage":2700}'
Debug Received MQTT message on 'zigbee2mqtt/Thermostat/set/occupied_heating_setpoint' with data '20.555555555555554'
Debug Publishing 'set' 'occupied_heating_setpoint' to 'Thermostat'
Debug Received MQTT message on 'zigbee2mqtt/Thermostat/set/occupied_cooling_setpoint' with data '22.22222222222222'
Debug Publishing 'set' 'occupied_cooling_setpoint' to 'Thermostat'
Debug Received Zigbee message from 'Thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":2056}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Thermostat', payload '{"battery":98,"fan_mode":"auto","fan_state":"ON","linkquality":90,"local_temperature":19.39,"local_temperature_calibration":null,"occupied_cooling_setpoint":22.22,"occupied_heating_setpoint":20.56,"running_state":"idle","system_mode":"cool","voltage":2700}'

mwav3, 2022-01-13:
I noticed the same thing with Home Assistant, and based on the documentation, https://www.home-assistant.io/integrations/climate/#service-climateset_hvac_mode , a call service to set the temperature is required when the mode is heat_cool, but this thermostat runs in either heat or cool mode (not full auto) as pointed out.  Probably the template needs updating which I can try and take a look at (I just did an update to add expose the hold function recently https://github.com/Koenkk/zigbee-herdsman-converters/pull/3627)

However, a bigger issue which I notice in your logs as well, is I keep getting "hardware_failure" errors in my logs, and the temperature doesn't update, and that's a bigger concern for me.  At first I thought it was something wrong with the template in Home Assistant similar to what was posted above, but posting from Node Red to MQTT direct (which completely bypasses Home Assistant) still results in the error occasionally, and the thermostat doesn't update.  I've started listening for the error on zigbee2mqtt/bridge/log and have Node Red try again when the error happens, and every time the thermostat updates the setpoint fine the second time around.  I have 3 different Centralite pearl 3157100-E (3 zone heat) and they all do the same thing, so unlikely its a defective product.  I've tried changing the optimistic, retain, and QOS settings but the error keeps happening.  I'm using the Node Red zigbee2mqtt pallette https://flows.nodered.org/node/node-red-contrib-zigbee2mqtt 

When I publish just a heat setpoint change using a zigbee2mqtt out node like this:

[image]

My log will occasionally just show an entry like this with an error and not update the thermostat:

```
error 2022-01-12 19:56:36: Publish 'set' 'occupied_heating_setpoint' to 'Upper Floor Heat' failed: 'Error: Write 0x000d6f000304c118/1 hvacThermostat({"occupiedHeatingSetpoint":1944.0000000000002}, {"sendWhenActive":true,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'HARDWARE_FAILURE')'
info  2022-01-12 19:56:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'occupied_heating_setpoint' to 'Upper Floor Heat' failed: 'Error: Write 0x000d6f000304c118/1 hvacThermostat({\"occupiedHeatingSetpoint\":1944.0000000000002}, {\"sendWhenActive\":true,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'HARDWARE_FAILURE')'","meta":{"friendly_name":"Upper Floor Heat"},"type":"zigbee_publish_error"}'
```

I've added repeaters and have a pretty good mesh, and the upstairs thermostat is in the same room as the coordinator with an LQI showing 200.  I'm using the ZZH coordinator.  Not really sure what is going wrong here.

mwav3, 2022-01-14:
I enabled debug logging to try and get more info.  Its starting to seem like an issue with the math calculation of publishing the temperature.  When you set the temp to 19.44 degrees, its supposed to move the decimal and publish 1944 to the cluster, but instead it is trying to publish 1944.0000000000002 with all those extra decimal places and failing.  The thermostat is running in Fahrenheit but needs the temperature published to the cluster in Celsius, and something seems to be going wrong with the process there.

I still don't get why most of the time its successful and sometimes not though.

Here's a change with the error

```
debug 2022-01-14 06:15:56: Received MQTT message on 'zigbee2mqtt/Upper Floor Heat/set' with data '{"occupied_heating_setpoint":19.44}'
debug 2022-01-14 06:15:56: Publishing 'set' 'occupied_heating_setpoint' to 'Upper Floor Heat'
error 2022-01-14 06:15:59: Publish 'set' 'occupied_heating_setpoint' to 'Upper Floor Heat' failed: 'Error: Write 0x000d6f000304c118/1 hvacThermostat({"occupiedHeatingSetpoint":1944.0000000000002}, {"sendWhenActive":true,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'HARDWARE_FAILURE')'
debug 2022-01-14 06:15:59: Error: Write 0x000d6f000304c118/1 hvacThermostat({"occupiedHeatingSetpoint":1944.0000000000002}, {"sendWhenActive":true,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'HARDWARE_FAILURE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:310:23)
    at Endpoint.write (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:392:22)
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1278:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:36)
info  2022-01-14 06:15:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'occupied_heating_setpoint' to 'Upper Floor Heat' failed: 'Error: Write 0x000d6f000304c118/1 hvacThermostat({\"occupiedHeatingSetpoint\":1944.0000000000002}, {\"sendWhenActive\":true,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'HARDWARE_FAILURE')'","meta":{"friendly_name":"Upper Floor Heat"},"type":"zigbee_publish_error"}'

```

As a contrast, here is a successful change:

```
info  2022-01-14 14:47:50: MQTT publish: topic 'zigbee2mqtt/Upper Floor Heat', payload '{"battery":100,"fan_mode":"auto","fan_state":"ON","linkquality":218,"local_temperature":20.37,"local_temperature_calibration":0,"occupied_cooling_setpoint":26.67,"occupied_heating_setpoint":20,"programming_operation_mode":"setpoint","running_state":"heat","system_mode":"heat","temperature_setpoint_hold":false,"voltage":2800}'
debug 2022-01-14 14:47:51: Received MQTT message on 'zigbee2mqtt/Upper Floor Heat/set/occupied_heating_setpoint' with data '18.333333333333332'
debug 2022-01-14 14:47:51: Publishing 'set' 'occupied_heating_setpoint' to 'Upper Floor Heat'
debug 2022-01-14 14:47:51: Received MQTT message on 'zigbee2mqtt/Upper Floor Heat/set/occupied_cooling_setpoint' with data '26.666666666666664'
debug 2022-01-14 14:47:51: Publishing 'set' 'occupied_cooling_setpoint' to 'Upper Floor Heat'
debug 2022-01-14 14:47:51: Received Zigbee message from 'Upper Floor Heat', type 'attributeReport', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":1832}' from endpoint 1 with groupID 0
info  2022-01-14 14:47:52: MQTT publish: topic 'zigbee2mqtt/Upper Floor Heat', payload '{"battery":100,"fan_mode":"auto","fan_state":"ON","linkquality":218,"local_temperature":20.37,"local_temperature_calibration":0,"occupied_cooling_setpoint":26.67,"occupied_heating_setpoint":18.32,"programming_operation_mode":"setpoint","running_state":"heat","system_mode":"heat","temperature_setpoint_hold":false,"voltage":2800}'
debug 2022-01-14 14:47:53: Received Zigbee message from 'Upper Floor Heat', type 'attributeReport', cluster 'hvacThermostat', data '{"runningState":0}' from endpoint 1 with groupID 0
info  2022-01-14 14:47:54: MQTT publish: topic 'zigbee2mqtt/Upper Floor Heat', payload '{"battery":100,"fan_mode":"auto","fan_state":"ON","linkquality":218,"local_temperature":20.37,"local_temperature_calibration":0,"occupied_cooling_setpoint":26.67,"occupied_heating_setpoint":18.32,"programming_operation_mode":"setpoint","running_state":"idle","system_mode":"heat","temperature_setpoint_hold":false,"voltage":2800}'
```

Justahobby01, 2022-02-05:
Bump - I'm really hoping someone can help me sort through this broken integration. There should only be a single temperature setpoint based on whether heating of cooling mode is selected. 

[image]

`
Debug Received MQTT message on 'zigbee2mqtt/Thermostat/set/occupied_heating_setpoint' with data '20.0'
Debug Publishing 'set' 'occupied_heating_setpoint' to 'Thermostat'
Debug Received MQTT message on 'zigbee2mqtt/Thermostat/set/occupied_cooling_setpoint' with data '27.77777777777778'
Debug Publishing 'set' 'occupied_cooling_setpoint' to 'Thermostat'
Error Publish 'set' 'occupied_cooling_setpoint' to 'Thermostat' failed: 'Error: Write 0x000d6f000ade91f6/1 hvacThermostat({"occupiedCoolingSetpoint":2778}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_VALUE')'
Debug Error: Write 0x000d6f000ade91f6/1 hvacThermostat({"occupiedCoolingSetpoint":2778}, {"sendWhen":"fastpoll","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_VALUE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28) at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:397:22)`

Exit2Studios, 2022-02-20:
I have the same issue/concern.  I just switched from ZHA where this device worked well, but now has a range.  If this isn't resolved, I will just go back to ZHA.

mwav3, 2022-03-02:
The problem is these devices control both heating and cooling, and the climate integration within Home Assistant requires specifying both the target_temp_low and target_temp_high to change either one, and it creates the lovelace card showing a range.

https://www.home-assistant.io/integrations/climate/#service-climateset_temperature

If your device just does heating, you can create a custom mqtt template in yaml to create a heat only thermostat without showing the range and just specifying one temperate in the service.climateset_temperature .

Here's a sample template

```
climate:
  - platform: mqtt
    name: Lower Test Heat
    action_template: "{% set values = {None:None,'idle':'off','heat':'heating','cool':'cooling','fan_only':'fan'}
  %}{{ values[value_json.running_state] }}"
    action_topic: zigbee2mqtt/Lower Floor Heat
    availability:
    - topic: zigbee2mqtt/bridge/state
    - topic: zigbee2mqtt/Lower Floor Heat/availability
    availability_mode: all
    current_temperature_template: "{{ value_json.local_temperature }}"
    current_temperature_topic: zigbee2mqtt/Lower Floor Heat
    json_attributes_topic: zigbee2mqtt/Lower Floor Heat
    max_temp: '30'
    min_temp: '10'
    mode_command_topic: zigbee2mqtt/Lower Floor Heat/set/system_mode
    mode_state_template: "{{ value_json.system_mode }}"
    mode_state_topic: zigbee2mqtt/Lower Floor Heat
    modes:
    - 'off'
    - heat
    temp_step: 1
    temperature_command_topic: zigbee2mqtt/Lower Floor Heat/set/occupied_heating_setpoint
    temperature_state_template: "{{ value_json.occupied_heating_setpoint }}"
    temperature_state_topic: zigbee2mqtt/Lower Floor Heat
    temperature_unit: C
    unique_id: lower_floor_test_climate_zigbee2mqtt
```

Resulting climate card:

[image]

Justahobby01, 2022-03-09:
Thank you for the reply and explanation.

Exit2Studios, 2022-03-09:
I guess I'm confused why it works (with both heat and cool) in ZHA but not M2Z?

Justahobby01, 2022-03-09:
My guess, HA created the dumpster fire previously, but you/I didn't notice until the climate config was replaced by switching to z2m or vice versa.

Justahobby01, 2022-03-09:
Had you deleted the climate entity from zha before you switched back to zha?

Exit2Studios, 2022-03-09:
Yes I did.  I recreated it each time when switching.

Justahobby01, 2022-03-23:
> Yes I did. I recreated it each time when switching.

I finally had a chance to switch back to ZHA and can also confirm it is fixed on their end. So, hopefully Z2M can implement a similar fix.

closed 2022-04-30 by github-actions[bot]

Exit2Studios, 2022-11-30:
I realize this was stale and is now closed, but has there been any progress to work like the ZHA implementation with only one set point?

Justahobby01, 2022-11-30:
I seriously doubt it.

chiebert, 2022-12-10:
> 

@mwav3 Where does one implement this custom mqtt template? On the z2m side, or HA? It's my first introduction to home automation and wouldn't you know it, this is the thermostat I have [rolls eyes].

Justahobby01, 2022-12-10:
It would go in your home assistant config.

chiebert, 2022-12-11:
lol - now I have just enough knowledge to be dangerous, and not enough to be helpful. I made some changes for my context to that mqtt template, popped it into the HA configuration.yaml, but I clearly screwed up _all_ of my climate sensors. I've searched through the HA documentation on templating and _thought_ I knew what was going on, but I was wrong.

If anyone has a moment to step me through how to implement that sort of thing in my system (2x Stelpro thermostats that work fine, and 2x Centralite Pearl 3157100). That card was so tantalizing - but as it stands, I can't get automations to correctly set_temperature on either of the Centralites (the Stelpros are fine)

chiebert, 2022-12-12:
So I can't seem to control the Centralite 3157100 from HA at all, via Z2M. If I try to use the developer tools console to set_temperature, or if I try to target_temp_high, I get the following in the logs:

`Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/connection.py:150 
Integration: Home Assistant WebSocket API (documentation, issues) 
First occurred: 8:20:10 PM (5 occurrences) 
Last logged: 8:29:55 PM

[140356957943120] Error handling message: some but not all values in the same group of inclusion 'temperature' @ data[<temperature>]. Got None (invalid_format) from 10.0.0.252 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/16.1 Safari/605.1.15)
[140356957943120] Error handling message: must contain at least one of temperature, target_temp_high, target_temp_low.. Got {'type': 'execute_script', 'sequence': [{'service': 'climate.set_temperature', 'data': {}, 'target': {'entity_id': ['climate.living_room_thermostat']}}], 'id': 244} (invalid_format) from 10.0.0.252 (Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/16.1 Safari/605.1.15)
`

Not sure where to go with this. Any advice?

randalotto, 2023-02-07:
I just installed 4 of these thermostats, and this issue is still very much alive. Any progress?

Justahobby01, 2023-02-07:
Since this one was closed it would be better to open a new issue and link to this one.

seanharsh, 2023-02-08:
This issue may be closed and unresolved in terms of dual-setpoint issues, but I have a quick and dirty workaround that allows the schedule card to work with this thermostat. Schedule card only sends one setpoint, but this automation can intercept and re-transmit it to the thermostat correctly. 

https://github.com/seanharsh/Zigbee2Mqtt-Dual-SetPoint-Fix

Maybe it will wok for you.

jdekarske, 2023-12-29:
I think this is working as intended, but has a UX issue.

`occupied_{heating,cooling}_setpoint` are the setpoints stored in the thermostat's memory, not a range where modes are shifted autonomously. If anything, the thermostat config needs an update to what it exposes to Home Assistant, or Home Assistant must handle this niche config.

XeKToReX, 2024-06-21:
I was having the same issue with my Pearl, after a bit of trial and error and some assistance from ChatGPT, I got the following config to work.
```
mqtt:
  - climate:
      name: Thermostat
      action_template: "{% set values = {None:None,'idle':'off','heat':'heating','cool':'cooling','fan_only':'fan'}  %}{{ values[value_json.running_state] }}"
      action_topic: zigbee2mqtt/Hallway Thermostat
      availability:
      - topic: zigbee2mqtt/bridge/state
        value_template: "{{ value_json.state }}"
        payload_available: "online"
        payload_not_available: "offline"
      availability_mode: any
      current_temperature_template: "{{ value_json.local_temperature }}"
      current_temperature_topic: zigbee2mqtt/Hallway Thermostat
      json_attributes_topic: zigbee2mqtt/Hallway Thermostat
      max_temp: '30'
      min_temp: '10'
      mode_command_topic: zigbee2mqtt/Hallway Thermostat/set/system_mode
      mode_state_template: "{{ value_json.system_mode }}"
      mode_state_topic: zigbee2mqtt/Hallway Thermostat
      modes:
      - 'off'
      - heat
      temp_step: 1
      temperature_command_topic: zigbee2mqtt/Hallway Thermostat/set/occupied_heating_setpoint
      temperature_state_template: "{{ value_json.occupied_heating_setpoint }}"
      temperature_state_topic: zigbee2mqtt/Hallway Thermostat
      temperature_unit: C
      unique_id: hall_climate_zigbee2mqtt
```

ebdavison, 2024-09-04:
> This issue may be closed and unresolved in terms of dual-setpoint issues, but I have a quick and dirty workaround that allows the schedule card to work with this thermostat. Schedule card only sends [...]

I am not sure how this would work.  I imported the blueprint but it adds an automation.  I am not sure how to trigger it or how to use it now that it is "implemented".

Can someone provide some feedback on this?

aspiringsage119, 2026-08-21:
I realize this issue is abhorrently stale, but it is still very much alive as I just took the plunge to make the switch to Zigbee2MQTT and experienced the same as others who are lucky enough to own this thermostat. However, in my quest for a more elegant solution that keeps auto discovery in tact and doesn't create any intermediary entities, I discovered Zigbee2MQTT's brilliant [discovery override feature](https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#overriding-discovery-properties) offers up the following (short) resolution:

<details>
<summary>Expand for heat only</summary>

```yaml
devices:
  '0x000xxxxxxxxxxxxx':
    friendly_name: Living Room Thermostat
    thermostat_unit: celsius
    homeassistant:
      climate:
        modes:
          - 'off'
          - heat
        temperature_command_topic: zigbee2mqtt/Living Room Thermostat/set/occupied_heating_setpoint
        temperature_state_topic: zigbee2mqtt/Living Room Thermostat
        temperature_state_template: "{{ value_json.occupied_heating_setpoint }}"
```
_Adjust the friendly name and topic to match your configuration._
</details>
<details>
<summary>Expand for heat and cool</summary>

```yaml
devices:
  '0x000xxxxxxxxxxxxx':
    friendly_name: Living Room Thermostat
    thermostat_unit: celsius
    homeassistant:
      climate:
        modes:
          - 'off'
          - cool
          - heat
        temperature_command_topic: zigbee2mqtt/Living Room Thermostat/set
        temperature_command_template: >
          {"{{ 'occupied_cooling_setpoint' if is_state('climate.living_room_thermostat', 'cool')
              else 'occupied_heating_setpoint' }}": {{ value | float }}}
        temperature_state_topic: zigbee2mqtt/Living Room Thermostat
        temperature_state_template: >
          {{ value_json.occupied_cooling_setpoint if value_json.system_mode == 'cool'
              else value_json.occupied_heating_setpoint if value_json.system_mode == 'heat'
              else none }}
``` 
_Adjust the friendly name, entity, and topic lines to match your configuration. The entity created is referenced in the command template, but it shouldn't be a problem since the template is only executed when you make a change to the set point (which can't be triggered if the entity doesn't exist in the first place). I don't use emergency heating so I clamped the modes to ```off```, ```cool```, and ```heat```.  Remove the ```modes``` block and adjust the templates if you want the extra option to show._
</details>

You'll want to put this in your **Zigbee2MQTT** (not Home Assistant) ```configuration.yaml``` file after editing the IEEE address to match your thermostat's address. Reboot Zigbee2MQTT once you have made the changes. This is all that is needed if your thermostat hasn't been added to Home Assistant yet. Any new discovery of the device should work out-of-the-box.

If you already added your thermostat to Home Assistant, you may need to remove the **device** your _entity_ belongs to in Home Assistant, then reboot Home Assistant followed by another reboot of Zigbee2MQTT. Home Assistant should rediscover the device after the second reboot of Zigbee2MQTT.

Sorry for the resurrection of such an old thread on a closed issue, but it is what comes up when searching the dual set point issue for this thermostat so I wanted to leave this for others.