Koenkk/zigbee2mqtt #10623: [New device support]: LEDVANCE SMART+ A60 TW (Tunable White)

baptou60, 2022-01-06:
### Link

https://www.ledvance.fr/particuliers/produits/smart-home/lampes-smart-home/lampes-smart-home/lampes-smart-home-c198385

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0xf0d1b800001c0af7","nwkAddr":39435,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Unknown","modelId":"A60S TW","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"A60S TW","manufacturerName":"LEDVANCE","powerSource":0,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"02136550"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":2,"swBuildId":"02136550","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1641491470099,"useImplicitCheckin":true}

### Comments

Bonjour
J'ai essayé d'inclure l'ampoule LEDVANCE Smart+ A60 TW mais Zigbee2MQTT m'indique que l'quipement est non supporté.
Je n'ai pas trop compris le role des external Converter.
Peut être dois je essayer cela mais je ne sais pas comment faire?
C'est la 1ère fois que je suis confronté à un équipement non reconnu.
Je suis désolé si je fais une demande pour rien.
Je vous remercie par avance,
Bien cordialement,
Baptiste

Keeze, 2022-01-08:
Today I had the same issue with a P40S TW

Create a file called AC33898.js where the configuration.yaml resides with the folowing content (already altered for your lamp):

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['A60S TW'],
    model: 'AC33898',
    vendor: 'LEDVANCE',
    description: 'ZolderSpiegel',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    extend: extend.light_onoff_brightness_colortemp(),
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    // extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;`

then add the line

`external_converters:
  - AC33903.js`

to configuration.yaml. restart Z2M and it c(sh)ould work ;)

Koenkk, 2022-01-08:
@Keeze if you could provide the color temp range I can add this device.

https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

Keeze, 2022-01-08:
Hi Koen,

values:
Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}

This is for the P40S TW with model AC33903 (E14 small round bulb)

Cheers,
Kees

Keeze, 2022-01-08:
How can I add the right thumnail?
[image]

Koenkk, 2022-01-09:
@Keeze thanks added, thumbnail will be available in the frontend with the next release (1 February)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-09 by Koenkk

baptou60, 2022-01-09:
Bonjour,
Thank you for your help.
But, after create file AC33898.js et add the line in configuration.yaml, the light don't work.
I delete the light and new include but the issue is same.

I receive this log : 
`
 Received message from unsupported device with Zigbee model 'A60S TW' and manufacturer name 'LEDVANCE'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'A60S TW' and manufacturer name 'LEDVANCE'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'A60S TW' and manufacturer name 'LEDVANCE'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'A60S TW' and manufacturer name 'LEDVANCE'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning Received message from unsupported device with Zigbee model 'A60S TW' and manufacturer name 'LEDVANCE'
Warning Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
`

The vendor is again "Not Supported".

If you have an idea, I'll take it.
Thanck you for your time.
Baptiste

Keeze, 2022-01-09:
Is the modelnumber AC33898 printed on the base of the bulb?

baptou60, 2022-01-09:
Yes, on the base of the bulb, there is write this : 
220-240V ~ AC33898 . TW
2700~6500k . 806lm . E27
9w . 50/60Hz . 55mA

You are reason for the model, it is , AC33898 and the vendor, it is LEDVANCE.
I don't understant why I have many message for unsupported devide.

Keeze, 2022-01-09:
Did you copy the whole section (including **module.exports = definition;**)?

baptou60, 2022-01-09:
Yes, in my file AC33898.js I paste this : 

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['A60S TW'],
    model: 'AC33898',
    vendor: 'LEDVANCE',
    description: 'YourLamp',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    extend: extend.light_onoff_brightness_colortemp(),
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    // extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;`

and in my configuration.yaml : 
`advanced:
  log_level: debug
  external_converters:
    - AC33898.js
`

Keeze, 2022-01-09:
Weird, I see no reason why this should not work...

baptou60, 2022-01-09:
I received that : 

Debug Received Zigbee message from '0xf0d1b800001c0af7', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":34825552,"imageType":162,"manufacturerCode":4489}' from endpoint 1 with groupID null

Is this can help you?

baptou60, 2022-01-09:
It works !!
I put 2 spaces before external_converters when I paste.

```
advanced:
  log_level: debug
  external_converters:
    - AC33898.js
```

I delete this space and I restart the service Z2M and welcome ;)
Thanks for your help !!!!

Keeze, 2022-01-09:
> @Keeze thanks added, thumbnail will be available in the frontend with the next release (1 February)
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Did you add 2 devices?
E14 small bulb: AC33903 / P40S TW with the whitecolorvalues I pasted and
E27 bulb: AC33898 / A60S TW (don't know the whitecolorvalues, perhaps @baptou60 can publish them here?)

bluesailor45, 2022-04-02:
I installed the latest development branch of zigbee2mqtt and it worked with the LEDVANCE AC33898. However only the On/Off function and the colortemperatur function does work.  The brightness function does not work through the GUI. I could change the brightness with a native mqtt message to the device, but not with the Home Assistant GUI. When analysing the zigbee2mqtt.herdsman converter 'ledvance.js' the line 11: extend is different for the AC33898 than the others ( extend.light_onoff_brightness_colortemp({colorTempRange: [153, 370]}) ) while others are in the form extend.ledvance.light_onoff.......) Maybe this line should be changed?