Koenkk/zigbee2mqtt #10641: [Problem]: TuYa TS0601_thermostat gui programing mode not working work only from publish mqtt

shirou93, 2022-01-07:
### What happened?

TuYa TS0601_thermostat gui programing mode not working work only from publish mqtt

[image]

Clicking on the gui got nothing happening.

### What did you expect to happen?

I should change programing mode in web gui.

### How to reproduce it (minimal and precise)

Open web gui and ctry change something in programming_mode section.

### Zigbee2MQTT version

 1.22.2 

### Adapter firmware version

20211217

### Adapter

Slaesh's CC2652RB

### Debug log

No appear error in logs.

closed 2022-02-14 by github-actions[bot]

buddy31130, 2022-10-01:
same for me. The values do not show up in GUI but when modified, it is published in MQTT

garykrige, 2022-11-09:
Same for me