Koenkk/zigbee2mqtt #10649: npi ci fails on RaspberryPi Buster NodeJS 16

edsub, 2022-01-07:
### What happened?

Tried to execute the last install command "npm ci".
Instead of correct end, it failed with
```

pi@xxxx:/opt $ cd /opt/zigbee2mqtt
pi@xxx:/opt/zigbee2mqtt $ npm ci
npm ERR! invalid options argument

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-01-07T18_58_59_137Z-debug.log
```
I noticed some other issues with "npm ci",  but none showed this behaviour.
I use NodeJS 16 as this is accompanied by a supported npm ( npm that is included in Raspbian Buster seem too old)

### What did you expect to happen?

Someting like
node-pre-gyp info ok
added 383 packages in 111.613s

### How to reproduce it (minimal and precise)

Follow the instructions on https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing
all the way to "npm ci" under "Installing"

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211115

### Adapter

CC2531

### Debug log

[2022-01-07T18_58_59_137Z-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/7830822/2022-01-07T18_58_59_137Z-debug.log)

Koenkk, 2022-01-07:
- what is the output of `npm --version`
- does `npm install` work?

edsub, 2022-01-07:
```
pi@xxxx:/opt/zigbee2mqtt $ npm --version
8.1.2
```
```
pi@xxxxt:/opt/zigbee2mqtt $ npm install
npm WARN old lockfile
npm WARN old lockfile The npm-shrinkwrap.json file was created with an old version of npm,
npm WARN old lockfile so supplemental metadata must be fetched from the registry.
npm WARN old lockfile
npm WARN old lockfile This is a one-time fix-up, please be patient...
npm WARN old lockfile
npm ERR! invalid options argument

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-01-07T19_51_35_293Z-debug.log
```
[2022-01-07T19_51_35_293Z-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/7831014/2022-01-07T19_51_35_293Z-debug.log)

Koenkk, 2022-01-07:
looks that something is wrong with your node installation: https://stackoverflow.com/questions/70264158/npm-err-invalid-options-argument/70271570

doesnt look like something that can be fixed from z2m

edsub, 2022-01-07:
Thanks for pointing in a direction i have not found myself. Will try and report back!

edsub, 2022-01-08:
I think I solved the problem, thanks to your tip @Koenkk
I removed all of nodejs and npm manualy (google on remove npm and raspbery pi helped me with the commands).
Then there remained some synmlinks and folders in /opt/local/lib/ which I also removed.
Then I rebooted and installed nodejs 16 based on [these instructions](https://www.makersupplies.sg/blogs/tutorials/how-to-install-node-js-and-npm-on-the-raspberry-pi)
For the remaining steps I just followed [the installing instructions](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing)
When I started zifbee2mqtt I noticed some logging when I set a device into pairing mode, soo al god I guess.

One more remark: After I started zigbee2mqtt manually it went wrong by trying to start it as a deamon.
It apeared I had to change the location of npm in the script. Verfied the correct npm location via 
```
pi@xxxx:/opt/zigbee2mqtt $ which npm
/usr/local/bin/npm
```

closed 2022-01-08 by edsub