Koenkk/zigbee2mqtt #10651: [Problem]: Can update Electorama zzh stick (CC2652R) to 20211217 Z-Stack

ipeacocks, 2022-01-08:
### What happened?

I have Electorama zzh stick (CC2652R), working on summer firmware (20210708). I've used it in Zigbee2MQTT installation. Today I wished to update it to newer 20211217 Z-Stack but got `ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'`. 

Do I need to clean previous firmware?

### What did you expect to happen?

New firmware `CC2652R_coordinator_20211217` appears on stick.

### How to reproduce it (minimal and precise)

That's why I've done:
* turned it to BSL mode https://electrolama.com/radio-docs/bsl/#how-to-enter-bsl-mode
* backuped it with `zigpy_znp`
```
$ sudo python3 -m zigpy_znp.tools.nvram_read /dev/ttyUSB0 -o backup.json_bak
                                       
2022-01-08 02:17:17.083 work-laptop __main__ INFO OsalNvIds.HAS_CONFIGURED_ZSTACK3 = '55'
2022-01-08 02:17:17.117 work-laptop __main__ INFO OsalNvIds.EXTADDR = 'cb69b721004b1200'
2022-01-08 02:17:17.141 work-laptop __main__ INFO OsalNvIds.STARTUP_OPTION = '00'
2022-01-08 02:17:17.164 work-laptop __main__ INFO OsalNvIds.START_DELAY = '0a'
...
2022-01-08 02:17:28.254 work-laptop __main__ INFO APS_KEY_DATA_TABLE[0x0000] = '000000000000000000000000000000000000000000000000'
2022-01-08 02:17:28.274 work-laptop __main__ INFO APS_KEY_DATA_TABLE[0x0001] = '000000000000000000000000000000000000000000000000'
2022-01-08 02:17:28.294 work-laptop __main__ INFO APS_KEY_DATA_TABLE[0x0002] = '000000000000000000000000000000000000000000000000'
2022-01-08 02:17:28.322 work-laptop __main__ INFO NWK_SEC_MATERIAL_TABLE[0x0000] = '010000004f16e8073886eb34'
2022-01-08 02:17:28.338 work-laptop __main__ INFO NWK_SEC_MATERIAL_TABLE[0x0001] = '010000003c897653f0fa6e60'
2022-01-08 02:17:28.358 work-laptop __main__ INFO NWK_SEC_MATERIAL_TABLE[0x0002] = '23250000d8838f6847cb3173'
2022-01-08 02:17:28.373 work-laptop __main__ INFO NWK_SEC_MATERIAL_TABLE[0x0003] = '92351900cb69b721004b1200'
2022-01-08 02:17:28.389 work-laptop __main__ INFO NWK_SEC_MATERIAL_TABLE[0x0004] = '000000000000000000000000'
```
* Downloaded and unpacked `CC2652R_coordinator_20211217.zip` from here https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin
* Launched `cc2538-bsl.py` for identifying that stick is working well:
```
$ sudo ./cc2538-bsl.py

Opening port /dev/ttyUSB0, baud 500000
Connecting to target...
CC1350 PG2.0 (7x7mm): 352KB Flash, 20KB SRAM, CCFG.BL_CONFIG at 0x00057FD8
Primary IEEE Address: 00:12:4B:00:21:B7:69:CB
```
* And started to update it:
```
$ sudo ./cc2538-bsl.py -p /dev/ttyUSB0 -evw CC2652R_coordinator_20211217.hex

Opening port /dev/ttyUSB0, baud 500000
Reading data from CC2652R_coordinator_20211217.hex
Your firmware looks like an Intel Hex file
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```
What am I doing wrong? Do I need to erase it before flashing?

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20210708

### Adapter

Electrolama zig-a-zig-ah!

### Debug log

Added some in `How to reproduce it` section.

Roosted7, 2022-01-10:
Do you have any service installed that might interfere with the serial port? Such as modemmanager, or brltty?

I did exactly the same (same fws, same device, etc), but it only worked after disabling `brltty`

passy05, 2022-01-11:
Using Windows, I had no issue updating my CC2652R to the new firmware. 
I had issues with some devices that needed repairing (I upgraded ZM at the same time, so I don't really know what caused the issue). So far, everything is stable.

ipeacocks, 2022-01-12:
Thank you for help guys. I've updated it on Windows successfully. Probably that was a problem with BSL mode. I had to press button longer.

closed 2022-01-12 by ipeacocks