Koenkk/zigbee2mqtt #10652: [New device support]: Jasco ZB3102 - ZigBee plug-in smart dimmer

guillaumemolter, 2022-01-08:
### Link

https://byjasco.com/ge-zigbee-plug-smart-dimmer

### Database entry

{"id":21,"type":"Router","ieeeAddr":"0x0022a300001bb902","nwkAddr":56698,"manufId":4388,"manufName":"Jasco Products","powerSource":"Mains (single phase)","modelId":"35938","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,1794,2821],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"35938","manufacturerName":"Jasco Products","powerSource":1,"zclVersion":1,"appVersion":18,"stackVersion":80,"hwVersion":1}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00258d3b3a","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":260,"inClusterList":[0,3,2821],"outClusterList":[3,6,8],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":18,"stackVersion":80,"hwVersion":1,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":91512617},"lastSeen":1641606237861,"useImplicitCheckin":true}

### Comments

This is essentially the same plug as the GE 45852GE but sold with a different Brand and model #.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
      zigbeeModel: ['35938'],
      model: 'ZB3102',
      vendor: 'Jasco Products',
      description: 'ZigBee plug-in smart dimmer',
      extend: extend.light_onoff_brightness({noConfigure: true}),
      configure: async (device, coordinatorEndpoint, logger) => {
          await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
          const endpoint = device.getEndpoint(1);
          await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
          await reporting.onOff(endpoint);
      },
};

module.exports = definition;
```

guillaumemolter, 2022-01-08:
Note for the external converter above I essentially copy pasted the entry for the GE 45852GE from https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/ge.js and it works great.

Koenkk, 2022-01-08:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/3642

closed 2022-01-08 by Koenkk