Koenkk/zigbee2mqtt #10655: [New device support]: Unsupported smart lock turns out to be.. Ewelink light switch?

flatsiedatsie, 2022-01-08:
### Link

https://nl.aliexpress.com/item/1005002656862349.html

### Database entry

{"id":28,"type":"Router","ieeeAddr":"0x00124b0023XXXXXX","nwkAddr":58712,"manufId":0,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"ZB-SW01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genBasic":{"attributes":{"modelId":"ZB-SW01","manufacturerName":"eWeLink","powerSource":1,"zclVersion":1,"appVersion":4,"hwVersion":1,"dateCode":"20201026","swBuildId":"1.0.3"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"hwVersion":1,"dateCode":"20201026","swBuildId":"1.0.3","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1641654603025,"useImplicitCheckin":true}

### Comments

I bought this piece of electronics because I wanted to see if I could use it to power an electronic lock. That's what it seems designed to do. It has some great features for this. 

I was expecting to have to get it supported as a new device in Z2M, but once I plugged it in and tried pairing it.. it paired immediately.

It reports itself as an `Smart light switch - 1 gang` from eWelink. Interestingly, unlike the normal light switch (I'm assuming), it turns itself off again after one second. So it really is designed to quicky unlock a door lock through a momentary pulse.

I was wondering what would be a logical next step. In a way, the device is already supported... though not correctly.
- Is there a way to distinguish the two devices somehow and create a separate converter for this one? The it might send the correct exposes data, and from there have an optimal representation in the UI of my smart home controller.
- Could this device be shown as supported in the database somehow?

Koenkk, 2022-01-08:
@dkmatt0  could you provide the data/database.db entry of the `ZB-SW01`?

dkmatt0, 2022-01-08:
Sorry @Koenkk, I don't use Z2M anymore.

Koenkk, 2022-01-08:
@flatsiedatsie you can use the following `fingerprint` in your definition (instead of the `zigbeeModel`):

```js
        fingerprint: [
            {type: 'Router', manufacturerName: 'eWeLink', modelID: 'ZB-SW01', applicationVersion: 4, dateCode: '20201026',
                softwareBuildID: '1.0.3', endpoints: [
                    {ID: 1, profileID: 260, deviceID: 256, inputClusters: [0, 3, 4, 5, 6], outputClusters: [0]}]},
        ],
```

flatsiedatsie, 2022-01-08:
In which file should I edit that? I don't see anything like that in `from_zigbee.js`?

Koenkk, 2022-01-08:
@flatsiedatsie example: https://github.com/Koenkk/zigbee-herdsman-converters/blob/629bbb0e1714e4ac83dabd27dc3b603bbbf7505b/devices/gledopto.js#L67 (`fingerprint` can be used instead of the `zigbeeModel` to distinguish devices with the same `zigebeeModel`). The fingerprint identification always takes precedence over the zigbeeModel identification.

flatsiedatsie, 2022-01-08:
Ah, so you mean `ewelink.js`

flatsiedatsie, 2022-01-08:
```
    {
        fingerprint: [
                    {type: 'Router', manufacturerName: 'eWeLink', modelID: 'ZB-SW01', applicationVersion: 4, dateCode: '20201026',
                        softwareBuildID: '1.0.3', endpoints: [
                            {ID: 1, profileID: 260, deviceID: 256, inputClusters: [0, 3, 4, 5, 6], outputClusters: [0]}]},
                ],
        zigbeeModel: ['ZB-SW01'],
        model: 'ZB-SW01',
        vendor: 'eWeLink',
        description: 'Smart light switch - 1 gang',
        extend: extend.switch(),
        fromZigbee: [fz.on_off_skip_duplicate_transaction],
        onEvent: async (type, data, device) => {
            device.skipDefaultResponse = true;
        },
    },
```

results in an error:

```
2022-01-08 18:04:26.696 INFO   : zigbee2mqtt-adapter: SyntaxError: Invalid or unexpected token
2022-01-08 18:04:26.697 INFO   : zigbee2mqtt-adapter:     at Module._compile (internal/modules/cjs/loader.js:723:23)
2022-01-08 18:04:26.697 INFO   : zigbee2mqtt-adapter:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
2022-01-08 18:04:26.699 INFO   : zigbee2mqtt-adapter:     at Module.load (internal/modules/cjs/loader.js:653:32)
2022-01-08 18:04:26.699 INFO   : zigbee2mqtt-adapter:     at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
2022-01-08 18:04:26.700 INFO   : zigbee2mqtt-adapter:     at Function.Module._load (internal/modules/cjs/loader.js:585:3)
2022-01-08 18:04:26.701 INFO   : zigbee2mqtt-adapter:     at Module.require (internal/modules/cjs/loader.js:692:17)
2022-01-08 18:04:26.702 INFO   : zigbee2mqtt-adapter:     at require (internal/modules/cjs/helpers.js:25:18)
2022-01-08 18:04:26.703 INFO   : zigbee2mqtt-adapter:     at Object.<anonymous> (/home/pi/.webthings/data/zigbee2mqtt-adapter/zigbee2mqtt/node_modules/zigbee-herdsman-converters/index.js:128:30)
2022-01-08 18:04:26.704 INFO   : zigbee2mqtt-adapter:     at Module._compile (internal/modules/cjs/loader.js:778:30)
2022-01-08 18:04:26.705 INFO   : zigbee2mqtt-adapter:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
```

Koenkk, 2022-01-09:
This should work:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition =  {
    fingerprint: [
                {type: 'Router', manufacturerName: 'eWeLink', modelID: 'ZB-SW01', applicationVersion: 4, dateCode: '20201026',
                    softwareBuildID: '1.0.3', endpoints: [
                        {ID: 1, profileID: 260, deviceID: 256, inputClusters: [0, 3, 4, 5, 6], outputClusters: [0]}]},
            ],
    zigbeeModel: ['ZB-SW01'],
    model: 'ZB-SW01',
    vendor: 'eWeLink',
    description: 'Smart light switch - 1 gang',
    extend: extend.switch(),
    fromZigbee: [fz.on_off_skip_duplicate_transaction],
    onEvent: async (type, data, device) => {
        device.skipDefaultResponse = true;
    },
};

module.exports = definition;
```

flatsiedatsie, 2022-01-10:
That worked.

I used it on a second test system. This time I did not change `ewelink.js` but only added the external converter option in configuration.yaml, refering to this code in a file.

My main goal is to have to `state` be exposed as `lock`. That way it will be correctly displayed in the Webthings Gateway UI.

Is there a way to get the `state` exposed as `lock` without adding a new converter? I tried using `e.lock().withEndpoint('state')` but that didn't seem to work. Will I have to create a new converter?

I'm trying to look ahead to how I can create a pull request for this new device. For my understanding: will this fingerprint allow this device to be distinguished from it's button-cousin? Can I now create a separate converter for it below the switch version of the device? 

On a side note, it seems the earlier `invalid or unexpected token` error was the result of Mac OS adding a hidden file, `._ewelink.js`...

`2022-01-10 10:41:42.014 INFO   : zigbee2mqtt-adapter: z2m stderr:  (node:956) UnhandledPromiseRejectionWarning: /home/pi/.webthings/data/zigbee2mqtt-adapter/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/._ewelink.js:1
2022-01-10 10:41:42.015 INFO   : zigbee2mqtt-adapter: 
2022-01-10 10:41:42.015 INFO   : zigbee2mqtt-adapter: 
2022-01-10 10:41:42.016 INFO   : zigbee2mqtt-adapter: 
2022-01-10 10:41:42.017 INFO   : zigbee2mqtt-adapter: SyntaxError: Invalid or unexpected token
2022-01-10 10:41:42.017 INFO   : zigbee2mqtt-adapter:     at Module._compile (internal/modules/cjs/loader.js:723:23)
2022-01-10 10:41:42.018 INFO   : zigbee2mqtt-adapter:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
2022-01-10 10:41:42.024 INFO   : zigbee2mqtt-adapter:     at Module.load (internal/modules/cjs/loader.js:653:32)
...
2022-01-10 10:41:42.030 INFO   : zigbee2mqtt-adapter: z2m stderr:  (node:956) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
`
Maybe the code could be more robust to this? Then again, I understand that's overhead for an edge case.

And on a very unrelated note: what's the best way to tell Z2M to forget about a device? I'm currently sending `{"id": "' + zigbee_id + '"}` to `/bridge/request/device/remove` whenever the user removes a device, but it seems that Z2M is still remembering the device somehow. After a restart, it still shows up in the data (the big overview) that arrives at `zigbee2mqtt/bridge/devices`. This makes development a bit harder. My experiments have lead to a frankenstein device:
[image]

Koenkk, 2022-01-10:
@flatsiedatsie I'm not sure if this should be exposed as a lock since this device looks more like a switch, with the difference being that it turns of automatically again. 

> I'm trying to look ahead to how I can create a pull request for this new device. For my understanding: will this fingerprint allow this device to be distinguished from it's button-cousin? Can I now create a separate converter for it below the switch version of the device?

Yes, the fingerprint will distinguish it

> And on a very unrelated note: what's the best way to tell Z2M to forget about a device? I'm currently sending {"id": "' + zigbee_id + '"} to /bridge/request/device/remove whenever the user removes a device, but it seems that Z2M is still remembering the device somehow. This might be an issue with the Webthings Gateway, but I just wanted to make sure.

device/remove is correct, probably there is an issue in the z2m <-> webthings gateway integration

flatsiedatsie, 2022-01-10:
Doing `e.lock().withEndpoint('onOff')` seemed to do something: it seemed to copy the `state` to `lock` in the exposes data, and made that `lock` a read-only value. Perhaps the `lock` exposes option is read-only.

I also tried creating two full-blown converters in order to translate `state` to `lock`, but wasn't succesful. I saw lots of different errors, such as.

```
UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'includes' of undefined
 at converters.find (/home/pi/.webthings/data/zigbee2mqtt-adapter/zigbee2mqtt/lib/extension/publish.ts:210:60)
```

This is how far I got:

External file:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition =  {
    fingerprint: [
                {type: 'Router', manufacturerName: 'eWeLink', modelID: 'ZB-SW01', applicationVersion: 4, dateCode: '20201026',
                    softwareBuildID: '1.0.3', endpoints: [
                        {ID: 1, profileID: 260, deviceID: 256, inputClusters: [0, 3, 4, 5, 6], outputClusters: [0]}]},
            ],
    zigbeeModel: ['ZB-SW01'],
    model: 'ZB-SW01',
    vendor: 'eWeLink',
    description: 'Door lock module',
    //extend: extend.switch(),
    fromZigbee: [fz.on_off_lock_skip_duplicate_transaction, fz.ignore_basic_report],
    toZigbee: [tz.on_off_lock_skip_duplicate_transaction],
    onEvent: async (type, data, device) => {
        device.skipDefaultResponse = true;
    },
    exposes: [e.lock()]
    //exposes: [e.lock()]
    //exposes: [e.lock().withEndpoint('onOff')]

};

module.exports = definition;
```

fromZigbee.js:
```
    on_off_lock_skip_duplicate_transaction: {
        cluster: 'genOnOff',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            meta.logger.debug("on_off_lock_skip_duplicate_transaction: " + JSON.stringify(model) + "\n" + JSON.stringify(msg) + "\n" + JSON.stringify(options));
            // Device sends multiple messages with the same transactionSequenceNumber,
            // prevent that multiple messages get send.
            // https://github.com/Koenkk/zigbee2mqtt/issues/3687
            if (msg.data.hasOwnProperty('onOff') && !hasAlreadyProcessedMessage(msg)) {
                //const property = postfixWithEndpointName('lock', msg, model);
                ///return {[property]: msg.data['onOff'] === 1 ? 'LOCK' : 'UNLOCK'};
                return {'lock': msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
            }
        },
    },
```

toZigbee.js
```
    on_off_lock_skip_duplicate_transaction: {
        key: ['lock'],
        convertSet: async (entity, key, value, meta) => {
            
            //meta.logger.debug("HIERO SENDING on_off_lock_skip_duplicate_transaction: ", key, value, meta.message);
            meta.logger.debug("HIERO on_off_lock_skip_duplicate_transaction: " + JSON.stringify(key) + "\n" + JSON.stringify(value));
            
            //const state = meta.message.hasOwnProperty('lock') ? meta.message.state.toLowerCase() : null;
            //const state = meta.message.hasOwnProperty('lock') ? meta.message.lock.toLowerCase() : null;
            //utils.validateValue(state, ['toggle', 'off', 'on']);

            //meta.logger.debug("SENDING state: ", state);
            //return {state: {state: state.toUpperCase()}};
            return {state: {state: 'ON'}};

        },
        convertGet: async (entity, key, meta) => {
            await entity.read('genOnOff', ['onOff']);
        },
    },
```

flatsiedatsie, 2022-01-10:
> I'm not sure if this should be exposed as a lock since this device looks more like a switch

I'm not sure what the difference is, but I am fine with keeping it as it is too.

closed 2022-02-18 by github-actions[bot]