Koenkk/zigbee2mqtt #10656: [Problem]: Device discovered as another

magostinelli, 2022-01-08:
### What happened?

My led lamp stopped working, so i remove it and rejoined again to my zigbee network.

After join the log says:
Device '0x00158d00045eeefc' is supported, identified as: Xiaomi MiJia temperature & humidity sensor (WSDCGQ01LM)
But my device is this one:
https://www.zigbee2mqtt.io/devices/ZNLDP12LM.html

It's possibile to change the device type manually?

Regards

### What did you expect to happen?

Discovered as https://www.zigbee2mqtt.io/devices/ZNLDP12LM.html

### How to reproduce it (minimal and precise)

Join the device to the network

### Zigbee2MQTT version

1.22.2 

### Adapter firmware version

20211115

### Adapter

cc2531

### Debug log

Device '0x00158d00045eeefc' is supported, identified as: Xiaomi MiJia temperature & humidity sensor (WSDCGQ01LM)

Koenkk, 2022-01-08:
Found the bug. To fix it:
- Update z2m to the latest dev branch
- force remove the LED bulb via the frontend
- repair the LED bulb

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

magostinelli, 2022-01-09:
I solved my problem, but I don't know how i do it.

I removed (force) che led lamp and powered it off for an hour. I restarted my home assistant, disconnected and reconnected the zigbee dongle.

closed 2022-01-09 by magostinelli