Koenkk/zigbee2mqtt #10660: [Problem]:

joselcaguilar, 2022-01-08:
### What happened?

Error `Failed to find endpoint which support OTA cluster` showed when I click on check updates for Xiaomi QBKG04LM using OTA.

### What did you expect to happen?

Xiaomi endpoint linked as expected in order to upgrade the firmware successfully.

### How to reproduce it (minimal and precise)

1. Go to OTA using the frontend UI
2. Click on check updates for the device Xiaomi QBKG04LM

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

CC1352P2_CC2652P_launchpad_coordinator

### Debug log

info  2022-01-08 17:17:14: Checking if update available for 'Terraza'
info  2022-01-08 17:17:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Terraza'","meta":{"device":"Terraza","status":"checking_if_available"},"type":"ota_update"}'
info  2022-01-08 17:17:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Terraza' (Failed to find endpoint which support OTA cluster)","meta":{"device":"Terraza","status":"check_failed"},"type":"ota_update"}'
info  2022-01-08 17:17:14: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Terraza"},"error":"Failed to check if update available for 'Terraza' (Failed to find endpoint which support OTA cluster)","status":"error","transaction":"ii8qw-1"}'
error 2022-01-08 17:17:14: Failed to check if update available for 'Terraza' (Failed to find endpoint which support OTA cluster)

joselcaguilar, 2022-02-09:
Follow up 😊, still present on 1.23.0

closed 2022-03-19 by github-actions[bot]

Mourticarius, 2022-10-08:
Same problem here with Xiaomi VOCKQJK11LM
Device : SONOFF ZigBee 3.0 USB Dongle Plus EFR32MG21
Cordinator type : EZSP v8
Zigbee2Mqtt version : 1.28.0

d96moe, 2022-10-19:
Same here: 
Xiaomi SSM-U01

d96moe, 2022-10-27:
Should this issue be re-opened?

AndrewHaluza, 2022-11-28:
same issue with MCCGQ14LM

protyposis, 2022-12-09:
Checked all of my Aqara devices:

3x MCCGQ11LM contact sensor
1x GZCGQ01LM light sensor
4x RTCGQ11LM motion sensor
1x RTCGQ14LM motion sensor
1x WXCJKG13LM switch
1x SRTS-A01 TRV
5x WSDCGQ11LM weather sensor

Only the motion sensors have a `genOta`, all others seem to be affected by this issue.

Koenkk, 2022-12-09:
@protyposis some of the device you lister are quite old and I know earlier Xiaomi products did not support this.

@AndrewHaluza MCCGQ14LM seems quite new, can you provide the herdsman debug log of:
- Force remove device through z2m frontend
- Restart z2m
- re-pair device.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

WernerRH21, 2023-01-08:
Hi there,
same problem with one Xiaomi Aqara RTCGQ14LM device.
One has ota cluster, the other not. Nothing helps: Batteries out, repair, press button

segdy, 2023-08-12:
I’m having the exact same issue for a new [ZNCLBL01LM](https://www.zigbee2mqtt.io/devices/ZNCLBL01LM.html#xiaomi-znclbl01lm)

also with Sonoff Zigbee dongle

really no fix?

zebrahosting, 2023-12-15:
Same here with Aqara presence sensor and Tubez co-ordinator. It shows as [Xiaomi](https://www.zigbee2mqtt.io/supported-devices/#v=Xiaomi)
Model
[RTCGQ13LM](https://www.zigbee2mqtt.io/devices/RTCGQ13LM.html#xiaomi-rtcgq13lm)

I do have another one connected to a SB6 co-ordinator (both POE) and from there I could update one.

CAVACO-PT, 2024-01-11:
[image]

Same Problem with Xiaomi Presence detector FP1 - RTCZCGQ11LM

ConfigMgrPro, 2024-01-12:
I have the same issue on (cheap) Tuya power monitoring devices:
https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1

MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Powermonitor koelkast"},"error":"Failed to check if update available for 'Powermonitor' (Failed to find endpoint which support OTA cluster)","status":"error","transaction":"bqwtv-1"}'

Also, i noticed both on ZHA and on Z2M that i don't see an OTA cluster for these devices (i have 6 of them).

Koenkk, 2024-01-13:
Pushed a fix for all Xiaomi devices!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

segdy, 2024-01-15:
I've just tried the dev branch with my [ZNCLBL01LM](https://www.zigbee2mqtt.io/devices/ZNCLBL01LM.html#xiaomi-znclbl01lm). Unfortunately no change :-(

Anyone else had success?

ConfigMgrPro, 2024-01-15:
> I've just tried the dev branch with my [ZNCLBL01LM](https://www.zigbee2mqtt.io/devices/ZNCLBL01LM.html#xiaomi-znclbl01lm). Unfortunately no change :-(
> 
> Anyone else had success?

Same here, no OTA cluster for Tuya device. (Z2M 1.35.1-dev commit: [2bd7df8](https://github.com/Koenkk/zigbee2mqtt/commit/2bd7df8))

thatslolo, 2024-01-18:
Having the same issue with the [Aqara T1 light intensity sensor](https://www.zigbee2mqtt.io/devices/GZCGQ11LM.html#xiaomi-gzcgq11lm).

Koenkk, 2024-01-18:
Can you try to reconfigure the device via the frontend? (yellow refresh icon) and see if that fixes the issue.

thatslolo, 2024-01-19:
Already tried.
Paired it with this version for the first time. 
another one which I’m using for a month already, has the same issue.

Koenkk, 2024-01-19:
Could you provide your data/database.db entry of the device and the debug log when configuring the device?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

thatslolo, 2024-01-20:
`{"id":93,"type":"EndDevice","ieeeAddr":"0x54ef441000a5f70e","nwkAddr":63050,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sen_ill.agl01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[0,1024,3,1,25],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.sen_ill.agl01","appVersion":27}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":19396}},"aqaraOpple":{"attributes":{"0":5,"247":{"type":"Buffer","data":[1,33,145,12,3,40,25,4,33,168,19,5,33,18,0,6,36,13,0,0,0,0,8,33,27,1,10,33,65,76,12,32,1,100,35,87,0,0,0]}}},"genPowerCfg":{"attributes":{"batteryVoltage":32}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":27,"stackVersion":2,"hwVersion":1,"dateCode":"20200312","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1705766694148}`

```
Debug 2024-01-20 17:19:19Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"0x54ef441000a5f70e","transaction":"jo0q6-3"}'
Info 2024-01-20 17:19:19Configuring '0x54ef441000a5f70e'
Debug 2024-01-20 17:19:22Received Zigbee message from '0x54ef441000a5f70e', type 'attributeReport', cluster 'genBasic', data '{"modelId":"lumi.sen_ill.agl01"}' from endpoint 1 with groupID 0
Debug 2024-01-20 17:19:22Received Zigbee message from '0x54ef441000a5f70e', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,145,12,3,40,25,4,33,168,19,5,33,18,0,6,36,4,0,0,0,0,8,33,27,1,10,33,65,76,12,32,1,100,35,85,0,0,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
Debug 2024-01-20 17:19:22GZCGQ11LM: Processed buffer into data {"1":3217,"3":25,"4":5032,"5":18,"6":4,"8":283,"10":19521,"12":1,"100":85}
Debug 2024-01-20 17:19:22GZCGQ11LM: unknown key 6 with value 4
Debug 2024-01-20 17:19:22GZCGQ11LM: Processed data into payload {"voltage":3217,"battery":100,"device_temperature":25,"power_outage_count":17,"state":"OFF"}
Debug 2024-01-20 17:19:22GZCGQ11LM: Processed data into payload {"voltage":3217,"battery":100,"device_temperature":25,"power_outage_count":17,"state":"OFF"}
Info 2024-01-20 17:19:22MQTT publish: topic 'zigbee2mqtt/0x54ef441000a5f70e', payload '{"battery":100,"detection_period":5,"device_temperature":25,"illuminance":19396,"illuminance_lux":87,"linkquality":116,"power_outage_count":17,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3217}'
Debug 2024-01-20 17:19:24Received Zigbee message from '0x54ef441000a5f70e', type 'readResponse', cluster 'aqaraOpple', data '{"0":5}' from endpoint 1 with groupID 0
Debug 2024-01-20 17:19:24GZCGQ11LM: Processed data into payload {"detection_period":5}
Info 2024-01-20 17:19:24MQTT publish: topic 'zigbee2mqtt/0x54ef441000a5f70e', payload '{"battery":100,"detection_period":5,"device_temperature":25,"illuminance":19396,"illuminance_lux":87,"linkquality":116,"power_outage_count":17,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3217}'
Info 2024-01-20 17:19:24Successfully configured '0x54ef441000a5f70e'
```

Koenkk, 2024-01-21:
Found the issue, update to latest-dev, reconfigure and update it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

segdy, 2024-01-22:
@Koenkk Does this apply to just one specific device or to ZNCLB01LM as well?

Switches to the dev version, reconfigured but sadly still no change :-( 

Thank you for your multiple efforts & attempts on this!

thatslolo, 2024-01-22:
The new converters are not yet in the dev release.
https://github.com/Koenkk/zigbee2mqtt/pull/20933

thatslolo, 2024-01-22:
tested with the 18.18.0 converter. Issue seems resolved now. Thanks!

segdy, 2024-01-23:
Unfortunately still not for me and my ZNCLB01LM :-( 

@Koenkk is this supposed to work for ZNCLB01LM too or just the T1 light intensity sensor?

Koenkk, 2024-01-24:
`ZNCLB01LM` is not supported, but it should be for `GZCGQ11LM`

lutz108, 2024-09-27:
> I have the same issue on (cheap) Tuya power monitoring devices: https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1
>
> MQTT publish: topic [...]

you found a solution to that? Updated z2m to current version, reconfigured and interviewed all again. I either get

`z2m: Failed to ping 'Plug 7 (Stehlampe, 20 A)' (attempt 1/1, ZCL command 0x70b3d52b6005cdda/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (no response received (75) waiting for response TIMEOUT))`

or

`Failed to check if update available for 'Plug 7 (Stehlampe, 16 A)' (Device didn't respond to OTA request)`

:/ Tried using and not using https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#alternative-firmware

baerakafubu, 2024-11-27:
@Koenkk Same problem here, with Third Reality Temp and Humidity Sensor [3RTHS0224Z](https://www.zigbee2mqtt.io/devices/3RTHS0224Z.html). One sensor can update, but the other one get the mentioned error above. Tried to re-configure and re-pair but didn't change the issue