Koenkk/zigbee2mqtt #10662: Error while starting zigbee-herdsman suddenly started after 1 year problem free running Domoticz

McGyver64, 2022-01-08:
### What happened?

Domoticz stopped communicating with all the Zigbee devices as of 2022-01-08 04:30:36

The error from the Domoticz log is:

2022-01-08 19:57:34.076 Error: Zigbee2MQTT: (Zigbee2MQTT) Failed to connect to: 172.16.10.249:1883, Description: Connection refused
2022-01-08 19:57:34.076 Error: Zigbee2MQTT: (Zigbee2MQTT) Disconnected from MQTT Server: 172.16.10.249:1883

No Zigbee device is connected and/or can be read from Domoticz

When I run nom start I get:

Zigbee2MQTT:info  2022-01-08 19:45:40: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-01-08.19-45-39' filename: log.txt
Zigbee2MQTT:info  2022-01-08 19:45:40: Starting Zigbee2MQTT version 1.18.2 (commit #abd8a092)
Zigbee2MQTT:info  2022-01-08 19:45:40: Starting zigbee-herdsman (0.13.88)
Zigbee2MQTT:error 2022-01-08 19:45:41: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-01-08 19:45:41: Failed to start zigbee
Zigbee2MQTT:error 2022-01-08 19:45:41: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2022-01-08 19:45:41: Exiting...
Zigbee2MQTT:error 2022-01-08 19:45:41: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.18.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-01-08T18_45_41_200Z-debug.log

### What did you expect to happen?

On 04:30 there is an automated script running update and upgrade. I think somewhere in the update something changed with rights or so, but my knowledge of Linux is not sufficient. 

### How to reproduce it (minimal and precise)

After a reboot, the connection is still not enabled

2022-01-08 20:05:24.117 Error: Zigbee2MQTT: (Zigbee2MQTT) Failed to connect to: 172.16.10.249:1883, Description: Connection refused
2022-01-08 20:05:24.117 Error: Zigbee2MQTT: (Zigbee2MQTT) Disconnected from MQTT Server: 172.16.10.249:1883

### Zigbee2MQTT version

1.18.2

### Adapter firmware version

cannot read 

### Adapter

usb-Silicon_Labs_slae.sh_cc2652rb_stick

### Debug log

pi@Domoticz:/opt/zigbee2mqtt/data/log/2022-01-08.19-47-02 $ sudo nano log.txt

  GNU nano 3.2                                                                  log.txt                                                                             

info  2022-01-08 19:47:03: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-01-08.19-47-02' filename: log.txt
info  2022-01-08 19:47:04: Starting Zigbee2MQTT version 1.18.2 (commit #abd8a092)
info  2022-01-08 19:47:04: Starting zigbee-herdsman (0.13.88)
info  2022-01-08 19:47:15: zigbee-herdsman started
info  2022-01-08 19:47:15: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210120,"transportrev":2},"type":$
info  2022-01-08 19:47:15: Currently 34 devices are joined:
info  2022-01-08 19:47:15: WCD Tafel (0x842e14fffe3666ef): TS0121_plug - TuYa 10A UK or 16A EU smart plug (Router)
info  2022-01-08 19:47:15: CV4 (0x5c0272fffec9da73): TS0601_thermostat - TuYa Radiator valve with thermostat (EndDevice)
info  2022-01-08 19:47:15: WCD Overloop (0x588e81fffed393ad): TS0121_plug - TuYa 10A UK or 16A EU smart plug (Router)
info  2022-01-08 19:47:15: WCD MK (0x588e81fffed39f9b): TS0121_plug - TuYa 10A UK or 16A EU smart plug (Router)
info  2022-01-08 19:47:15: WCD TV (0x588e81fffeff5cbb): TS0121_plug - TuYa 10A UK or 16A EU smart plug (Router)
info  2022-01-08 19:47:15: WCD Garage (0x842e14fffe626176): HG06337 - Lidl Silvercrest smart plug (EU, CH, FR, BS, DK) (Router)
info  2022-01-08 19:47:15: Lidl Motion Garage (0xbc33acfffe5e4ee2): HG06335 - Lidl Silvercrest smart motion sensor (EndDevice)
info  2022-01-08 19:47:15: S7 Bijkeuken (0x00158d0004a008fb): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-01-08 19:47:15: S8 Garage (0x00158d000485254c): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-01-08 19:47:15: S6 Zuid (0x00158d000485259f): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-01-08 19:47:15: CV1 (0x60a423fffeab24a7): TS0601_thermostat - TuYa Radiator valve with thermostat (EndDevice)
info  2022-01-08 19:47:15: CV2 (0x5c0272fffedfbca0): TS0601_thermostat - TuYa Radiator valve with thermostat (EndDevice)
info  2022-01-08 19:47:15: CV3 (0xbc33acfffe458434): TS0601_thermostat - TuYa Radiator valve with thermostat (EndDevice)
info  2022-01-08 19:47:15: CV5 (0x5c0272fffedab20a): TS0601_thermostat - TuYa Radiator valve with thermostat (EndDevice)
info  2022-01-08 19:47:15: S5 Badkamer (0x00158d00048525dc): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-01-08 19:47:15: S4 MK (0x00158d00054bdd02): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-01-08 19:47:15: S3 Oost (0x00158d00054be2ad): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-01-08 19:47:15: S2 West (0x00158d00054b6c1c): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-01-08 19:47:15: S1 Zolder (0x00158d00054be26f): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-01-08 19:47:15: CV6 (0x847127fffe0eba58): TS0601_thermostat - TuYa Radiator valve with thermostat (EndDevice)
info  2022-01-08 19:47:15: CS2 (0x00158d0006c3d07d): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-01-08 19:47:15: CS1 (0x00158d0006bf439e): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-01-08 19:47:15: 0x00158d0006b8aa55 (0x00158d0006b8aa55): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-01-08 19:47:15: 0x00124b00234d5f96 (0x00124b00234d5f96): ZB-RGBCW - Lonsonho Zigbee 3.0 LED-bulb, RGBW LED (Router)
info  2022-01-08 19:47:15: Tuya Motion (0x00158d0005bcdba8): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
info  2022-01-08 19:47:15: 0x00158d0006c418ea (0x00158d0006c418ea): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-01-08 19:47:15: CS3 (0x00158d0006b88cd0): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-01-08 19:47:15: CS4 (0x00158d0006c3fee1): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)

Koenkk, 2022-01-09:
> Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'

https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-resource-temporarily-unavailable-cannot-lock-port

McGyver64, 2022-01-09:
I checked all the things listed in the link all dmesg return no error, Power is ok, Extension Cable Ok, Connection to USB device is Ok, raised PAN-ID from 6757 to 6758 (reboot) but still no respons from CC2526

This configuration has been running for 1 year without any problem, and suddenly in the night (+- 03:00 am) the devices stoped communication with domoticz. There was no change or intervention from outside at that moment.

McGyver64, 2022-01-09:
I started the TI FlashProgrammer 2 tool, but it didn't detect the CC2652 device. I had to select it manually and then I was able tho flash the latest firmware
 
[image]

[image]

When I use the UNIFlash tool, the CC2652 is also not recognized

Is the CC2652 Broken?

McGyver64, 2022-01-18:
Did a new fresh install same problem other error. I am opening new problem case

closed 2022-01-18 by McGyver64