Koenkk/zigbee2mqtt #10666: [Problem]: E1743 Update stucks at 100 %

ahemwe, 2022-01-08:
### What happened?

I've paired a brand new Ikea E1743 switch.
```
Firmware Date: 20190723
Firmware Version: 2.2.010
```

Then, I've started OTA update via front end while putting the device less then 10 cm away from the adapter. Battery has been 100 %.

The update took more than two hours. Eventually, I got "100 %" and "6 minutes remaining" for more than another two hours.

Then, I've tried to restart Zigbee2MQTT which lead to:
```

{
    "battery": 87,
    "brightness": 115,
    "last_seen": "2022-01-08T23:00:22+01:00",
    "linkquality": 255,
    "update": {
        "state": "updating"
    },
    "update_available": false
}
```

It might be worth to mention, that I've had enabled Ikea Tradfri OTA test url some month ago. After issues mentioned [here](https://github.com/Koenkk/zigbee2mqtt/issues/7886) I've disabled that option.
Is there any local cache in the file system with possibly corrupted fw?

How about removing battery of the remote? Does that brick the hardware?

### What did you expect to happen?

Normal firmware update.

### How to reproduce it (minimal and precise)

Pairing.
Frontend -> OTA -> Update Firmware

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26680700

### Adapter

 ConBee2/RaspBee2

### Debug log

Debug: 
`Received Zigbee message from 'Remote_Licht', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null`

Info:
`MQTT publish: topic 'zigbee2mqtt/Remote_Licht', payload '{"action":"on","battery":87,"brightness":115,"elapsed":7078,"last_seen":"2022-01-08T23:30:35+01:00","linkquality":63,"update":{"state":"updating"},"update_available":false}'`

ahemwe, 2022-01-09:
After night, the remote was still in update state. Eventually, I've removed the device, removed battery and tried to reconnect.
It took several tries but worked well in the end.

Then, I did the same for an E1524/E1810 Ikea Tradfri Remote.

Here, the same happend, but now, I've better logs as I've activated debug logs beforehand;
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7835780/log.txt)

Interestingly, the update took place between 15:07 and 17:20:59, but afterwards I still get the "updating" state.

```
info  2022-01-09 17:20:59: MQTT publish: topic 'zigbee2mqtt/My_Remote', payload '{"battery":100,"last_seen":"2022-01-09T17:20:59+01:00","linkquality":255,"update":{"progress":99.67,"remaining":26,"state":"updating"},"update_available":false}'
debug 2022-01-09 17:20:59: Got upgrade end request for '0x000d6ffffe1f2624': {"status":0,"manufacturerCode":4476,"imageType":4545,"fileVersion":587728433}
debug 2022-01-09 17:20:59: Update succeeded, waiting for device to restart
info  2022-01-09 17:21:14: MQTT publish: topic 'zigbee2mqtt/My_Remote', payload '{"battery":100,"last_seen":"2022-01-09T17:21:14+01:00","linkquality":255,"update":{"progress":99.67,"remaining":26,"state":"updating"},"update_available":false}'
debug 2022-01-09 17:21:14: Device 'My_Remote' announced itself
info  2022-01-09 17:21:14: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"My_Remote","ieee_address":"0x000d6ffffe1f2624"},"type":"device_announce"}'
info  2022-01-09 17:21:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"My_Remote"},"type":"device_announced"}'
debug 2022-01-09 17:21:16: Retrieving state of 'My_Remote' after reconnect
debug 2022-01-09 17:22:06: Received Zigbee message from 'My_Remote', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587728433,"imageType":4545,"manufacturerCode":4476}' from endpoint 1 with groupID null
info  2022-01-09 17:22:06: MQTT publish: topic 'zigbee2mqtt/My_Remote', payload '{"battery":100,"last_seen":"2022-01-09T17:22:06+01:00","linkquality":255,"update":{"progress":99.67,"remaining":26,"state":"updating"},"update_available":false}'

```

Is there any issue with resetting the update state?

Thanks for your support.

Koenkk, 2022-01-09:
> Is there any issue with resetting the update state?

This is cached in the data/state.json, you can stop z2m and remove it from there.

ahemwe, 2022-01-09:
> > Is there any issue with resetting the update state?
> 
> This is cached in the data/state.json, you can stop z2m and remove it from there.

Can't find anything there, for that IEEE address I only see:
```

    "0x000d6ffffe1f2624": {
        "battery": 87
    }
```

There is also no update-section with other content than "idle", while the frontend still shows updating...

ahemwe, 2022-01-10:
After ~24 h I just wanted to change `state.json `as mentioned above @Roosted7 as I saw that the updating-state disappeared meanwhile.
Looks like it simply takes very long.

I still think that this is a (minor) bug. May we keep this issue open? @Koenkk

Koenkk, 2022-01-11:
Found the issue, will check for a proper fix (discussion: https://github.com/Koenkk/zigbee-herdsman/commit/3ee4af3c13a2a4d186c6d79d451c1c314a4e3d0f#r63645018)

satter, 2022-01-12:
I had a troublesome [update](https://github.com/Koenkk/zigbee2mqtt/discussions/2921#discussioncomment-1938434) of my E1810 remote, with stuck 'updating' status

Do not know if related, but when I check updates (from web interface) for remote which is offline, Z2M service will crash:

```
Jan 12 19:02:32 homie1 npm[79593]: Zigbee2MQTT:debug 2022-01-12 19:02:32: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"ikea_remote1","transaction":"j6cbw-7"}'
Jan 12 19:02:32 homie1 npm[79593]: Zigbee2MQTT:info  2022-01-12 19:02:32: Checking if update available for 'ikea_remote1'
Jan 12 19:02:32 homie1 npm[79593]: Zigbee2MQTT:debug 2022-01-12 19:02:32: Check if update available for '0x000d6ffffe409baa' (TRADFRI remote control)
Jan 12 19:02:32 homie1 npm[79593]: Zigbee2MQTT:debug 2022-01-12 19:02:32: Using endpoint '1'
Jan 12 19:03:02 homie1 npm[79593]: Error: Timeout - 21387 - 1 - null - 25 - 1 after 30000ms
Jan 12 19:03:02 homie1 npm[79593]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
Jan 12 19:03:02 homie1 npm[79593]:     at listOnTimeout (node:internal/timers:557:17)
Jan 12 19:03:02 homie1 npm[79593]:     at processTimers (node:internal/timers:500:7)
Jan 12 19:03:02 homie1 systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Jan 12 19:03:02 homie1 systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 12 19:03:03 homie1 systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 1.
Jan 12 19:03:03 homie1 systemd[1]: Stopped zigbee2mqtt.
Jan 12 19:03:03 homie1 systemd[1]: Started zigbee2mqtt.
Jan 12 19:03:03 homie1 npm[79632]: > zigbee2mqtt@1.22.2 start
Jan 12 19:03:03 homie1 npm[79632]: > node index.js
```
```
Z2m 1.22.2
node v16.13.1
sonoff zStack3x0 20211217
```

Koenkk, 2022-01-17:
This issue should be fixed now in the latest dev branch. To verify this I've updated 1 IKEA switch, 1 IKEA remote and 1 Hue light, all also show the correct version afterwards.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-17 by Koenkk

GillesVermeulen, 2022-01-25:
Should the dev-version with this fix be available in the "edge" branch of the Home Assistant addon right now?
Because I just tested the edge version out in home assistant, and sadly I'm still not able to update the on/off switch.

I had the update stuck at 100% issue, and I followed your instructions  ("stop z2m and remove it manually from data/state.json"), but the device is now stuck on firmware version 2.2.010, and if I try to update it again, it says "Failed to check if update available for 'On/off switch sofa' (Device didn't respond to OTA request)".

I removed the battery multiple times and restarted Zigbee2MQTT.

DesertDwellerEv, 2022-04-17:
I know it's probably too late since the fix is in dev right now and this issue has been closed.
However, I stumbled on how to update E1812 without issue.

1. Press switch in back 4 times to get into pairing mode
2. Watch the logs for device_announce
3. Wait for glowing red led to stop
4. OTA tab and press Update device firmware
5. Rinse and repeat for other devices