Koenkk/zigbee2mqtt #10673: [Problem]: Eglo/Awox Remote control detected as light bulb

fariouche, 2022-01-09:
### What happened?

I tried to pair the RCU from Eglo (https://media.adeo.com/marketplace/LMFR/83330031/f2e1c7aa-eb2e-4b4d-94b9-1d7258832444.png?width=1920)

Detected as a light bulb instead of a switch

### What did you expect to happen?

Detect and working as a remote controller

### How to reproduce it (minimal and precise)

Just pair the rcu (after long pressing the power on button)

### Zigbee2MQTT version

1.22.2-dev commit: 24d9065d

### Adapter firmware version

20210708

### Adapter

zzh!

### Debug log

info  2022-01-09 13:09:41: Zigbee: allowing new devices to join.
info  2022-01-09 13:09:41: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"bragw-5"}'
info  2022-01-09 13:10:00: Device '0xa4c138f7acbcc55c' joined
info  2022-01-09 13:10:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138f7acbcc55c","ieee_address":"0xa4c138f7acbcc55c"},"type":"device_joined"}'
info  2022-01-09 13:10:00: Starting interview of '0xa4c138f7acbcc55c'
info  2022-01-09 13:10:00: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138f7acbcc55c","ieee_address":"0xa4c138f7acbcc55c","status":"started"},"type":"device_interview"}'
info  2022-01-09 13:10:01: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138f7acbcc55c","ieee_address":"0xa4c138f7acbcc55c"},"type":"device_announce"}'
info  2022-01-09 13:11:08: Successfully interviewed '0xa4c138f7acbcc55c', device has successfully been paired
info  2022-01-09 13:11:08: Device '0xa4c138f7acbcc55c' is supported, identified as: AwoX LED white (33951/33948)
info  2022-01-09 13:11:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"LED white","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"33951/33948","options":[{"access":2,"description":"Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to `0` (no transition).","name":"transition","property":"transition","type":"numeric","value_min":0}],"supports_ota":false,"vendor":"AwoX"},"friendly_name":"0xa4c138f7acbcc55c","ieee_address":"0xa4c138f7acbcc55c","status":"successful","supported":true},"type":"device_interview"}'
info  2022-01-09 13:11:08: Configuring '0xa4c138f7acbcc55c'
info  2022-01-09 13:11:08: MQTT publish: topic 'homeassistant/light/0xa4c138f7acbcc55c/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/0xa4c138f7acbcc55c/set","device":{"identifiers":["zigbee2mqtt_0xa4c138f7acbcc55c"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c138f7acbcc55c","sw_version":"0122052017"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"name":"0xa4c138f7acbcc55c","schema":"json","state_topic":"zigbee2mqtt/0xa4c138f7acbcc55c","unique_id":"0xa4c138f7acbcc55c_light_zigbee2mqtt"}'
info  2022-01-09 13:11:08: MQTT publish: topic 'homeassistant/sensor/0xa4c138f7acbcc55c/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138f7acbcc55c"],"manufacturer":"AwoX","model":"LED white (33951/33948)","name":"0xa4c138f7acbcc55c","sw_version":"0122052017"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0xa4c138f7acbcc55c linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138f7acbcc55c","unique_id":"0xa4c138f7acbcc55c_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2022-01-09 13:11:08: Successfully configured '0xa4c138f7acbcc55c'
info  2022-01-09 13:12:20: Zigbee: disabling joining new devices.

The relevant database.db line is:
{"id":19,"type":"EndDevice","ieeeAddr":"0xa4c138f7acbcc55c","nwkAddr":5189,"manufId":4417,"manufName":"AwoX","powerSource":"Battery","modelId":"TLSR82xx","epList":[1,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":2048,"inClusterList":[0,3,4,4096],"outClusterList":[0,3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":3,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":2048,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1641733868347,"useImplicitCheckin":true}

fariouche, 2022-01-09:
Here is a full log if this helps:

```
Zigbee2MQTT:info  2022-01-09 13:10:56: Device '0xa4c138f7acbcc55c' joined
Zigbee2MQTT:info  2022-01-09 13:10:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138f7acbcc55c","ieee_address":"0xa4c138f7acbcc55c"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-01-09 13:10:56: Configuring '0xa4c138f7acbcc55c'
Zigbee2MQTT:info  2022-01-09 13:10:56: Starting interview of '0xa4c138f7acbcc55c'
Zigbee2MQTT:info  2022-01-09 13:10:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138f7acbcc55c","ieee_address":"0xa4c138f7acbcc55c","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-01-09 13:10:56: Successfully configured '0xa4c138f7acbcc55c'
Zigbee2MQTT:debug 2022-01-09 13:10:56: Device '0xa4c138f7acbcc55c' announced itself
Zigbee2MQTT:info  2022-01-09 13:10:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138f7acbcc55c","ieee_address":"0xa4c138f7acbcc55c"},"type":"device_announce"}'
Zigbee2MQTT:debug 2022-01-09 13:11:01: Received Zigbee message from '0xa4c138f7acbcc55c', type 'readResponse', cluster 'genBasic', data '{"appVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-01-09 13:11:01: Received Zigbee message from '0xa4c138f7acbcc55c', type 'raw', cluster 'genBasic', data '{"data":[25,229,1,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-01-09 13:11:01: Received Zigbee message from '0xa4c138f7acbcc55c', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-01-09 13:11:01: Received Zigbee message from '0xa4c138f7acbcc55c', type 'raw', cluster 'genBasic', data '{"data":[25,230,1,0,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-01-09 13:11:01: Received Zigbee message from '0xa4c138f7acbcc55c', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-01-09 13:11:01: Received Zigbee message from '0xa4c138f7acbcc55c', type 'raw', cluster 'genBasic', data '{"data":[25,231,1,134],"type":"Buffer"}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-01-09 13:11:02: Successfully interviewed '0xa4c138f7acbcc55c', device has successfully been paired
Zigbee2MQTT:info  2022-01-09 13:11:02: Device '0xa4c138f7acbcc55c' is supported, identified as: AwoX LED white (33951/33948)
Zigbee2MQTT:info  2022-01-09 13:11:02: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"LED white","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"33951/33948","options":[{"access":2,"description":"Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to `0` (no transition).","name":"transition","property":"transition","type":"numeric","value_min":0}],"supports_ota":false,"vendor":"AwoX"},"friendly_name":"0xa4c138f7acbcc55c","ieee_address":"0xa4c138f7acbcc55c","status":"successful","supported":true},"type":"device_interview"}'
```

Koenkk, 2022-01-09:
You can use a fingerprint to separately identify this device, example: https://github.com/Koenkk/zigbee-herdsman-converters/blob/8a828859609e1b120fe031b19db6aec58a4f07eb/devices/awox.js#L12

fariouche, 2022-01-09:
Thank you.
So I've used this (naive) extend switch() and it is detected. I will need to implement a custom converter of course but I will do it later when I have some time:
```
const extend = require('zigbee-herdsman-converters/lib/extend');

module.exports = [
    {
	fingerprint: [
		{type: 'Router', manufacturerName: 'AwoX', modelID: 'TLSR82xx', endpoints: [
                {ID: 1, profileID: 260, deviceID: 2028, inputClusters: [0, 3, 4, 4096], outputClusters: [0, 3, 4, 5, 6, 8, 768, 4096]},
                {ID: 3, profileID: 4751, deviceID: 2048, inputClusters: [65360, 65361], outputClusters: [65360, 65361]},
            ]},
	],
        zigbeeModel: ['TLSR82xx'],
        model: '33951/33948',
        vendor: 'AwoX',
        description: 'Remote controller',
        extend: extend.switch(),
    },
];

```

fariouche, 2022-01-09:
Finally, didn't took me that long. Still cannot get the color buttons to work (most likely custom to awox because I get this ```
Zigbee2MQTT:debug 2022-01-09 20:08:02: Received Zigbee message from '0xa4c138f7acbcc55c', type 'raw', cluster 'lightingColorCtrl', data '{"data":[17,21,48,0,42,8,0],"type":"Buffer"}' from endpoint 1 with groupID 32776
Zigbee2MQTT:debug 2022-01-09 20:08:02: No converter available for '33952' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[17,21,48,0,42,8,0],"type":"Buffer"}'

``` )

Here is the full converter:
```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
	fingerprint: [
		{type: 'Router', manufacturerName: 'AwoX', modelID: 'TLSR82xx', endpoints: [
                {ID: 1, profileID: 260, deviceID: 2028, inputClusters: [0, 3, 4, 4096], outputClusters: [0, 3, 4, 5, 6, 8, 768, 4096]},
                {ID: 3, profileID: 4751, deviceID: 2048, inputClusters: [65360, 65361], outputClusters: [65360, 65361]},
            ]},
	],
        zigbeeModel: ['TLSR82xx'],
        model: '33952',
        vendor: 'AwoX',
        description: 'Remote controller',
        fromZigbee: [fz.command_on, fz.command_off, fz.command_step, fz.command_move, fz.command_stop, fz.command_recall, fz.command_step_color_temperature],
        toZigbee: [],
        exposes: [e.action(['on', 'off', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up', 'brightness_move_down', 'brightness_stop', 'recall_1', 'color_temperature_step_up', 'color_temperature_step_down'])],
	configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'genOnOff', 'genLevelCtrl', 'genScenes', 'lightingColorCtrl']);
        },
    },
];

```

closed 2022-02-17 by github-actions[bot]