Koenkk/zigbee2mqtt #10674: [New device support]: Philips Hue gradient lightstrip

antonholmstedt, 2022-01-09:
### Link

https://www.philips-hue.com/sv-se/p/hue-white-and-color-ambiance-ambiance-gradient-lightstrip-2-meter/8719514339965

### Database entry

{"id":30,"type":"Router","ieeeAddr":"0x001788010b8f09ae","nwkAddr":2802,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LCX004","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LCX004","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20210407","swBuildId":"1.86.6"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500,"colorMode":2,"currentX":29967,"currentY":26869,"enhancedCurrentHue":8417,"currentSaturation":140,"colorTemperature":366}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20210407","swBuildId":"1.86.6","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1641732827294,"useImplicitCheckin":true}

### Comments

Got it working using external converter. Added hueExtend but only including the needed parameters below. zigbeeModel and model are correct.

Picture: https://www.assets.signify.com/is/image/Signify/6d60ca09a85a44dc912cad24009a09b3?wid=1500&hei=1125&qlt=82

### External converter

```shell
module.exports = [{
  zigbeeModel: ['LCX004'],
  model: '929002994901',
  vendor: 'Philips',
  description: 'Hue gradient lightstrip',
  meta: { turnsOffAtBrightness1: true },
  extend: hueExtend.light_onoff_brightness_colortemp_color({ colorTempRange: [153, 500] }),
  ota: ota.zigbeeOTA,
}];
```

### Supported color modes

color temperature, color (rbg)

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

antonholmstedt, 2022-01-09:
Btw there is no possibility to set multiple colors of the strip through z2m, right?

Koenkk, 2022-01-09:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

> Btw there is no possibility to set multiple colors of the strip through z2m, right?

Can this device display different colours in different segments of the strip?

antonholmstedt, 2022-01-09:
> Can this device display different colours in different segments of the strip?

Yes it's a digital strip so each diode can display a color independent of the others. Think of it like ambilight. Here is a picture example:

https://www.assets.signify.com/is/image/Signify/0fc36f3f51c44a80b4caad2d00c8ae51?wid=960&hei=720&qlt=82

Koenkk, 2022-01-09:
@antonholmstedt I see, they probably use something custom from that, the Zigbee spec doesn't support such feature. You will have to sniff the traffic with the hue gateway to reverse engineer how this works. https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

antonholmstedt, 2022-01-09:
Thanks, will give it a try!

mluecke09, 2022-01-17:
@Koenkk I captured traffic from hue bridge to the gradient lightstrip while setting the colors for the zones. How can I transfer the pcap files and screenshots from the hue app to you?

Koenkk, 2022-01-18:
You can dump the pacpng files here, also share your network key.

mluecke09, 2022-01-18:
Network key: 51:D2:39:0B:60:1F:39:FB:3A:57:19:F2:6C:9D:4B:EA
[image]
[image]
[gradient_lightstrip_captures.zip](https://github.com/Koenkk/zigbee2mqtt/files/7887854/gradient_lightstrip_captures.zip)

closed 2022-02-26 by github-actions[bot]

tader, 2022-06-17:
> @mluecke09 thanks, as can be seen in your sniff, the device uses a manufacturer specific command for this:
> 
> [image]
> 
> We can also send these commands from z2m, however someone has to reverse engineer the data payload (circled in red). Seems quite complex so not sure how to do that.

I have a hue signe floor lamp with similar functionality. I have been able to sniff a lot of these packets while changing some variables individually.

Could you please point me in the right direction on how to send the manufacturer specific commands so that I can experiment?

Koenkk, 2022-06-18:
I added the command in the latest dev branch now.  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Update to the latest dev and load the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    multicolor: {
        key: ['multicolor'],
        convertSet: async (entity, key, value, meta) => {
            const payload = {data: Buffer.from([0])}; // <-- take from right click on data -> copy hex value 
            await entity.command('manuSpecificPhilips2', 'multiColor', payload);
        },
    },
}

const definition = {
    zigbeeModel: ['TODO'], // <-- update this with the modelID from your data/database.db prob something similar to LCF003, 4080248P7 or 915005987201
    model: 'multicolortest',
    vendor: 'Philips',
    description: 'Hue Signe floor light',
    meta: {turnsOffAtBrightness1: true},
    extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    toZigbee: [...extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}).toZigbee],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- I put two comments in the file, update them accordingly
- start z2m
- sending to `zigbee2mqtt/MY_DEVICE_NAME/set` payload `{"multicolor": ""}` will trigger the command

tader, 2022-06-18:
@Koenkk awesome, thank you!!!

I changed `ext_converter.js` into the following so that I can just send data packets like `{"multicolor": "500104001350000000d649510b5c4deb7c5d8b6d6d2b9b802800"}` to `zigbee2mqtt/MY_DEVICE_NAME/set`.

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const Buffer = require('buffer').Buffer;

const tzLocal = {
    multicolor: {
        key: ['multicolor'],
        convertSet: async (entity, key, value, meta) => {
            const payload = {data: Buffer.from(value, 'hex')};
            await entity.command('manuSpecificPhilips2', 'multiColor', payload);
        },
    },
}

const definition = {
    zigbeeModel: ['4080248U9', '915005987201'],
    model: 'multicolortest',
    vendor: 'Philips',
    description: 'Hue Signe floor light multicolortest',
    meta: {turnsOffAtBrightness1: true},
    extend: extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}),
    toZigbee: [tzLocal.multicolor, ...extend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]}).toZigbee],
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

For now, I have only replayed some sniffed commands.

Koenkk, 2022-06-18:
@tader great to hear!

tader, 2022-06-20:
Progress so far in deciphering the "multicolor" command on my Philips Hue Gradient Signe floor lamp.

The lamp itself seems to have 10 zones, in the command you can configure the color of 5 points on the lamp, these colors "bleed" into adjacent zones. Let's call the lamp zones 0-9 and the points which you can assign a color a-e.

```
      [0][1][2][3][4][5][6][7][8][9]
a     ===---...
b        ---===---...
c              ---======---
d                    ...---===---   
e                          ...---===

===: bleeds heavily into this zone
---: bleeds into this zone
...: bleeds a tiny bit into this zone
```

If you configure the entire lamp white and a single point blue, then the most bluest point is less blue than when you configure two adjacent points blue.

Now command, is made up of `[prefix][point a]...[point e][suffix]`, with prefix `500104001350000000`, and suffix `2800`. I have not looked into the meaning of the prefix and suffix.

Each of the 5 points consists of a 6 digit hexadecimal number, ie., 3 bytes.
Meaning of the digits:

0: color temp? white channel? saturation?
1: no visible effect
2: no visible effect
3: color coordinate (0 blue -- 7 white -- f red)
4-5: color coordinate (0 purple -- 7 white -- f green)

A white point would therefore be: `000FFF`, a warmer white point `f00770`, a blue point `0000100`...

classicsdriver, 2022-08-06:
Wow. This looks like real brain work to figure this out. It would be really great if we could see this feature for the gradient devices (mine is a signe table lamp) in the future. Thanks for all your efforts.

tader, 2022-08-08:
So, I have made no progress in reverse engineering the gradient commands other than as described in my comment above.

On https://developers.meethue.com I did use the contact form to ask for documentation on the command, but I never received a response.

I did however capture all gradient scenes in the Hue app, these are sufficient for my personal use. I could provide a list of those to be used with the multicolor command. However, these gradients are probably under copyright by Signify/Philips.

@Koenkk do you have an idea on how to continue?

Koenkk, 2022-08-08:
@tader what about adding support for all the gradient scenes? This would be a good first step IMO. Reverse engineering these messages tends to be very hard 😄

tader, 2022-08-15:
@Koenkk my [ext_converter.js](https://gist.github.com/tader/4a4f235a8c21b453c921fb6bb591789a) with all gradients from the Hue scenes

Koenkk, 2022-08-16:
@tader 

Thanks for the converter, code looks good, if you want I can also integrate it.

> I'm also still a bit worried about copyright issues.

I'm not really worried tbh. The only thing that has been copied are the names, it doesn't contain the x/y or rgb codes of the colors (technically yes of course, but nobody can reverse engineer them from the payloads).

tader, 2022-08-17:
@Koenkk I would like to take you up on your offer to integrate it. I'm not familiar enough with the project to do it the proper way. Thank you!

Koenkk, 2022-08-17:
Integrated it, thanks @tader 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

tader, 2022-08-17:
Thank you @Koenkk !!!

notorioustubze, 2022-10-02:
>             const payload = {data: Buffer.from([0])}; // <-- take from right click on data -> copy hex value 

Hi, firstly thank you for all the great work you do I really appreciate the effort you put in. Sorry if it's really simple but could you advise how I obtain the HEX please I'm editing this file via shell on Proxmox do I need to be editing somewhere else to be able to right click? TIA

tader, 2022-10-03:
Hi @notorioustubze, Koenk is here talking about clicking in [Wireshark](https://www.wireshark.org) when [Sniffing Zigbee traffic](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html)

zegl, 2022-10-19:
I saw that @ebaauw from the deconz project had successfully _decoded_ the payload (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5891#issuecomment-1280886750). I've been testing it, and can say that I think that the mystery is solved! 🎉 Huge thanks to @ebaauw and @manup!

Each "point" is a 24 bit unsigned integer, of two 12 bit numbers for the XY coordinates, each individually scaled.

On the Signe floor lamp (`915005987201`), the points are ordered bottom-to-top, i.e. the first "point" in the payload sets the color of the leds at the bottom of the device.

The values are decoded like so:

* `0edb63` physically looks to be a red color (from the "ruby_romance" preset)
* `0edb63` is read as `[ b0e, 63d ]` (unsigned 24-bit integer, split into high and low bits for the x and y values)
* `x = hextobin(b0e) * 0.7347 / 4095 = 0.5077`
* `y = hextobin(63d) * 0.8413 / 4095 = 0.3280`
* `xy(0.5077, 0.3280)` is `#ff4739`!

Encoding can be done by applying the same process in reverse.

-----

**IRL testing:**

Picking two colours, one red (0edb63 / `#ff4739`) and a green (37a7a3 / `#7dff20`), and constructing a repeated pattern of alternating colors:

```
50010400135000000037a7a30edb6337a7a30edb6337a7a32800

500104001350000000|37a7a3|0edb63|37a7a3|0edb63|37a7a3|2800
                  |      |      |      |      |      |
                  |111111|222222|333333|444444|555555|
                  |green |red   |green |red   |green |   
```

Sending this payload to the device and... success!

[image]

----

**Example code**, for converting to and from the wire format is included below. I'm not entirely sure that it's correct, but it seems to do the trick! I've been testing it with various different RGB colours to create my own gradient presets.

```js
import { ColorRGB, ColorXY } from './lib/color.js'

function rgbtohex(rgb) {
    var r = rgb.red * 255; if (r < 0) { r = 255 };
    var g = rgb.green * 255; if (g < 0) { g = 255 };
    var b = rgb.blue * 255; if (b < 0) { b = 255 };
    return [
        "#",
        Math.round(r).toString(16).padStart(2, "0"),
        Math.round(g).toString(16).padStart(2, "0"),
        Math.round(b).toString(16).padStart(2, "0"),
    ].join("")
}

function decodePointToHex(p) {
    var x = p[3] + p[0] + p[1]
    var y = p[4] + p[5] + p[2]

    var xx = (parseInt(x, 16) * 0.7347 / 4095).toFixed(4)
    var yy = (parseInt(y, 16) * 0.8413 / 4095).toFixed(4)

    var rgb = new ColorXY(xx, yy).toRGB()
    var hex = rgbtohex(rgb)
    return hex
}

function encodeHexToPoint(hex) {
    var rgb = ColorRGB.fromHex(hex)
    var xy = rgb.toXY()
    var x = xy.x * 4095 / 0.7347
    var y = xy.y * 4095 / 0.8413
    var xx = Math.round(x).toString(16)
    var yy = Math.round(y).toString(16)

    return [
        xx[1], xx[2],
        yy[2], xx[0],
        yy[0], yy[1],
    ].join("")
}

// "chinatown"
var input = "500104001350000000b33e5b408e59d90d5b4c6c6360ac792800"
var s = input.slice("500104001350000000".length, -4)
var pairs = s.match(/.{6}/g)

var hexes = pairs.map(decodePointToHex)
// [ '#ff0506', '#ff070d', '#ff0d11', '#ff2f20', '#ff5404' ]
console.log(hexes)

var points = hexes.map(encodeHexToPoint)
// [ 'b44e5b', '41ae59', 'da0d5b', '4b7c63', '62bc79' ]
console.log(points)
```

ebaauw, 2022-10-19:
We found out some more stuff, with help from Thomas Vos, the author of Hue Essentials, but that's currently only mentioned on Discord.  The number of colours is actually variable, encoding as follows:
```
50010400|13|50|000000|37a7a3|0edb63|37a7a3|0edb63|37a7a3|28|00
        |  |  |      |      |      |      |      |      |
        |ll|cc|000000|111111|222222|333333|444444|555555|ss|oo
                     |green |red   |green |red   |green |  
```
- `ll` is the length of the variable part: 1 + 3 * (#colours + 1), so in this case 19 = 0x13;
- `cc` is the number of colours left shifted 4 bits, so in this case 5 << 4 = 0x50;
- Not sure what the `000000` does.  Haven't yet tried changing that;
- `ss` is the number of segments, left shifted 3 bits, so in this case 5 << 3 = 0x28;
- `oo` is the offset of the colour for the first segment, also left shifted 3 bits, so in this case 0, meaning the first segment takes takes the first colour.

The number of segments can differ from the number of colours.  If it's larger, the colours repeat, if smaller, superfluous colours are simply not used.  Apparently you can specify up to 9 colours (even though the Hue bridge won't go beyond 5, and the app only shows 3), and possibly up to 31 segments (0xF8).  Not yet sure if that changes with extensions.

So you could have set the same effect by:
```
50010400|0a|20|000000|37a7a3|0edb63|28|00
```
or by
```
50010400|0a|20|000000|0edb63|37a7a3|28|08
```

The Hue API v2 also exposes `"mode": "interpolated_palette"` and `"pixel_count": 16`, don't know yet if/how these are encoded.  There could be something in the least significant three bits (0x07) of the last two bytes, in the `000000` or maybe somewhere in the `50010400`.

tader, 2022-10-19:
Awesome findings @zegl @ebaauw !!!! 😍

Filoumino, 2022-11-05:
Hello to all! Thank you for your sensational work. I am just starting to integrate all my Hue lights into Zigbee2MQTT. 
Unfortunately, the Philips Hue Play Gradient 75" lightstrip is not yet included in the dev branch. Is it possible to add it?

I guess it would have to look like this:

zigbeeModel: ['LCX003'],
        model: '929002422901',
        vendor: 'Philips',
        description: 'Hue Play gradient lightstrip 75',
        extend: hueExtend.light_onoff_brightness_colortemp_color({colorTempRange: [153, 500]})

I would be totally grateful!

jb123567, 2022-12-08:
Looks like this is now sorted? @zegl am I right in thinking you've worked out how to achieve this? I'm assuming it will hit edge soon?

zegl, 2022-12-08:
@jb123567 Yes, it should already be out on the latest dev build! There is still some work to be done to support more devices and to add gradient support on the frontend. 

But starting now you can send something like

```json
{"gradient":[ "#0a96ff","#623aff","#ff51ea","#ff13de","#3e01ff"]}
```

to `"zigbee2mqtt/${NAME}/set"` to test things out.

Zegorax, 2022-12-25:
@zegl Thank you very much for your work! I have a Table Sign Gradient and I tried it on the latest-dev docker image tag, however it does not seem to be working. The device ID is 915005987001.

I'm assuming it is because the converter has not been assigned to this device ID, is that correct ?

zegl, 2023-01-02:
@Zegorax Support for your device is coming in https://github.com/Koenkk/zigbee-herdsman-converters/pull/5248 :-)

Zegorax, 2023-01-04:
@zegl Thanks a lot ! I just tested and it works!
The effect is not "rolling" yet, is it an optional option to add in the payload ? In the Hue app, if I remember correctly, it is called "live scene" or something like this.

tader, 2023-01-04:
If I remember correctly from sniffing the zigbee traffic, you would first have to define the scene on the lamp itself, and then you can send a command to start it as a live scene.

marblecoma, 2023-01-05:
> There is still some work to be done to support more devices and to add gradient support on the frontend.

@zegl When I look at the Z2M frontend for the Hue Gradient Lightstrip (LCX004), I see "Unknown access "Color in RGB HEX format (eg #663399)"" in each space where a gradient hex code could be input (screenshot below). Is this expected and what you meant with still needing to add gradient support to the frontend? 

[image]

notorioustubze, 2023-01-27:
Hi all, I’m running the latest dev build but I can’t see any options for the gradient. I’ve got the 55inch Gradient Light strip. Can anyone advise what I’m doing wrong. I know front end is missing but I can’t see anything in z2m either? Tia

zegl, 2023-01-28:
@notorioustubze Could you please share the entry of the light strip from the zigbee2mqtt database?

marblecoma, 2023-01-28:
@notorioustubze What options are you expecting and where are you expecting to see them? What front end are you referring to? You refer to Z2M as something separate from your "front end". In the Z2M "Exposes" tab, you should be able to select a "gradient_scene" from the drop down menu.

notorioustubze, 2023-01-29:
@zegl - {"id":20,"type":"Router","ieeeAddr":"0x00178801095ea075","nwkAddr":61171,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LCX001","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LCX001","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20221006","swBuildId":"1.101.2"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genLevelCtrl":{"attributes":{}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20221006","swBuildId":"1.101.2","zclVersion":2,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1674993302677,"defaultSendRequestWhen":"immediate"}

notorioustubze, 2023-01-29:
> @notorioustubze What options are you expecting and where are you expecting to see them? What front end are you referring to? You refer to Z2M as something separate from your "front end". In the Z2M "Exposes" tab, you should be able to select a "gradient_scene" from the drop down menu.

Exactly this a gradient_scene option in the drop down menu in exposes however I only have the following options:
[image]

marblecoma, 2023-01-29:
> Exactly this a gradient_scene option in the drop down menu in exposes however I only have the following options: [image]

That's Effect drop down. You should see a separate one called gradient_scene

notorioustubze, 2023-01-29:
Yeah I don’t see anything referencing the gradient option. Any ideas?Sent from my iPhone

marblecoma, 2023-01-30:
@notorioustubze I don't have any Hue Play Gradient lightstrips to test. If you pair it with the Hue app, are you able to set a gradient? Are you sure you're on the dev version of Z2M? You can check in the Settings --> About tab in the Z2M web ui. Have you updated the firmware of the lightstrip to the most recent version?

zegl, 2023-01-30:
@notorioustubze The `LCX001` should support gradients just fine. What version of Z2M are you running? You can get the version number from "Settings" > "About" in the UI.

If the issue persists after making sure that you're on the latest version, please open a new issue for improving the support of your device, instead of commenting on this old issue that's already implemented and closed.

notorioustubze, 2023-01-30:
[image]
Running the following version and yeah gradient works fine with Hue app.

Sirgrant618, 2023-10-26:
> [image]
> 
> Running the following version and yeah gradient works fine with Hue app.

Hey @notorioustubze, 

How are you accessing Z2M in the Hue app? Are you using DiyHue?