Koenkk/zigbee2mqtt #10675: [New device support]: Support needed for 'BSEED ZigBee Wall Sockets Power Smart Sockets '

JawaharN, 2022-01-09:
### Link

https://www.aliexpress.com/item/1005003594581166.html?spm=a2g0o.9042311.0.0.46114c4dRODQGO

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c1383e4c71d7b7","nwkAddr":36678,"manufId":4417,"manufName":"_TZ3000_o1jzcxou","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_o1jzcxou","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53264":2,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1641734368313,"useImplicitCheckin":true}

### Comments

I do not know how to work with new device

JawaharN, 2022-01-09:
I created BSEED.js, and it is working with switch option

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
	fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_o1jzcxou'}],
	model: 'TS011F_socket_module',
	vendor: 'TuYa',
	description: 'Socket module',
	extend: extend.switch(),
	whiteLabel: [{vendor: 'BSEED', model: 'UNKNOWN'}],
};

module.exports = definition;
```

Koenkk, 2022-01-09:
Added this device, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-09 by Koenkk