Koenkk/zigbee2mqtt #10677: [New device support]: PIR motion detector alarm tuya TS0202 _TZ3000_tiwq83wk

jocamane, 2022-01-09:
### Link

https://pt.aliexpress.com/item/1005002686808050.html?spm=a2g0s.12269583.0.0.116d208eswfgmB

### Database entry

warn  2022-01-09 16:36:55: Device '0xa4c1382256c542a2' with Zigbee model 'TS0202' and manufacturer name '_TZ3000_tiwq83wk' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html warn  2022-01-09 16:36:56: Received message from unsupported device with Zigbee model 'TS0202' and manufacturer name '_TZ3000_tiwq83wk' warn  2022-01-09 16:36:56: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html warn  2022-01-09 16:36:57: Received message from unsupported device with Zigbee model 'TS0202' and manufacturer name '_TZ3000_tiwq83wk' warn  2022-01-09 16:36:57: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### Comments

I tried in edge version and prodution version

Danit2, 2022-01-11:
Try this as an external converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =[
    {
        fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_ef5xlc9q'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_vwqnz1sn'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_2b8f6cio'},
            {modelID: 'TS0202', manufacturerName: '_TZE200_bq5c8xfe'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_qjqgmqxr'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_mmtwjmaq'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_kmh5qpmb'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_zwvaj5wy'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_mcxw5ehu'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_msl6wxk9'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_tv3wxhcz'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_hqbdru35'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_tiwq83wk'},
            {modelID: 'WHD02', manufacturerName: '_TZ3000_hktqahrq'}],
        model: 'TS0202',
        vendor: 'TuYa',
        description: 'Motion sensor',
        whiteLabel: [{vendor: 'Mercator Ikuü', model: 'SMA02P'}, {vendor: 'TuYa ', model: 'TY-ZPR06'}],
        fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report],
        toZigbee: [],
        exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery()],
    },
]

module.exports = definition;
```

When it works, i will add it.

jocamane, 2022-01-12:
Please tell me how todo it ?
I don't know 
Thanks

Danit2, 2022-01-12:
Make a file with name `pir_test.js` and put my code in. This file has to be created next to the `configuration.yaml`
Add in `configuration.yaml` under external converter `pir_test.js` and restart Z2M.

Like in this doc: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device

jocamane, 2022-01-12:
I already tested and it worked. Thanks 
how do I know when I can remove this situation and that the Z2M is already implemented?

Danit2, 2022-01-12:
After the next update.

Koenkk, 2022-01-13:
added

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-13 by Koenkk

jocamane, 2022-01-30:
hi,

I noticed now that this sensor does not have a battery % indication. What do I need to do?

Thanks

{"id":35,"type":"EndDevice","ieeeAddr":"0xa4c138981d3988ff","nwkAddr":51615,"manufId":4417,"manufName":"_TZ3000_mmtwjmaq","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"powerSource":3,"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"currentZoneSensitivityLevel":1,"iasCieAddr":"0x00212effff07afaa","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1643543150792,"useImplicitCheckin":true}

jocamane, 2022-01-30:
[image]

jocamane, 2022-01-30:
If i implement the external converters, the sensor does not work. does not detect movement.
if i delete the external converter works ok.
in both cases the battery % is not detected 
thanks again

jocamane, 2022-02-19:
Hi, Any news about battery % ?

Thanks