Koenkk/zigbee2mqtt #10678: [Problem]:  Danfoss 014G2461

2Max2Pax, 2022-01-09:
### What happened?

I have 9 Danfoss 014G2461, every 2-3 hours zigbee2mqtt stops communicating with one or two devices (not always the same one) in both directions.
The only way to restart the communication is restarting zigbee2mqtt.
I cannot see any error in the logs of z2m or mosquitto.

### What did you expect to happen?

Once started the communication shold be stable.

### How to reproduce it (minimal and precise)

Try to change the temperature publishing:
Topic: zigbee2mqtt/FRIENDLY_NAME/set
Payload: {"occupied_heating_setpoint":TEMPERATURE}

### Zigbee2MQTT version

1.22.2 commit: 414c51f

### Adapter firmware version

20211217

### Adapter

Gio-dot Z-Bee Duo with CC2652P

### Debug log

[logz2m.txt](https://github.com/Koenkk/zigbee2mqtt/files/7835606/logz2m.txt)
The device that is not working is :
```
0x84fd27fffed4388c:
friendly_name: trv_ingresso
```
All TRVs' friendly names are: trv_XXXX

lux73, 2022-01-09:
how many other Devices so you have?

How many Routers/Bulb, which Manufacturer?

I have some Danfoss too - there ist no problem with communication so far...

2Max2Pax, 2022-01-09:
28 devices:
17 Vimar switches that act as routers
9 Danfoss 014G2461 (battery)
1 Aqara WSDCGQ11LM (battery)
1 Aqara MCCGQ11LM (battery)

lux73, 2022-01-09:
don't know the mentioned Vimar Switches

but had same some serious issues with some Zigbee Devices that act as Router: eg IKEA Bulbs und lastly with a MiBoxer RGB CCT Controller - they work terrible as Router. I had to remove them to stabilize my Network :-(

even don't know your Coordintor: Gio-dot Z-Bee Duo with CC2652P

lastly another User had serious Problems with a ConBee Stick: https://github.com/Koenkk/zigbee2mqtt/discussions/10119#discussioncomment-1865407

2Max2Pax, 2022-01-09:
> 
> don't know the mentioned Vimar Switches
> 

See: [Vimar 14592](https://www.zigbee2mqtt.io/devices/14592.0.html) 

> but had same some serious issues with some Zigbee Devices that act as Router: eg IKEA Bulbs und lastly with a MiBoxer RGB CCT Controller - they work terrible as Router. I had to remove them to stabilize my Network :-(
> 

This would be the last chance, I spent over 500 Eur for those switches. 
For the time being I created an automation that restarts z2m every hour .....

> even don't know your Coordintor: Gio-dot Z-Bee Duo with CC2652P

See: [Z Bee Duo](https://github.com/Gio-dot/Z-Bee-Duo)
There are no issues about it as far as I know

lux73, 2022-01-09:
> This would be the last chance, I spent over 500 Eur for those switches. For the time being I created an automation that restarts z2m every hour .....

could not imagine thats good for your z2m Mesh Network...  I'm using slaesh's CC2652RB and stopping my z2m for Updates or slightly changes in the configuration only

I'm even not using Docker - runs on bare Metal APU 2C4 Debian Buster

2Max2Pax, 2022-01-10:
I forgot that I have also a Vimar 14594 (cover) , which is zigbee 2.0.
I've excluded it from my zigbee network.... I hope the problem was only that device.

If not I must build a second zigbee network. In this case, could I run a second instance of z2m in a docket container? 
For sure buying a new zigbee coordinator + a router is cheaper than changing my end devices.

lux73, 2022-01-10:
> ... could I run a second instance of z2m in a docket container? For sure buying a new zigbee coordinator + a router is cheaper than changing my end devices.

no problem - make sure you use a **different Channel, PAN ID & ext PAN ID**

btw the Problem that some zigbee Devices crashes the Zigbee Mesh has nothing to do with zigbee 2.0 - even 3.0 Devices are crappy...

2Max2Pax, 2022-01-11:
The problem still happens. I cannot exclude a problem with the routers/devices but the behavior of the web frontend is strange: when one of the TRV does not communicate and I set the temperature (occupied_heating_setpoint), or any other attribute,  through the web frontend nothing is published.

antst, 2022-01-11:
I also got issues with those TRVs lately. Can not pair properly new TRVs, even if they are 20 cm from router. Interview never completes anymore.

antst, 2022-01-11:
Was to quick, old TRVs now also went to 
```"interview_completed":false,```

antst, 2022-01-11:
After  many retries, I managed to get them somehow working again, but still they marked as failed interview, even if they seems to pass interview and bind to coordinator for reporting again.
[image]

antst, 2022-01-11:
There should be some regression somewhere.

2Max2Pax, 2022-01-12:
> After many retries, I managed to get them somehow working again, but still they marked as failed interview, even if they seems to pass interview and bind to coordinator for reporting again.

Try this:
unpair the 2 new TRVs and make the factory reset procedure.
pair them again.

2Max2Pax, 2022-01-13:
New update.
Since the day before yesterday everything works fine. This makes finding the reason of the problem even more difficult.

2Max2Pax, 2022-01-17:
I haven't seen the problem in the last 4 days.
There are no more communication problems.

closed 2022-01-17 by 2Max2Pax

DonDonatello, 2022-10-08:
This resembles the issue in #13478, might be related